google-coral/pycoral #3: Siamese network support

Cganzorig, 2020-11-13:
The topic is not about an issue with pycoral. Just would like to ask if there is any way to support seamese network on edge TPU. Any suggestions will be appreciated.

Namburger, 2020-11-13:
@Cganzorig I'm not so sure about Siamese, since from what I can tell it will requires 2 inputs at the same time, which the edgetpu wouldn't be able to support. 
@Naveen-Dodda do you know much about this network?

Naveen-Dodda, 2020-11-18:
@Cganzorig officially we haven't tested a seamese network on edgetpu. But we do support multiple input/output  tensors. As long as tensors fit in on chip memory and all the ops are supported by compiler. I do n't see any problem. 

I recommend you to give a try : first step is test a Int8 quantised model.

Hope this help

Naveen-Dodda, 2021-05-13:
@Cganzorig 

Are you still working on seamese netwrok ?

closed 2021-05-26 by ngrutas