google-coral/pycoral #5: PyCoral Vs TfLite_runtime

vinceab, 2020-11-24:
Hello,

By reading the python quickstart:
https://www.tensorflow.org/lite/guide/python
I understand that the tflite_runtime is installed thank to python wheel packages regenerated via pycoral.
Is my understanding right?

In that case what is their any adherence with the tflite_runtime generated via TensorFlow Lite github.
The version also does not match. Official Tensorflow Lite version (so runtime) is v2.3.1 and in the python quick start guide I can see a tflite_runtime version 2.5.0.

Further, running a TPU model using the Tensorflow Lite runtime v2.3.1 is failing:
RuntimeError: Internal: Unsupported data type in custom op handler: 5105160Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.

Can you please provide more information regarding pyCoral Vs TfLite_runtime?

Thanks
Vincent

Namburger, 2020-11-24:
Hi @vinceab 
The only different between pycoral and tflite_runtime is the API. We added adapters for easy parsing of outputs for common models like classification and detections. [Here](https://github.com/google-coral/pycoral/blob/master/examples/detect_image.py#L85) is an example of that.

Regarding to version, you should be using 2.5.0 for the tflite_runtime package, not  v2.3.1, may I know where you got v2.3.1 from?

vinceab, 2020-11-24:
Hi @Namburger 

I actually get the TFLite_intepreter by regenerating the python package from TensorFlow github version 2.3.1:
tensorflow/lite/tools/pip_package/build_pip_package.sh
As the last stable version is the version 2.3.1 I am expecting the TFLite_interpreter version is v2.3.1.

May I ask you same question? :) From wich source code did you get the 2.5.0 TFLite_runtime version?

I am a little bit puzzled between TFLite_runtime generated via tensorflow (that's seems not support EdgeTPU) and the information I read regarding pyCoral and the Python quickstart guide.

Many thanks for your lights.
Vincent

Vincent

Namburger, 2020-11-24:
hello @vinceab 
Oh I see, the tflite_runtime package that we built and release is from [this TensorFlow commit](https://github.com/google-coral/pycoral/blob/master/WORKSPACE#L18). The special thing about it is just that is is the same commit that we used to load `libedgetpu` dynamic library, so it is crucial that both matches or else you'll get:
```
RuntimeError: Internal: Unsupported data type in custom op handler: 5105160Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.
```

vinceab, 2020-11-24:
@Namburger 

So if I understand well, the version of the tflite_runtime v2.5.0 you released is not correlated with the TensorFlow versioning.
The branch your are referring to is in the master branch? The ancestor seems to be v1.12.1 which is old no? How far are is this commit from the v2.3.1?
Frankly speaking it is a really difficult to understand the merging flow.
Is it planned to align both Coral and TensorFLow?
Thanks
Vincent

vinceab, 2020-11-25:
Its is now pretty clear by reading this:
https://coral.ai/docs/notes/build-coral

I understand that I need to rebuild the libedgetpu against the tensorflow lite v2.3.1 if I want it working with my runtime.

Vincent

Namburger, 2020-11-25:
@vinceab yes, and I understand it could get confusing. Anyhow:

> Is it planned to align both Coral and TensorFLow?

Yes we do, and actually, it is aligned, it's just that any of the stable released branches right now doesn't contains all the features that libedgetpu currently use. So that commit is actually from the master branch, ahead of all the stable release!

Closing for now, feel free to continue the questions here!

closed 2020-11-25 by Namburger