google-coral/pycoral #6: Python 3.9 support

fgervais, 2020-12-04:
I'd like to try and build tflite_runtime myself for python 3.9.

Any guidelines on how I would go about doing that?

Namburger, 2020-12-04:
Hello @fgervais 
You'd have to create a target for python3.9 and build from here as we don't have supports for 3.9 at this time: 
https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/tools/pip_package

Also, make sure that you checked out [this commit](https://github.com/google-coral/pycoral/blob/master/WORKSPACE#L18) of tensorflow before building it!

fgervais, 2020-12-08:
I want to report my progress as it seems I might have a working milestone.

What I did is I went in there:

https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/tools/pip_package

at current `tensorflow` `master` (`5453cf0b1c8359c0ee6afef8393b350c832a9123`)

Then I commented out the python2 stuff as it doesn't seem to be packaged for newer debian version:

```patch
diff --git a/tensorflow/lite/tools/pip_package/Dockerfile b/tensorflow/lite/tools/pip_package/Dockerfile
index 834fe5efca7..cc8c22acf89 100644
--- a/tensorflow/lite/tools/pip_package/Dockerfile
+++ b/tensorflow/lite/tools/pip_package/Dockerfile
@@ -10,15 +10,15 @@ RUN apt-get update && \
     apt-get install -y \
       debhelper \
       dh-python \
-      python-all \
-      python-setuptools \
-      python-wheel \
-      python-numpy \
-      python-pip \
+      # python-all \
+      # python-setuptools \
+      # python-wheel \
+      # python-numpy \
+      # python-pip \
       pybind11-dev \
-      libpython-dev \
-      libpython-dev:armhf \
-      libpython-dev:arm64 \
+      # libpython-dev \
+      # libpython-dev:armhf \
+      # libpython-dev:arm64 \
       python3-all \
       python3-setuptools \
       python3-wheel \
```

Then I built using bullseye as base as it comes with python `3.9.0-4`

`make BASE_IMAGE=debian:bullseye PYTHON=python3 TENSORFLOW_TARGET=aarch64 docker-build`

For now it seems to work:

```
root@buildroot:~# pip install /tflite_runtime-2.5.0-cp39-cp39-linux_aarch64.whl 
Processing /tflite_runtime-2.5.0-cp39-cp39-linux_aarch64.whl
Requirement already satisfied: numpy>=1.16.0 in /usr/lib/python3.9/site-packages (from tflite-runtime==2.5.0) (1.18.2)
Installing collected packages: tflite-runtime
Successfully installed tflite-runtime-2.5.0
root@buildroot:~# python
Python 3.9.0 (default, Dec  8 2020, 17:05:44) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tflite_runtime.interpreter as tflite
>>> 
```

Namburger, 2020-12-09:
Nice!!!!

fgervais, 2020-12-11:
I tested the package I built on an M.2 edgetpu module and it does inference fine.

Should I close this issue or do we leave it open until official support for python 3.9?

Namburger, 2020-12-11:
@fgervais yes, please keep it open until we have an official package, thanks!

thekazak, 2021-01-02:
@fgervais it helps build install tensorflow/lite with python 3.9
but what about pycoral? I need it on python 3.9 together with tensorflow lite

fgervais, 2021-01-02:
Humm I'm not sure, I haven't used pycoral for my inferences.

susmitpy, 2021-06-26:
> `5453cf0b1c8359c0ee6afef8393b350c832a9123`

Can you provide the tflite runtime wheel file you have made ?

dmitriykovalev, 2021-07-28:
We have `Python 3.9` wheels now, please check [releases](https://github.com/google-coral/pycoral/releases) page.

closed 2021-07-28 by dmitriykovalev

jogugil, 2025-03-27:
Hi, I'm working with a Coral Dev Board (Raspberry Pi + Edge TPU) and I’m trying to set up a container to install everything I need. However, I’m facing an issue: it says that it’s not compatible with Python 3.9. Can you help me figure out what I need to modify to make it work? I'm even unable to install the examples or run inference examples. Here is my Dockerfile:

dockerfile
 
```
# Use Debian Bullseye (Debian 11) as the base image
FROM debian:bullseye

# Set the environment to be non-interactive for automation
ENV DEBIAN_FRONTEND=noninteractive

# Update existing packages and install necessary tools
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
    sudo \
    openssh-server \
    python3 \
    python3-pip \
    python3-venv \
    python3-dev \
    git \
    wget \
    curl \
    nano \
    udev \
    usbutils \
    libusb-1.0-0-dev \
    libopencv-dev \
    python3-opencv \
    python3-numpy \
    python3-scipy \
    && apt-get clean && \
    rm -rf /var/lib/apt/lists/*

# Install PyCoral API
RUN apt-get update && \
    apt-get install -y python3-pycoral && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

# Add Google Coral repositories to ensure the latest version of libedgetpu
RUN echo "deb http://packages.cloud.google.com/apt edgetpu-stable main" | tee -a /etc/apt/sources.list.d/edgetpu.list && \
    curl -sS https://packages.cloud.google.com/apt/doc/apt-key.gpg | apt-key add - && \
    apt-get update

# Install libedgetpu and python3-edgetpu
RUN apt-get install -y libedgetpu1-std python3-edgetpu && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

# Clone PyCoral examples repo and download necessary models
RUN git clone https://github.com/google-coral/pycoral /pycoral && \
    cd /pycoral && \
    bash examples/install_requirements.sh

# Create a non-root user with sudo privileges
RUN useradd -m -s /bin/bash coral && \
    echo 'coral:coral' | chpasswd && \
    usermod -aG sudo coral

# Configure SSH service
RUN mkdir /var/run/sshd && \
    echo 'coral ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers && \
    sed -i 's/#PermitRootLogin prohibit-password/PermitRootLogin no/' /etc/ssh/sshd_config && \
    sed -i 's/#PasswordAuthentication yes/PasswordAuthentication yes/' /etc/ssh/sshd_config

# Configure udev to detect TPU and USB devices
RUN echo 'SUBSYSTEM=="usb", ATTR{idVendor}=="1a6e", MODE="0666"' > /etc/udev/rules.d/99-edgetpu.rules

# Expose port 22 for SSH access
EXPOSE 22

# Set working directory
WORKDIR /home/coral

# Switch to non-root user
USER coral

# Keep the container running with SSH and Bash
CMD /usr/sbin/sshd -D & /bin/bash
```

Here is the error I get when building the Docker image:
 
 
```
 => CACHED [ 1/10] FROM docker.io/library/debian:bullseye@sha256:5c028c19adaab7849171183060074  0.0s
 => [ 2/10] RUN apt-get update &&     apt-get install -y --no-install-recommends     sudo      75.1s
 => ERROR [ 3/10] RUN apt-get update &&     apt-get install -y python3-pycoral &&     apt-get   2.9s
------
 > [ 3/10] RUN apt-get update &&     apt-get install -y python3-pycoral &&     apt-get clean &&     rm -rf /var/lib/apt/lists/*:
0.409 Get:1 http://deb.debian.org/debian bullseye InRelease [116 kB]
0.492 Get:2 http://deb.debian.org/debian-security bullseye-security InRelease [27.2 kB]
0.513 Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
0.612 Get:4 http://deb.debian.org/debian bullseye/main amd64 Packages [8066 kB]
0.767 Get:5 http://deb.debian.org/debian-security bullseye-security/main amd64 Packages [352 kB]
0.822 Get:6 http://deb.debian.org/debian bullseye-updates/main amd64 Packages [18.8 kB]
1.662 Fetched 8624 kB in 1s (6375 kB/s)
1.662 Reading package lists...
2.139 Reading package lists...
2.622 Building dependency tree...
2.739 Reading state information...
2.851 E: Unable to locate package python3-pycoral
------
  1 warning found (use docker --debug to expand):
 - JSONArgsRecommended: JSON arguments recommended for CMD to prevent unintended behavior related to OS signals (line 75)
Dockerfile:31
--------------------
  30 |     # Instala la API de PyCoral
  31 | >>> RUN apt-get update && \
  32 | >>>     apt-get install -y python3-pycoral && \
  33 | >>>     apt-get clean && \
  34 | >>>     rm -rf /var/lib/apt/lists/*
  35 |
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-get update &&     apt-get install
```