google-coral/pycoral #7: support for armv6l?

prettyflyforabeeguy, 2020-12-15:
Is there support for armv6l  (i.e. Raspberry Pi Zero W)
If no support exists currently, is there a way I can build this natively to support this ARM32 device? 

More about the device I'm attempting to use the tflite_runtime on:
$ python3 --version
	Python 3.7.3
	
$ cat /etc/os-release
	PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
	NAME="Raspbian GNU/Linux"
	VERSION_ID="10"
	VERSION="10 (buster)"
	VERSION_CODENAME=buster
	ID=raspbian
	ID_LIKE=debian
	HOME_URL="http://www.raspbian.org/"
	SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
	BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
		
$ cat /proc/cpuinfo
	processor       : 0
	model name      : ARMv6-compatible processor rev 7 (v6l)
	BogoMIPS        : 697.95
	Features        : half thumb fastmult vfp edsp java tls
	CPU implementer : 0x41
	CPU architecture: 7
	CPU variant     : 0x0
	CPU part        : 0xb76
	CPU revision    : 7

	Hardware        : BCM2835
	Revision        : 9000c1
	Serial          : 00000000fdca881b
	Model           : Raspberry Pi Zero W Rev 1.1

$ uname -a
	Linux experimentalpiZero 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l GNU/Linux
	
$ uname -m
	armv6l

$ gcc -v
	Using built-in specs.
	COLLECT_GCC=gcc
	COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
	Target: arm-linux-gnueabihf
	Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc- 
        8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only -- 
       program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib -- 
       without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu -- 
       enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object -- 
       disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target- 
       system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with- 
       float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf -- 
       target=arm-linux-gnueabihf
		Thread model: posix
		gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)

Namburger, 2020-12-15:
Hello, this issue have been addressed before and users have been success in doing this. You can take a look at this thread for more info: https://github.com/google-coral/edgetpu/issues/229#issuecomment-704576309

closed 2020-12-15 by Namburger

prettyflyforabeeguy, 2020-12-15:
Thank you for the quick reply, but I'm am not using docker and I am not trying to use libedgetpu, however I'm willing to try anything if I can get tflite working on the armv6l
I'm trying to follow the instructions https://www.tensorflow.org/lite/guide/build_rpi and https://www.tensorflow.org/lite/guide/python 
I can import tensorflow just fine, but trying to use the tflite_runtime is where things fall appart.  I get an illegal instruction the moment I try import tflite_runtime.interpreter as tflite

~/tensorflow_src $ python3 -c 'print(__import__("tflite_runtime").__git_version__)'
0.6.0-76902-gd855adfc5a

~/tensorflow_src $ git rev-parse 0.6.0-76902-gd855adfc5a
d855adfc5a0195788bf5f92c3c7352e638aa1109

I'm trying to piece the conversation together from the #229 comment, but I'm not quite sure i'm followling the necessary steps.
I appreciate any addtional guidence you may have!

Namburger, 2020-12-15:
@prettyflyforabeeguy hello, docker is specifically and it is very important piece for building and you won't be able to do anything without libedgetpu.so. 
Basically you'll need to compile the needed libraries on an x86_64 machineand cross compile it for armv6l. We already provided compatible docker containers to get this going so without that it'll only make it harder for you to get thins going!!

Here is the gist:

1) The 3 peaces that is necessary for you to use our HW from here on out is : 
* libedgetpu
* tflite_runtime
* tensorflow-lite.a

2) Since tensorlfow-lite.a is statically linked to both libedgetpu AND tflite_runtime, you won't need that for your application it self, however you'll NEED to make sure that libedgetpu and the tflite_runtime are both linked against the same exact tensorflow commit (hence we use git rev-parse to find out the exact tensorflow commit).

Okay, so here is where the fun begins :)

> I get an illegal instruction the moment I try import tflite_runtime.interpreter as tflite

That is because the tflite_runtime package was not built to supports the RPI0 with armv6l architecture. So you'll need to build your own package, the instruction to build it is [here](https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/tools/pip_package. You'll need to dig into those build fles to find out which instruction is causing the illegal instructions on your pi. We don't support this arch anymore so we can't fix those built file to bring support back, unfortunately..... HOWEVER, in #229, @rhadnum already mentioned that he found a known working tflite_runtime package that is working with the pi0 and even include the [download link](https://drive.google.com/file/d/1mW8QGmhfk4kRYXqUTEH1ckPUNIsl7S3d/view?usp=sharing).
Therefore my suggestion is to remove the currently installed package and start with that link. 
Then get the tensorflow commit version (same command you posted).

```
~/tensorflow_src $ python3 -c 'print(import("tflite_runtime").git_version)'
should be different from what you posted since this is a different package version

~/tensorflow_src $ git rev-parse 0.6.0-76902-gd855adfc5a
should be different from what you posted since this is a different package version
```

Now that you have a working tflite_runtime package, you'll need a libedgetpu.so. That's why you'll need to build it from our libedgetpu repo. Just make sure that when you build, you are linking against the same tensorflow version by replacing that line with the commit you got from that above git-revparse command. Then build like this:
```
$ DOCKER_CPUS="armv6" DOCKER_TARGETS=libedgetpu make docker-build
```

A couple of `libedgetpu.so*` binaries will be in the out directory. You'll need to install them to your library paths and that should be it as far as libraries installation.

One last step, which may not be necessary but just for assurance. Sometimes, linux doesn't allows user to have plugdev group by default, and this is necessary for interacting with the edgetpu. So I'd also do this:
```
$ sudo usermod -aG plugdev $USER
$ sudo reboot now
```

That is all!

prettyflyforabeeguy, 2020-12-15:
@Namburger you are my new best friend whom I've never met!!! 
Somehow I missed the link to the .whl file that rhadnum had shared.  Thank you thank you thank you for taking the time to write up such a wonderful explanation but in the end I didn't need to do any of that.   I simply downloaded the [tflite_runtime-2.3.1-cp37-cp37m-linux_armv6l.whl](https://drive.google.com/file/d/1mW8QGmhfk4kRYXqUTEH1ckPUNIsl7S3d/view?usp=sharing) and installed it and everything works.  This is so awesome!

Namburger, 2020-12-15:
Nicee and no probs :)