google-coral/pycoral #8: Is there a special reason for armv6 to be ignored?

ricardodeazambuja, 2020-12-26:
The Coral TPU USB version would be a lovely alternative to add inference powers to a RPI 0, but since the last release from https://github.com/google-coral/edgetpu-platforms it seems to only support Python 3.5.

I would like to try to make things work with the RPI 0, but before I spend time on this I would be really glad to know whether or not it's feasible ;)

Namburger, 2020-12-26:
I guess you already found this? https://github.com/google-coral/pycoral/issues/7

ricardodeazambuja, 2020-12-27:
Hi @Namburger, yes I did find it. However, I'm still lost. I can see there's a version of the runtime available:
$ sudo apt install python3-tflite-runtime
It installs the 2.5.0 and I can import it (python3):
```
>>> import tflite_runtime
>>> tflite_runtime.__git_version__
'48c3bae94a8b324525b45f157d638dfd4e8c3be1'
```
However, every example I tried gives me "Illegal instruction".

Actually, there are a lot of things available: 
$ sudo apt list | grep edgetpu
edgetpu-examples/coral-edgetpu-stable 15.0 all
gasket-dkms/coral-edgetpu-stable 1.0-14 all
libedgetpu-dev/coral-edgetpu-stable 15.0 armhf
libedgetpu-legacy-dev/coral-edgetpu-stable 15.0 armhf
libedgetpu1-legacy-max/coral-edgetpu-stable 15.0 armhf
libedgetpu1-legacy-std/coral-edgetpu-stable 15.0 armhf
libedgetpu1-max/coral-edgetpu-stable 15.0 armhf
libedgetpu1-std/coral-edgetpu-stable,now 15.0 armhf
pycoral-examples/coral-edgetpu-stable 1.0 all
python3-edgetpu/coral-edgetpu-stable 15.0 armhf
python3-pycoral/coral-edgetpu-stable,now 1.0 armhf
python3-tflite-runtime/coral-edgetpu-stable,now 2.5.0 armhf [installed]
I tried to install everything, but the system gives error messages about incompatibilities.

Installing using the whl from https://github.com/prettyflyforabeeguy/tf_lite_on_pi_zero allows me to run the example that comes with the repo, but I couldn't run any examples from https://github.com/google-coral/tflite.

This is my system (RPI 0 W):
Linux 5.4.79+ armv6l GNU/Linux
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

I have already accumulated lots of frustration with my [AIY Vision Kit](https://github.com/thecognifly/AIY-Vision-Kit-Utils) (although it works on the RPI0 while the TPU USB...), but I used to blame Intel... now I'm starting to think it was not Intel's fault after all. It would be so nice to have a low power device like a RPI0+TPU USB...

ricardodeazambuja, 2020-12-27:
I just tried to remove the python3-tflite-runtime and install python3-pycoral. It installed python3-tflite-runtime again (lol) and that made me feel like it should work (after all, the maintainers would not push packages that ask for the wrong dependencies...). Then I cloned https://github.com/google-coral/pycoral and tried the examples, but I always get the "Illegal instruction".
```
python3 examples/classify_image.py   --model test_data/mobilenet_v2_1.0_224_inat_bird_quant.tflite    --labels test_data/inat_bird_labels.txt   --input test_data/parrot.jpg
```
and 
```
python3 examples/classify_image.py   --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite    --labels test_data/inat_bird_labels.txt   --input test_data/parrot.jpg
```
both fail with "Illegal instruction"

ricardodeazambuja, 2020-12-28:
Ok, I posted a comment directly on [#229](https://github.com/google-coral/edgetpu/issues/229#issuecomment-751504398) where I fully describe my attempt to generate my own pip packed for the RPI 0.

What would be the advantage of adding this [extra / optional layer (pycoral)](https://coral.ai/docs/edgetpu/inference/) to the process? Will google give up the tflite_runtime way in the near future?

Namburger, 2020-12-28:
@ricardodeazambuja use this package instead, someone has already built the tfltie_package before and proved to be working: https://github.com/google-coral/edgetpu/issues/229#issuecomment-708478704

We won't be able to change the tflite_runtime package builds since it is owned by the tensorflow team and changes there can break compatibility to newer OS. I suggest using that package and ust build your own libedgetpu.so which a much simpler approach.

ricardodeazambuja, 2020-12-28:
Thanks @Namburger for your reply, but if someone managed to build the pip package for the RPI0 that confirms it's possible, so I want to know the "secret sauce" to make sure I will be able to build it in the future myself :)

I'm planning to use the Coral TPU USB for my research project (up to 3 years!) and I can't depend on something downloaded from a random gdrive link.

Namburger, 2020-12-28:
@ricardodeazambuja fair enough :)

scottamain, 2020-12-28:
Hey @ricardodeazambuja, the pycoral library is built on top of (it depends on) the tflite_runtime library, so we're not going to give up on the Edge TPU support for those TF Lite APIs. The pycoral advantage is some convenience APIs for inferencing, additional APIs for on-device transfer learning, and pipelining with multiple Edge TPUs. It's not required.

Also, you should find everything you need to know about building the tflite_runtime wheel and the Edge TPU runtime for any other platform here: https://coral.ai/docs/notes/build-coral/

Namburger, 2020-12-28:
@scottamain the problem here is that the tflite_runtime package isn't built for armv6 any longer, this hasn't ben working for a while now, so @ricardodeazambuja needs to rebuild that wheel for his specific need. 
He actually done it and post progress here :)
https://github.com/google-coral/edgetpu/issues/229#issuecomment-751504398

closed 2020-12-28 by Namburger

scottamain, 2020-12-28:
Yeah, and building tflite_runtime is covered in the doc I shared. 
(We never claimed we would support armv6; that one-time build was never official.)

ricardodeazambuja, 2020-12-28:
By just following the instructions [here](https://coral.ai/docs/notes/build-coral/#build-the-tf-lite-runtime-tflite_runtime), it's not possible to build it for the RPI 0. In the best scenario, it doesn't link latomic...

ricardodeazambuja, 2020-12-29:
Ok, I forked the repos, made some modifications and now I am able to generate the tflite_runtime (compiled directly on the RPI0) and the libedgetpu :)
https://github.com/ricardodeazambuja/libedgetpu/releases/tag/rpi0_tflite_edgetpu