google-coral/pycoral #10: Installation issue on x64

bpinaya, 2021-01-12:
Hi there, I've followed your building guide, and while installing the wheels built from source I had the following error.
```
The directory '/home/bpinaya/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/bpinaya/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Processing ./dist/pycoral-1.0.0-cp36-cp36m-linux_x86_64.whl
Collecting tflite-runtime==2.5.0 (from pycoral==1.0.0)
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 342, in run
    requirement_set.prepare_files(finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python3/dist-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 795, in get_page
    resp.raise_for_status()
  File "/usr/share/python-wheels/requests-2.18.4-py2.py3-none-any.whl/requests/models.py", line 935, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://pypi.org/simple/tflite-runtime/
```
Same thing happens if I try to install from any of the release wheels for x64. Wonder if you have any ideas.

Namburger, 2021-01-14:
Hello, could you point me to the build from source guide that you are following?

bpinaya, 2021-01-14:
The ones in your [readme](https://github.com/google-coral/pycoral/blob/master/README.md).

```
scripts/build.sh
```
Then I want to install the wheels. Installing the `tflite_runtime-2.5.0-cp36-cp36m-linux_x86_64.whl` works ok.
but when installing `pip3 install pycoral-1.0.0-cp36-cp36m-linux_x86_64.whl` I get the error mentioned above.

My python version is `3.6.9`. Since then I tried building in a conda environment with python `3.6.10` and it work properly, the wheel installed and I was able to run the segmentation example. Very weird behavior, I'll try to build on an conda env of that specific version and update the post.

Namburger, 2021-01-14:
@bpinaya I see, that's actually really odd, look forward to hear more

bpinaya, 2021-01-14:
Well I'm closing this since I already have it running inside conda. I also manage to install it outside conda by just using `sudo pip3 install ...` which is of course [not a recommended practice](https://askubuntu.com/questions/802544/is-sudo-pip-install-still-a-broken-practice).
Tried in a conda env with python `3.6.9` and it works too.

I also tried using your release versions `https://github.com/google-coral/pycoral/releases/download/release-frogfish/pycoral-1.0.0-cp36-cp36m-linux_x86_64.whl` and that one fails outside conda without sudo.

Would be interesting to know what caused it outside conda without sudo but I just want to benchmark coral so can't spend a lot of time in this, inside a conva env it works just fine. Closing

closed 2021-01-14 by bpinaya

Namburger, 2021-01-14:
@bpinaya 
outside of conda without sudo is usually normal since th OS would block you from installing it for other user. You can install it like this:
```
python3 -m pip install package --user
```

But I'm still not sure about your other errors :/

bpinaya, 2021-01-14:
Tried outside conda without sudo like:
```
python3 -m pip install dist/pycoral-1.0.0-cp36-cp36m-linux_x86_64.whl --user
```
and still nothing:
```
The directory '/home/bpinaya/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/bpinaya/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Processing ./dist/pycoral-1.0.0-cp36-cp36m-linux_x86_64.whl
Collecting Pillow>=4.0.0 (from pycoral==1.0.0)
  Downloading https://files.pythonhosted.org/packages/b6/c0/442d9d87e0da00bf856ef6dd4916f84a2d710b5f1a367d42d7f3c4e99a6c/Pillow-8.1.0-cp36-cp36m-manylinux1_x86_64.whl (2.2MB)
    100% |████████████████████████████████| 2.2MB 683kB/s 
Collecting tflite-runtime==2.5.0 (from pycoral==1.0.0)
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 342, in run
    requirement_set.prepare_files(finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python3/dist-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 795, in get_page
    resp.raise_for_status()
  File "/usr/share/python-wheels/requests-2.18.4-py2.py3-none-any.whl/requests/models.py", line 935, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://pypi.org/simple/tflite-runtime/
```
It's fine, as long as I can run it on conda I'm ok, thanks for the interest!