google-coral/pycoral #14: Image with 640x480 resolution breaks the Example in Detect_Image.py

mitchross, 2021-02-07:
python3 examples/detect_image.py   --model test_data/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite   --labels test_data/coco_labels.txt   --input test_data/bear480.png   --output ${HOME}/bear480_processed.png
Traceback (most recent call last):
  File "examples/detect_image.py", line 112, in <module>
    main()
  File "examples/detect_image.py", line 79, in main
    _, scale = common.set_resized_input(
  File "/usr/lib/python3/dist-packages/pycoral/adapters/common.py", line 99, in set_resized_input
    tensor[:h, :w] = np.reshape(result, (h, w, channel))
  File "<__array_function__ internals>", line 5, in reshape
  File "/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py", line 301, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py", line 58, in _wrapfunc
    return _wrapit(obj, method, *args, **kwds)
  File "/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py", line 47, in _wrapit
    result = getattr(asarray(obj), method)(*args, **kwds)
ValueError: cannot reshape array of size 270000 into shape (225,300,3)

[image]

Namburger, 2021-02-08:
@mitchross that's odd that it is trying to reshape it to (225, 300, 3) when the model is (300, 300, 3). If the picture you posted the original picture with full size (I see the picture here is only 397x297)? If not, could you attach it here?

mitchross, 2021-02-08:
@Namburger I've zipped it up. Lets try that. should be 640x480.
[bear480.zip](https://github.com/google-coral/pycoral/files/5944152/bear480.zip)

Namburger, 2021-02-09:
Thanks for reporting, looks like it is a reproducible bug

closed 2021-05-13 by Naveen-Dodda

ghbn8, 2022-10-17:
How was this resolved? This appears to still be an issue.

hjonnala, 2022-10-18:
@ghbn8 Please add the below line [here](https://github.com/google-coral/pycoral/blob/master/examples/detect_image.py#L77) to run the demo with the image mentioned at: https://github.com/google-coral/pycoral/issues/14#issuecomment-775176844. Thanks!

image = image.convert('RGB')