google-coral/pycoral #15: Jetson Nano + Coral USB - RuntimeError: Internal: Unsupported data type in custom op handler: 0Node number 1 (EdgeTpuDelegateForCustomOp) failed to prepare.

mitchross, 2021-02-09:
```
:~/coral/pycoral$ dpkg -l | grep libedgetpu                                                                                            
ii  libedgetpu1-max:arm64                      15.0                                             arm64        Support library for Edge TPU
```

```
 python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 84, in <module>
    main()
  File "examples/classify_image.py", line 62, in main
    interpreter.allocate_tensors()
  File "/home/vanillax/.local/lib/python3.6/site-packages/tflite_runtime/interpreter.py", line 242, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/home/vanillax/.local/lib/python3.6/site-packages/tflite_runtime/interpreter_wrapper.py", line 115, in AllocateTensors
    return _interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Internal: Unsupported data type in custom op handler: 0Node number 1 (EdgeTpuDelegateForCustomOp) failed to prepare.
```

Namburger, 2021-02-09:
@mitchross can you share this output?
```
python3 -c 'print(__import__("tflite_runtime").__version__)'
```

mitchross, 2021-02-09:
@Namburger 

`$ python3 -c 'print(__import__("tflite_runtime").__version__)'
2.1.0.post1
`

Namburger, 2021-02-09:
@mitchross that's an older version of tflite_rutime, please upgrade to this one and it should works!
```
pip3 install --extra-index-url https://google-coral.github.io/py-repo/ tflite_runtime
```

mitchross, 2021-02-09:
@Namburger ```
 pip3 install --extra-index-url https://google-coral.github.io/py-repo/ tflite_runtime
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://google-coral.github.io/py-repo/
Requirement already satisfied: tflite_runtime in /home/vanillax/.local/lib/python3.6/site-packages (2.1.0.post1)
Requirement already satisfied: numpy>=1.12.1 in /usr/local/lib/python3.6/dist-packages (from tflite_runtime) (1.16.1)
WARNING: You are using pip version 20.2.4; however, version 21.0.1 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
vanillax@vanillax-jetson:~/coral/pycoral$

```

Namburger, 2021-02-09:
Ahh, an issue similar to pi as described here:
https://www.tensorflow.org/lite/guide/python#install_tensorflow_lite_for_python

You can just install it from these released packages, pick the one that matches your platform: 
 https://github.com/google-coral/pycoral/releases/

mitchross, 2021-02-10:
Working!

Namburger, 2021-02-10:
Nice, I've also put in a bug for your other issues, I'll give you update when it's fixed

closed 2021-02-10 by Namburger