google-coral/pycoral #17: Right whl package for Raspberry PI 4 64 bit OS beta, Kernel 5.10

hackebike, 2021-02-16:
I tried to install the 
pip3 install https://github.com/google-coral/pycoral/releases/download/release-frogfish/tflite_runtime-2.5.0-cp37-cp37m-linux_armv7l.whl

But I am getting an error:
"tflite_runtime-2.5.0-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform."

Is there a whl package that I can install my the RPI4 64 bit OS beta?
thx

Namburger, 2021-02-16:
@hackebike if it's a 64 bits OS, can you try the aarch64 wheel instead?
https://github.com/google-coral/pycoral/releases/

hackebike, 2021-02-16:
I was using a script from and got errors:
https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/97

Is this the right whl for Raspberry PI 64 bit OS beta correct?
https://github.com/google-coral/pycoral/releases/download/v1.0.1/pycoral-1.0.1-cp38-cp38-linux_aarch64.whl

I can update the script

hackebike, 2021-02-16:
got this:
"pycoral-1.0.1-cp38-cp38-linux_aarch64.whl is not a supported wheel on this platform."

Namburger, 2021-02-16:
@hackebike and which python version do you have?
You tried a python3.7 for the 32bits wheel but a python3.8 for the 64bits wheel lol

hackebike, 2021-02-16:
I have python3.7 installed and I am on 64 bit.
Sorry I am confused with the naming,
Which is the correct whl file I need to install?

hackebike, 2021-02-16:
ok I got installed !

pip3 install https://github.com/google-coral/pycoral/releases/download/v1.0.1/tflite_runtime-2.5.0-cp37-cp37m-linux_aarch64.whl

I was confusing the python naming in the file name! Sorry!
Well at least I can report to the other project the update they need to do for Python3.7 and RPI 64 bit OS
thx

Namburger, 2021-02-16:
@hackebike nice!

closed 2021-02-16 by Namburger