google-coral/pycoral #18: ImportError: DLL load failed while importing _pywrap_coral: The specified module could not be found.

arun-kumark, 2021-02-19:
Hello,

I am trying M.2 card on Mini PCIe bridge with Windows 10 OS.

While trying to test the Google Coral card for the example program, I am facing the folloiwng error:
https://coral.ai/docs/m2/get-started/#requirements

```
C:\Users\227G\pycoral>python examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 39, in <module>
    from pycoral.utils.edgetpu import make_interpreter
  File "C:\Users\227G\AppData\Local\Programs\Python\Python38\lib\site-packages\pycoral\utils\edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: DLL load failed while importing _pywrap_coral: The specified module could not be found.
```

The device manager is able to detect the device as Google Coral device correctly.
My system comfiguration:
Intel Elkhart Lake 
RAM 16GB

I couldn't find the similar or related issue, could you help me in resolving it?

Kind Regards,
Arun

Namburger, 2021-02-26:
@arun-kumark this seems like you some how have a missing dll, which I'm not sure why. Could you check what's under this path in your machine?
```
C:\Users\227G\AppData\Local\Programs\Python\Python38\lib\site-packages\pycoral\
```

manoj7410, 2021-03-12:
@arun-kumark Is this issue still reproducible ?

arun-kumark, 2021-03-12:
Hi Manoj,
Unfortunately, I moved from the Widnows OS to Debian. I cannot further check this, so would like to close. 
Kind Regards
Arun

closed 2021-03-12 by arun-kumark

arun-kumark, 2024-01-09:
Hi Manoj,
I am again testing this issue long after.

I find the folliwing files in the path you mentioned:

IPC@DESKTOP-2771BIE MINGW64 ~/AppData/Local/Programs/Python/Python38/Lib/site-packages/pycoral
$ ls -l
total 5
-rw-r--r-- 1 IPC 197121 683 Jan  9 08:48 __init__.py
drwxr-xr-x 1 IPC 197121   0 Jan  9 08:48 __pycache__/
drwxr-xr-x 1 IPC 197121   0 Jan  9 08:48 adapters/
drwxr-xr-x 1 IPC 197121   0 Jan  9 08:48 learn/
drwxr-xr-x 1 IPC 197121   0 Jan  9 08:48 pipeline/
drwxr-xr-x 1 IPC 197121   0 Jan  9 08:48 pybind/
drwxr-xr-x 1 IPC 197121   0 Jan  9 08:48 utils/

Could you help further

thanks