google-coral/pycoral #19: Error running face detection demo

Tylersuard, 2021-02-24:
Input:

 edgetpu_detect_server \
> --source /dev/video1:YUY2:800x600:24/1  \
> --model ${DEMO_FILES}/ssd_mobilenet_v2_face_quant_postprocess_edgetpu.tflite

Error:
INFO:OpenGL.acceleratesupport:No OpenGL_accelerate module loaded: No module named 'OpenGL_accelerate'
Error: gst-stream-error-quark: Internal data stream error. (1): gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
^CINFO:edgetpuvision.streaming.server:Server is shutting down
INFO:edgetpuvision.streaming.server:Camera stop recording

Using a webcam for inference

Namburger, 2021-02-24:
@Tylersuard where are you running this from?

Tylersuard, 2021-02-26:
I found the problem: my USB camera runs at 20fps, not 24 fps.
I fixed it by changing `YUY2:800x600:24/1`
to `YUY2:800x600:20/1`

closed 2021-02-26 by Tylersuard