google-coral/pycoral #24: No module named 'pycoral.pybind._pywrap_coral' (aarch64, debian bullseye, python 3.9)

leoniloris, 2021-03-31:
Hello folks!
So, I have the following system
```
debian bullseye
python 3.9
architecture aarch64 (arm64 or armv8)
```

after some time struggling, I compiled manually and generated the both tflite and pycoral for my board, installed them both and I'm trying to follow https://coral.ai/docs/m2/get-started/#4-run-a-model-on-the-edge-tpu
But I'm facing the following problem
```ModuleNotFoundError: No module named 'pycoral.pybind._pywrap_coral'```

I have no Idea what I did wrong. do you guys know?

paullee2018, 2021-04-14:
I have something similar using python 3.7.

`Traceback (most recent call last):
  File "test_coral.py", line 8, in <module>
    from pycoral.utils import edgetpu
  File "/home/pi/projects/coral/pycoral/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ModuleNotFoundError: No module named 'pycoral.pybind'
`

abelhouchat, 2021-05-01:
I ran into a similar issue with pycoral. I believe it's because classify_image.py is confusing the pycoral package with the pycoral folder. I was able to resolve it by renaming the pycoral folder to something else.

Naveen-Dodda, 2021-05-10:
@abelhouchat May I know on which platform you have this issue. Can you mention how you have installed[ python3-pycoral ?](https://coral.ai/software/#pycoral-api)

abelhouchat, 2021-05-10:
@Naveen-Dodda This is on Ubuntu 18.04. I installed pycoral as recommended in the tutorial, using `sudo apt-get install python3-pycoral`

Naveen-Dodda, 2021-05-13:
Can you try a pip install instead 
https://coral.ai/software/#pycoral-api

hjonnala, 2021-07-12:
Hello, does anyone here still facing the issue. If yes, please let me know the python version and operating system have been used.

maazullah96, 2021-07-13:
Solved the issue by installing through the whl file 
 
[image]

hjonnala, 2021-07-13:
@leoniloris are you able to fix the issue.

mprhode, 2021-07-14:
I'm still having the issue with Ubuntu 18.04 and python 3.8.8 (apt install -> ModuleNotFound, pip3 install -> https://pypi.org/simple/python3-coral/)

maazullah96, 2021-07-15:
@mprhode  Did you have tried the installation from whl file?

japm48, 2021-07-16:
I had the exact same problem in Windows, and after hours of guessing/debugging I realised that I forgot to install the Edge TPU runtime (step 1 in [the instructions](https://coral.ai/docs/accelerator/get-started)...).
Note that, at least in my case, the "No module named" error is misleading: the pyd module is actually found but it fails to load because it cannot find some DLL/.so dependency (my candidates are libusb, UsbDk or libedgetpu). This can be debugged in Windows by loading with `ctypes.WinDLL` (possibly something similar with Linux).

Possibly related to #32.

hjonnala, 2021-07-19:
The issue(No module named 'pycoral.pybind._pywrap_coral') can be resolved using the whl files from this source : https://github.com/google-coral/pycoral/releases for python vesrions 3.5 - 3.8.

closed 2021-07-19 by hjonnala

minhhoai1001, 2021-09-20:
> I ran into a similar issue with pycoral. I believe it's because classify_image.py is confusing the pycoral package with the pycoral folder. I was able to resolve it by renaming the pycoral folder to something else.

Thank you. I work for me.

kzgrey, 2022-09-18:
> ModuleNotFoundError: No module named 'pycoral.pybind._pywrap_coral'

If you get this error, its because there's a directory in the repo called pycoral.  If you `mv pycoral pycoral2` everything starts working.  No need to downgrade Python.

bme323, 2023-03-06:
Hi, I am having a similar problem. 

When I enter step 3 from https://coral.ai/docs/accelerator/get-started/#3-run-a-model-on-the-edge-tpu  I get error code:
Traceback (most recent call last):
  File "examples/classify_image.py", line 37, in <module> from pycoral.adapters import classify 
ModuleNotFoundError: No module named 'pycoral'

I have downgraded to python 3.7.12 from 3.9.2 (RPi 4), running on arm7l 32-bit. 
I have also tried installing the whl file and tried renaming the pycoral folder name to pycoral2 as suggested.

All have had no success. Any help on this?

Nick-P-Orr, 2024-01-31:
> I'm having the same issue. I can run the test on the host machine but when I run it in docker i get this same error. Did you get it working?

I'm having this issue as well. Any update?