google-coral/pycoral #26: Can't build pycoral, includes not found

Spanni26, 2021-05-11:
Hello,

I try to build pycoral on opensuse. I cloned the git, but at a "make CPU=k8 build" it failed:
 src/coral_wrapper.cc:15:10: fatal error: numpy/arrayobject.h: No such file or directory
 #```
include <numpy/arrayobject.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
src/main/tools/linux-sandbox-pid1.cc:437: waitpid returned 2
src/main/tools/linux-sandbox-pid1.cc:457: child exited with code 1
src/main/tools/linux-sandbox.cc:204: child exited normally with exitcode 1
Target //src:_pywrap_coral failed to build
```
The numpy-devel-package is installed, the file are located at /usr/lib64/python3.6/site-packages/numpy/core/include/numpy/arrayobject.h

How can I tell the build process where to find the numpy includes?

Regards

Daniel

Naveen-Dodda, 2021-05-13:
@Spanni26 

Not exactly sure how to reproduce this issue on different x86_64 platform. But these are docker builds things should fairly straight forward if you use docker.

Can you follow steps below and see if you still have similar issue. 

cd pycoral
make docker-shell 
make CPU=k8  (in the docker shell)

Spanni26, 2021-05-26:
Hello,

I can't use docker cause I build on a build server (open build service), which has no connection to the internet.
I downloaded all the dependencies this package needs, so it doesn't claim any missing packages.

There is another error about wrong include path also. I will open another issue about that.

manoj7410, 2021-05-28:
@Spanni26 Did you try to include /usr/lib64/python3.6/site-packages/numpy/core/include/numpy/ in your PATH environment ?

Spanni26, 2021-05-28:
Which "PATH" do you mean? ISt the a variable for python includes?

manoj7410, 2021-05-28:
> Which "PATH" do you mean?

In your system's PATH environment and in your build's environment. 

Although the recommended way to build is still Docker as it will take care of all the deps and includes.

Spanni26, 2021-06-09:
I can't use docker cause I'm in an environment without internet access.
"set |grep include" finds nothing, so no include-path is set in my environment.