google-coral/pycoral #28: Coral Mini Pycoral example error

SunBeenMoon, 2021-05-14:
Hardware : Google Coral Board Mini
Host Computer : RPI 4 

I'm just testing the Coral board.
An error occurred at the beginning part as shown in the picture below.
[image]

I got the error code 
`mendel@undefined-orange:~/coral/pycoral$ python3 examples/classify_image.py \
> --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
> --labels test_data/inat_bird_labels.txt \
> --input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 84, in <module>
    main()
  File "examples/classify_image.py", line 62, in main
    interpreter.allocate_tensors()
  File "/home/mendel/.local/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 242, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/home/mendel/.local/lib/python3.7/site-packages/tflite_runtime/interpreter_wrapper.py", line 115, in AllocateTensors
    return _interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Internal: Unsupported data type in custom op handler: 0Node number 1 (EdgeTpuDelegateForCustomOp) failed to prepare.`

Errors continue to occur in the same interpreter part.

I need help.

Regards,
SunBeenMoon

manoj7410, 2021-05-18:
Just for the record. This got resolved by reflashing the board.

SunBeenMoon, 2021-05-18:
Thanks to reply @manoj7410 

I solved this problem with reflashing the board with option -H
bash flash.sh -H

closed 2021-05-18 by SunBeenMoon

ergoego, 2021-09-24:
This exact same thing is happening for me as well, but on a raspberry pi 4. Is there a way to flash he USB Accelerator or do I need to flash my raspberry pi and start from scratch?