google-coral/pycoral #30: Pycoral with GRPC error

bomerzz, 2021-06-29:
While utilizing the GRPC python library on a raspberry pi, once the `import grpc` line of code is run, the load delegate function will fail to connect to the USB Coral accelerator.  Can someone please advise what steps I can take?

Thank you.

python3-pycoral version: 1.0.1

Using load delegate:
[image]
Using a detection engine:
[image]

manoj7410, 2021-07-01:
@bomerzz I see that you are working with virtual environment. Is the USB Accelerator visible in your environment. 
Can you share the output of 'lsusb' ?

bomerzz, 2021-07-07:
[image]

Hi @manoj7410  above is the lsusb output

manoj7410, 2021-07-07:
@bomerzz lsusb output looks okay. Are you able to run the demo given at : https://coral.ai/docs/accelerator/get-started/#3-run-a-model-on-the-edge-tpu ?

bomerzz, 2021-07-07:
Yes i am able to however if i add `import grpc` to the demos it will not run as well

manoj7410, 2021-07-07:
Okay.  Can you paste the detailed error here ?

closed 2021-10-18 by hjonnala