google-coral/pycoral #31: examples fail to run on coral Mini PCIe version - Could not set performance expectation

pejotigrek, 2021-07-01:
my system:

- HP t620 Quad Core
- AMD GX-415GA SOC with Radeon(tm) HD Graphics
- 8GB RAM
- Coral Device: Mini PCIe

```
> uname -m -r
5.12.13-051213-generic x86_64

> lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal
```

I followed the guide from [coral.ai/docs/m2/get-started](https://coral.ai/docs/m2/get-started/), including the need of blacklisting  prebuilt/preinstalled `apex` and `gasket` to install provided driver and Edge TPU runtime:
```
gasket-dkms: newest version (1.0-16).
libedgetpu1-std: newest version (15.0).
```

according to driver and module verification - everything is ok:
```
> lspci -nn | grep 089a
01:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]

> ls /dev/apex_0
/dev/apex_0
```

moved on to installing the PyCoral library - succeeded.
moved on to run examples from - failed with below report:

```
> python3 examples/classify_image.py \
>> --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
>> --labels test_data/inat_bird_labels.txt \
>> --input test_data/parrot.jpg

W :131] Could not set performance expectation : 4 (Inappropriate ioctl for device)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 152, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 111, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 84, in <module>
    main()
  File "examples/classify_image.py", line 61, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 66, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 42, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 154, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1
```

I'm new to Coral Edge TPU & somewhere between _beginner_ and _intermediate_ in linux [far, far away from advanced user ;)] - wanted to use Coral TPU as accelerator for object detection in my smarthome system [[HomeAssistant](https://www.home-assistant.io/) with [Frigate NVR](https://github.com/blakeblackshear/frigate) installed]. It looks like it is working - I see no issues/error reports within Frigate, and it's log looks fine, with a small exception - at startup I can see similar error like during the example scripts (check the 3rd line below):
```
detector.coral_pci             INFO    : Starting detection process: 39
frigate.edgetpu                INFO    : Attempting to load TPU as pci
W :131] Could not set performance expectation : 52 (Inappropriate ioctl for device)
frigate.edgetpu                INFO    : TPU found
```

is it something I:
- can fix easily?
- shouldn't be bothered with, it's fine.

manoj7410, 2021-07-02:
Can you paste the output of:
-modinfo gasket
- modinfo apex
- lsmod | grep apex

pejotigrek, 2021-07-04:
here you are:

```
> modinfo gasket
filename:       /lib/modules/5.12.13-051213-generic/kernel/drivers/staging/gasket/gasket.ko
author:         Rob Springer <rspringer@google.com>
license:        GPL v2
version:        1.1.2
description:    Google Gasket driver framework
srcversion:     E96C15597C5747EE336FF6D
depends:
staging:        Y
retpoline:      Y
intree:         Y
name:           gasket
vermagic:       5.12.13-051213-generic SMP mod_unload modversions
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        2E:8A:06:6E:C4:C2:88:BB:05:78:A1:8D:B3:8B:04:74:E9:8A:77:F0
sig_hashalgo:   sha512
signature:      90:5E:EB:12:3B:23:DE:7B:E7:69:9B:A6:2C:E3:58:81:CC:3E:1F:A2:
                0B:3D:CB:21:2B:7D:D5:B7:4F:AF:BD:B4:DB:7F:9D:9D:73:A1:2E:07:
                F3:74:2B:B9:9D:10:1F:48:24:55:82:3A:BE:B2:EC:ED:8C:D8:14:7C:
                86:90:1C:A6:7D:7F:DF:D8:FE:F9:23:27:28:72:E6:3F:55:A3:DD:C8:
                DD:C2:BA:96:4C:DC:AE:0B:B9:82:14:69:A8:96:95:B7:9D:48:7B:96:
                B0:6B:A5:D4:37:EF:95:77:AF:BD:4E:96:5D:15:17:41:28:B3:7A:06:
                D3:3F:20:0B:C8:A3:0C:CD:CC:EE:18:E7:76:1A:CC:C4:73:FF:EC:C8:
                5A:DF:D8:7B:FB:D9:15:F4:59:1C:A8:4E:FB:9A:5D:EC:39:36:56:25:
                7C:4D:E4:58:77:EF:73:A6:B3:35:1A:2A:A7:E1:1E:4B:A4:E5:ED:37:
                1F:D7:4C:90:18:6B:E5:33:52:C9:8C:98:BD:94:86:6C:06:4A:A6:1D:
                FF:DE:D6:97:7A:B4:6E:D3:18:F9:CA:5C:85:1A:A8:97:0C:49:AA:09:
                A1:54:0D:9A:D5:41:BD:CE:96:59:F3:1E:46:D6:57:4D:B6:EF:CE:ED:
                80:58:6F:91:80:7E:E9:2F:28:30:69:C9:7E:E8:D6:8C:CE:DB:D0:AE:
                68:43:38:26:E6:70:E1:58:5A:FF:7B:76:0E:40:22:82:A3:AE:EA:F1:
                3E:05:A6:AB:81:16:B1:E0:6F:1E:B7:15:B4:7D:52:49:E5:2C:30:AB:
                D5:CC:7C:C4:71:16:46:F7:13:A8:BC:6C:98:92:4C:63:FC:24:4D:89:
                DF:24:52:98:92:59:07:AB:8F:BD:83:96:AB:5C:B0:0F:32:F5:1D:7C:
                AE:81:C3:0F:EF:26:BF:97:D0:14:EB:6B:69:A7:C0:82:D9:C7:6F:FA:
                78:4E:E6:54:62:92:FB:96:D4:65:39:7A:DF:52:47:4F:B0:0B:5A:C3:
                DA:F3:E6:33:66:25:A0:F3:24:72:0B:8D:FA:DF:27:8E:72:37:8D:D5:
                44:7E:90:4E:21:CA:3B:F5:30:B6:D9:15:85:BE:8B:BF:7E:DB:35:82:
                06:91:B2:64:9E:78:ED:DF:40:FF:8F:05:88:6C:73:6C:B5:38:CA:57:
                D5:00:12:D6:B9:BF:9F:C9:9C:ED:0F:59:E1:92:63:5F:C7:51:40:10:
                F3:67:05:1B:6A:89:7A:FB:18:C1:41:E5:8B:7B:3C:36:17:2E:1F:9A:
                84:20:CC:3D:B5:41:5C:53:1A:2C:CD:4F:AC:D0:46:13:6D:13:A8:97:
                72:71:36:55:8B:10:88:3B:29:EA:E7:87

> modinfo apex
filename:       /lib/modules/5.12.13-051213-generic/kernel/drivers/staging/gasket/apex.ko
author:         John Joseph <jnjoseph@google.com>
license:        GPL v2
version:        1.0
description:    Google Apex driver
srcversion:     A59D430D2A073385D2E7062
alias:          pci:v00001AC1d0000089Asv*sd*bc*sc*i*
depends:        gasket
staging:        Y
retpoline:      Y
intree:         Y
name:           apex
vermagic:       5.12.13-051213-generic SMP mod_unload modversions
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        2E:8A:06:6E:C4:C2:88:BB:05:78:A1:8D:B3:8B:04:74:E9:8A:77:F0
sig_hashalgo:   sha512
signature:      32:A9:6F:95:56:7C:15:CE:C4:47:BE:2E:4E:21:C4:89:8E:81:72:CF:
                8E:86:28:07:D0:A1:23:70:69:6F:F0:A2:0A:AD:04:96:D1:90:C8:11:
                6C:1D:FF:AC:6C:83:4B:9D:34:ED:47:4B:E5:5F:5A:F1:A5:8E:5C:64:
                9E:7C:F9:6B:8E:9C:0B:3B:B1:5F:C2:29:81:B2:0A:8D:66:E6:ED:78:
                50:77:47:DC:E5:54:7D:FB:FB:F1:3A:62:A8:04:62:4B:2F:BD:C9:48:
                0E:40:DA:3D:5B:C7:0C:A7:97:1F:EA:CB:14:63:74:86:38:03:CB:3C:
                7F:A6:8F:24:68:C9:94:5B:BD:49:32:0E:08:2A:51:6D:87:45:99:9D:
                0B:2F:90:DE:AE:28:B3:FA:F0:47:B8:0C:E3:E6:67:85:D0:03:8D:9C:
                A1:90:54:1C:00:C6:F7:4F:9E:AA:B2:3E:70:B9:6A:FF:3D:B8:46:CD:
                31:42:39:50:31:96:8C:5E:86:DE:5E:41:85:14:DC:C7:00:8D:D6:ED:
                49:C3:56:76:55:95:BB:13:74:8A:40:DA:02:4B:47:28:21:68:23:0A:
                1D:8E:4D:46:01:07:57:18:1D:E9:78:26:15:7A:86:A2:07:65:BC:D5:
                58:04:D4:EE:2B:3E:64:83:66:65:DC:0F:4B:01:2E:09:46:74:FB:B3:
                6D:64:C5:01:F5:D1:F6:43:BB:F8:60:BA:C3:13:28:73:D5:F2:17:24:
                04:6B:EC:03:76:AF:83:8A:EA:94:0F:6F:01:46:2D:47:B1:95:12:AD:
                CD:D7:28:52:0C:72:38:EC:BA:11:61:CF:76:D3:19:24:F7:DA:30:82:
                12:C1:92:EC:91:0D:FE:08:4A:2D:F8:25:BE:4E:16:5C:5B:5E:10:9E:
                8F:BF:9E:7E:6F:11:B3:61:D9:C4:33:39:11:B9:0D:DD:6F:F8:F8:2A:
                0C:5C:24:46:E3:74:98:57:4A:A0:8A:E1:28:D0:F1:B7:74:FE:6D:4A:
                D6:B5:DE:6A:9E:78:B8:35:A4:2B:18:1C:C9:51:8F:39:11:5F:15:6E:
                3C:F2:3F:10:D8:BF:36:4F:E1:7A:CC:8F:46:F2:FF:85:70:2B:4D:15:
                5C:93:16:32:32:7B:D7:F6:B2:3B:25:85:9D:7E:05:79:78:B2:55:18:
                FD:47:30:58:EA:E1:33:13:39:A8:C4:7A:98:4C:95:AD:F0:CE:23:C9:
                EB:EC:98:AB:53:01:F8:C4:C6:4E:ED:4E:0B:D0:25:F6:21:02:9A:A1:
                8A:E0:C6:82:33:6C:ED:8F:B2:C2:9C:39:84:05:64:89:AF:87:23:4B:
                13:F7:EB:6B:A1:5D:FB:91:47:66:32:3F
parm:           allow_power_save:int
parm:           allow_sw_clock_gating:int
parm:           allow_hw_clock_gating:int
parm:           bypass_top_level:int

> lsmod | grep apex
apex                   20480  5
gasket                106496  6 apex
```

I am running the demo straight in the system's shell. no docker involved here.

manoj7410, 2021-07-05:
@pejotigrek You have older Gasket and Apex modules installed on your machine. 
Did you follow the instructions given at https://coral.ai/docs/m2/get-started/#2a-on-linux ?

pejotigrek, 2021-07-05:
> @pejotigrek You have older Gasket and Apex modules installed on your machine.
> Did you follow the instructions given at https://coral.ai/docs/m2/get-started/#2a-on-linux ?

yes, I did followed the instructions - also with the workaround to disable Apex and Gasket by creating the `/etc/modprobe.d/blacklist-apex.conf` file with needed blacklists. then I even tried to do that by the second time [to be sure I did everything ok] and then [on the second time] `sudo apt-get install gasket-dkms libedgetpu1-std` told me that I already have newest drivers.

I can try this one more time, but to be clear please confirm if I'm doing it right to resolve:
1. use workaround with `/etc/modprobe.d/blacklist-apex.conf` + reboot
2. uninstall drivers with `udo apt-get remove gasket-dkms libedgetpu1-std` [to be sure it won't interfere]
3. reboot [just in case]
4. install drivers as mentioned in [coral.ai/docs/m2/get-started/#2a-on-linux](https://coral.ai/docs/m2/get-started/#2a-on-linux)
5. reboot
6. delete `/etc/modprobe.d/blacklist-apex.conf` and reboot for the last time

I guess there's too many reboots above, but I just want to be sure installation is clear - and if above steps are valid, I'll try this as soon as I'll be back at home [working right now].

pejotigrek, 2021-07-06:
@manoj7410 all right, I did as I wrote above - all the 6 steps - and seems like it's partially working [looks like the 2nd step was needed to properly install the drivers].

now, the command:

```
python3 examples/classify_image.py \
--model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
--labels test_data/inat_bird_labels.txt \
--input test_data/parrot.jpg
```
gives this output:
```
W :131] Could not set performance expectation : 4 (Inappropriate ioctl for device)
----INFERENCE TIME----
Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
17.0ms
7.1ms
7.0ms
7.1ms
7.1ms
-------RESULTS--------
Ara macao (Scarlet Macaw): 0.75781
```

shoud I be concerned about the `W :131] Could not set performance expectation : 4 (Inappropriate ioctl for device)` message?

FYI modinfo commands give very similar output..:
```
> modinfo gasket
filename:       /lib/modules/5.12.13-051213-generic/kernel/drivers/staging/gasket/gasket.ko
author:         Rob Springer <rspringer@google.com>
license:        GPL v2
version:        1.1.2
description:    Google Gasket driver framework
srcversion:     E96C15597C5747EE336FF6D
depends:
staging:        Y
retpoline:      Y
intree:         Y
name:           gasket
vermagic:       5.12.13-051213-generic SMP mod_unload modversions
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        2E:8A:06:6E:C4:C2:88:BB:05:78:A1:8D:B3:8B:04:74:E9:8A:77:F0
sig_hashalgo:   sha512
signature:      90:5E:EB:12:3B:23:DE:7B:E7:69:9B:A6:2C:E3:58:81:CC:3E:1F:A2:
                0B:3D:CB:21:2B:7D:D5:B7:4F:AF:BD:B4:DB:7F:9D:9D:73:A1:2E:07:
                F3:74:2B:B9:9D:10:1F:48:24:55:82:3A:BE:B2:EC:ED:8C:D8:14:7C:
                86:90:1C:A6:7D:7F:DF:D8:FE:F9:23:27:28:72:E6:3F:55:A3:DD:C8:
                DD:C2:BA:96:4C:DC:AE:0B:B9:82:14:69:A8:96:95:B7:9D:48:7B:96:
                B0:6B:A5:D4:37:EF:95:77:AF:BD:4E:96:5D:15:17:41:28:B3:7A:06:
                D3:3F:20:0B:C8:A3:0C:CD:CC:EE:18:E7:76:1A:CC:C4:73:FF:EC:C8:
                5A:DF:D8:7B:FB:D9:15:F4:59:1C:A8:4E:FB:9A:5D:EC:39:36:56:25:
                7C:4D:E4:58:77:EF:73:A6:B3:35:1A:2A:A7:E1:1E:4B:A4:E5:ED:37:
                1F:D7:4C:90:18:6B:E5:33:52:C9:8C:98:BD:94:86:6C:06:4A:A6:1D:
                FF:DE:D6:97:7A:B4:6E:D3:18:F9:CA:5C:85:1A:A8:97:0C:49:AA:09:
                A1:54:0D:9A:D5:41:BD:CE:96:59:F3:1E:46:D6:57:4D:B6:EF:CE:ED:
                80:58:6F:91:80:7E:E9:2F:28:30:69:C9:7E:E8:D6:8C:CE:DB:D0:AE:
                68:43:38:26:E6:70:E1:58:5A:FF:7B:76:0E:40:22:82:A3:AE:EA:F1:
                3E:05:A6:AB:81:16:B1:E0:6F:1E:B7:15:B4:7D:52:49:E5:2C:30:AB:
                D5:CC:7C:C4:71:16:46:F7:13:A8:BC:6C:98:92:4C:63:FC:24:4D:89:
                DF:24:52:98:92:59:07:AB:8F:BD:83:96:AB:5C:B0:0F:32:F5:1D:7C:
                AE:81:C3:0F:EF:26:BF:97:D0:14:EB:6B:69:A7:C0:82:D9:C7:6F:FA:
                78:4E:E6:54:62:92:FB:96:D4:65:39:7A:DF:52:47:4F:B0:0B:5A:C3:
                DA:F3:E6:33:66:25:A0:F3:24:72:0B:8D:FA:DF:27:8E:72:37:8D:D5:
                44:7E:90:4E:21:CA:3B:F5:30:B6:D9:15:85:BE:8B:BF:7E:DB:35:82:
                06:91:B2:64:9E:78:ED:DF:40:FF:8F:05:88:6C:73:6C:B5:38:CA:57:
                D5:00:12:D6:B9:BF:9F:C9:9C:ED:0F:59:E1:92:63:5F:C7:51:40:10:
                F3:67:05:1B:6A:89:7A:FB:18:C1:41:E5:8B:7B:3C:36:17:2E:1F:9A:
                84:20:CC:3D:B5:41:5C:53:1A:2C:CD:4F:AC:D0:46:13:6D:13:A8:97:
[... 27 lines cut ...]
                7F:A6:8F:24:68:C9:94:5B:BD:49:32:0E:08:2A:51:6D:87:45:99:9D:
                0B:2F:90:DE:AE:28:B3:FA:F0:47:B8:0C:E3:E6:67:85:D0:03:8D:9C:
                A1:90:54:1C:00:C6:F7:4F:9E:AA:B2:3E:70:B9:6A:FF:3D:B8:46:CD:
                31:42:39:50:31:96:8C:5E:86:DE:5E:41:85:14:DC:C7:00:8D:D6:ED:
                49:C3:56:76:55:95:BB:13:74:8A:40:DA:02:4B:47:28:21:68:23:0A:
                1D:8E:4D:46:01:07:57:18:1D:E9:78:26:15:7A:86:A2:07:65:BC:D5:
                58:04:D4:EE:2B:3E:64:83:66:65:DC:0F:4B:01:2E:09:46:74:FB:B3:
                6D:64:C5:01:F5:D1:F6:43:BB:F8:60:BA:C3:13:28:73:D5:F2:17:24:
                04:6B:EC:03:76:AF:83:8A:EA:94:0F:6F:01:46:2D:47:B1:95:12:AD:
                CD:D7:28:52:0C:72:38:EC:BA:11:61:CF:76:D3:19:24:F7:DA:30:82:
                12:C1:92:EC:91:0D:FE:08:4A:2D:F8:25:BE:4E:16:5C:5B:5E:10:9E:
                8F:BF:9E:7E:6F:11:B3:61:D9:C4:33:39:11:B9:0D:DD:6F:F8:F8:2A:
                0C:5C:24:46:E3:74:98:57:4A:A0:8A:E1:28:D0:F1:B7:74:FE:6D:4A:
                D6:B5:DE:6A:9E:78:B8:35:A4:2B:18:1C:C9:51:8F:39:11:5F:15:6E:
                3C:F2:3F:10:D8:BF:36:4F:E1:7A:CC:8F:46:F2:FF:85:70:2B:4D:15:
                5C:93:16:32:32:7B:D7:F6:B2:3B:25:85:9D:7E:05:79:78:B2:55:18:
                FD:47:30:58:EA:E1:33:13:39:A8:C4:7A:98:4C:95:AD:F0:CE:23:C9:
                EB:EC:98:AB:53:01:F8:C4:C6:4E:ED:4E:0B:D0:25:F6:21:02:9A:A1:
                8A:E0:C6:82:33:6C:ED:8F:B2:C2:9C:39:84:05:64:89:AF:87:23:4B:
                13:F7:EB:6B:A1:5D:FB:91:47:66:32:3F
parm:           allow_power_save:int
parm:           allow_sw_clock_gating:int
parm:           allow_hw_clock_gating:int
parm:           bypass_top_level:int
```

pejotigrek, 2021-07-06:
..well, that was too soon I guess - second try goes back to the errors:

```
W :131] Could not set performance expectation : 4 (Inappropriate ioctl for device)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 152, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 111, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 84, in <module>
    main()
  File "examples/classify_image.py", line 61, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 66, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 42, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 154, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1
```

manoj7410, 2021-07-07:
> @manoj7410 all right, I did as I wrote above - all the 6 steps - and seems like it's partially working [looks like the 2nd step was needed to properly install the drivers].
> 
> installation feedback:
> 
> ```
> > sudo apt-get install gasket-dkms libedgetpu1-std
> 
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> The following packages were automatically installed and are no longer required:
>   libgfortran5 libimagequant0 liblapack3 liblcms2-2 libwebpdemux2 python3-numpy python3-olefile python3-pil python3-tflite-runtime
> Use 'sudo apt autoremove' to remove them.
> The following NEW packages will be installed:
>   gasket-dkms libedgetpu1-std
> 0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
> Need to get 363 kB of archives.
> After this operation, 1,261 kB of additional disk space will be used.
> Get:1 https://packages.cloud.google.com/apt coral-edgetpu-stable/main amd64 gasket-dkms all 1.0-16 [47.8 kB]
> Get:2 https://packages.cloud.google.com/apt coral-edgetpu-stable/main amd64 libedgetpu1-std amd64 15.0 [315 kB]
> Fetched 363 kB in 0s (899 kB/s)
> Selecting previously unselected package gasket-dkms.
> (Reading database ... 149231 files and directories currently installed.)
> Preparing to unpack .../gasket-dkms_1.0-16_all.deb ...
> Unpacking gasket-dkms (1.0-16) ...
> Selecting previously unselected package libedgetpu1-std:amd64.
> Preparing to unpack .../libedgetpu1-std_15.0_amd64.deb ...
> Unpacking libedgetpu1-std:amd64 (15.0) ...
> Setting up libedgetpu1-std:amd64 (15.0) ...
> Setting up gasket-dkms (1.0-16) ...
> Loading new gasket-1.0 DKMS files...
> Building for 5.12.13-051213-generic
> Module build for kernel 5.12.13-051213-generic was skipped since the
> kernel headers for this kernel does not seem to be installed.
> Processing triggers for libc-bin (2.31-0ubuntu9.2) ...
> ```
> 
> now, the command:
> 
> ```
> python3 examples/classify_image.py \
> --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
> --labels test_data/inat_bird_labels.txt \
> --input test_data/parrot.jpg
> ```
> 
> gives this output:
> 
> ```
> W :131] Could not set performance expectation : 4 (Inappropriate ioctl for device)
> ----INFERENCE TIME----
> Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
> 17.0ms
> 7.1ms
> 7.0ms
> 7.1ms
> 7.1ms
> -------RESULTS--------
> Ara macao (Scarlet Macaw): 0.75781
> ```
> 
> shoud I be concerned about the `W :131] Could not set performance expectation : 4 (Inappropriate ioctl for device)` message?
> 
> FYI modinfo commands give very similar output..:
> 
> ```
> > modinfo gasket
> filename:       /lib/modules/5.12.13-051213-generic/kernel/drivers/staging/gasket/gasket.ko
> author:         Rob Springer <rspringer@google.com>
> license:        GPL v2
[... 72 lines cut ...]
>                 6D:64:C5:01:F5:D1:F6:43:BB:F8:60:BA:C3:13:28:73:D5:F2:17:24:
>                 04:6B:EC:03:76:AF:83:8A:EA:94:0F:6F:01:46:2D:47:B1:95:12:AD:
>                 CD:D7:28:52:0C:72:38:EC:BA:11:61:CF:76:D3:19:24:F7:DA:30:82:
>                 12:C1:92:EC:91:0D:FE:08:4A:2D:F8:25:BE:4E:16:5C:5B:5E:10:9E:
>                 8F:BF:9E:7E:6F:11:B3:61:D9:C4:33:39:11:B9:0D:DD:6F:F8:F8:2A:
>                 0C:5C:24:46:E3:74:98:57:4A:A0:8A:E1:28:D0:F1:B7:74:FE:6D:4A:
>                 D6:B5:DE:6A:9E:78:B8:35:A4:2B:18:1C:C9:51:8F:39:11:5F:15:6E:
>                 3C:F2:3F:10:D8:BF:36:4F:E1:7A:CC:8F:46:F2:FF:85:70:2B:4D:15:
>                 5C:93:16:32:32:7B:D7:F6:B2:3B:25:85:9D:7E:05:79:78:B2:55:18:
>                 FD:47:30:58:EA:E1:33:13:39:A8:C4:7A:98:4C:95:AD:F0:CE:23:C9:
>                 EB:EC:98:AB:53:01:F8:C4:C6:4E:ED:4E:0B:D0:25:F6:21:02:9A:A1:
>                 8A:E0:C6:82:33:6C:ED:8F:B2:C2:9C:39:84:05:64:89:AF:87:23:4B:
>                 13:F7:EB:6B:A1:5D:FB:91:47:66:32:3F
> parm:           allow_power_save:int
> parm:           allow_sw_clock_gating:int
> parm:           allow_hw_clock_gating:int
> parm:           bypass_top_level:int
> ```

At least, 
modinfo gasket should display : version: 1.1.3

modinfo apex should display : version: 1.1

Looks like you still have the older version.

pejotigrek, 2021-07-07:
> At least,
> modinfo gasket should display : version: 1.1.3
> 
> modinfo apex should display : version: 1.1
> 
> Looks like you still have the older version.

is there any other way than mentioned here [coral.ai/docs/m2/get-started/#2a-on-linux](https://coral.ai/docs/m2/get-started/#2a-on-linux) and here [coral.ai/docs/m2/get-started/#troubleshooting-on-linux](https://coral.ai/docs/m2/get-started/#troubleshooting-on-linux) to install drivers or force update gasket and apex? I did all the steps mentioned in the official installation guide, and as you saw in my log packages are installed from the right repo but still in wrong version, right?