google-coral/pycoral #32: No module named 'pycoral.pybind._pywrap_coral' (am64, Windows 10, python 3.7-3.9)

gdownes6, 2021-07-11:
Hello,

Similar to a previously reported issue (#24) I am having similar failures under:
Windows 10
python 3.7 (have tried 3.8 and 3.9 to cover all bases)
architecture amd64
* Preliminary setup is done - driver, support files, coral and runtimes all installed without issue.
Windows environment variable PATH limit is disabled and environment variables are present  for python (C:\Pyton37; C:\Python37\Scripts\; C:\Python37\Lib;)

When I try to run the first example classify_image.py, I get the error:
ImportError: DLL load failed: The specified module could not be found.

I have tried:
1. cloning the git repo and installing from there
2. manually installing version 13 of runtime edgetpu_runtime_20210119.zip
3. manually installing PyCoral API using whl's for versions 3.7 (and 3.8 and 3.9 with associated python amd 64 package)
4. manually installing Edge TPU Python API (just in case) - understood its depreciated

All required prerequisite files/packages are installed including numpy, Pillow, etc.

The thing to be imported is present on the system at C:\Python37\Lib\site-packages\pycoral\pybind.

However, what gets installed is named _pywrap_coral.cp37-win_amd64.pyd.
Depending on what version of python, its cp38, cp39, respectively.
I've tried making a copy of the file named _pywrap_coral.pyd, but no luck.

The trace error message is:
...File "C:\Python37\lib\site-packages\pycoral\utils\edgetpu.py", line 24, in <module>
from pycoral.pybind_pyrwap_coral import GetRuntimeVersion as get_runtime_version
ImportError: DLL load failed: The specified module could not be found.

The same happens if you try python versions like 3.8 and 3.9.

Everything appears to be correct but for some strange reason, python can't see it.

I've looked at what is visible to Python by dumping the sys.path
 C:\Python37\python37.zip
 C:\Python37\DLLs
 C:\Python37\lib
 C:\Python37
 C:\Python37\site-packages\
 C:\Python37\site-packages\win32
 C:\Python37\site-packages\win32\lib
 C:\Python37\site-packages\Pythonwin

Can't wrap my head about this one.

Any takers?

gdownes6, 2021-07-11:
Just for run, tried Python 3.6 from scratch and obtained the same result.

hjonnala, 2021-07-12:
Hi,
As you are using python 3.7 and windows can you confirm you have installed pycoral using PIP3 with the below command:

`pip3 install https://github.com/google-coral/pycoral/releases/download/v1.0.1/pycoral-1.0.1-cp37-cp37m-win_amd64.whl`

Let us know whether you are using USB accelerator or Coral Dev Board?

gdownes6, 2021-07-12:
I did not use the pip3 command, but rather pip.
That command returns "ERROR: pycoral-1.0.1-cp37-cp37m-win_amd64.whl is not a supported wheel on this platform"

The command I used, per the instructions on the coral.ai page was: pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral

I am using the Mini PCIe Accelerator.

gdownes6, 2021-07-12:
pycoral-1.0.1.dist-info and pycoral both appear in my site-packages.

hemanthreddyjonnala, 2021-07-12:
What is your default version of python.
That error might be coming because your default version 3.6 and wheel link I have provided is 3.7
Release specific to versions can be found here: https://github.com/google-coral/pycoral/releases

For edgeptu runtime can you confirm you have followed these instructions: https://coral.ai/docs/accelerator/get-started/#runtime-on-windows

can you try to create a virtual env using your desired python version and install pycoral specific to that version.
[image]

gdownes6, 2021-07-12:
Ok, I've corrected the pip3 syntax for Python 3.6.

I cannot create any virtual environments because this machine doesn't support it (ie. the CPU lacks the feature support).

Pycoral 1.0.1 downloads and installs.
> Requirement already satisfied: tflite-runtime==2.50
> Requirement already satisfied: numpy>=1.12.1
> Requirement already satisfied: Pillow>=4.0.0
* olefile does not show up.

For edgecpu runtime, yes I can confirm (part of my opening in this thread).

Same problem.

hemanthreddyjonnala, 2021-07-12:
Hmm.. can you try uninstalling tflite_runtime and pycoral and installing from wheel packages.
Please refer to this [pycoral_windows.txt](https://github.com/google-coral/pycoral/files/6798045/pycoral_windows.txt) for links specific to python3.6

gdownes6, 2021-07-12:
Attempting to work on it.  Git for Windows destroyed my Windows/Bash capabilities.  The script to run examples refuses to work properly.

gdownes6, 2021-07-12:
Same result, it fails.  My console output is attached.

C:\Projects\coral\pycoral>python examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 39, in <module>
    from pycoral.utils.edgetpu import make_interpreter
  File "C:\Python36\lib\site-packages\pycoral\utils\edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: DLL load failed: The specified module could not be found.

[testing.txt](https://github.com/google-coral/pycoral/files/6798263/testing.txt)

gdownes6, 2021-07-12:
Here is the interesting part, if I launch IDLE from the Windows Start menu and do a few more checks some things stand out.
Path Browser > all paths appear correctly including navigating down into C:\Python36\lib\site-packages\pycoral\pybind (where the problem appears to be.

I can get to C:\Python36\lib\site-packages\pycoral\utils where edgetpu.py is found and the browser will expand edgetpu and show all the definitions contained in side it.

Navigate back up to /pycoral/pybind and it stops at _pywrap_coral.cp36-win_amd64.pyd.  It cannot go any further.

[image]

gdownes6, 2021-07-12:
Finally - I have figured it out.

The error “The specified module could not be found” is a bit misleading on Windows because it means either the DLL you are trying to load or any of its dependencies cannot be located.

Having said that, the Microsoft Visual C++ 2019 redistributable was installed but didn't actually register its payload even though it said it did.  Opening the entry in add-remove programs (control panels) and choosing repair solved it.

The examples now run without issue.

Never would have taken this path without your assistance.

Thank you

closed 2021-07-12 by gdownes6

hjonnala, 2021-07-12:
Great, Thanks for sharing fixing information.