google-coral/pycoral #34: ModuleNotFoundError: No module named 'pycoral.pybind'

AkkiSony, 2021-07-23:
I am new to Coral TPU. I am using this USB accelerator and trying to get the initial demo run and monitor the inference time. 
But, I am getting this error. Please let me know if you have any leads?
Thanks in advance guys! :)

[image]

hjonnala, 2021-07-23:
Hi, Let me know your python version. 
please uninstall tflite_runtime and pycoral install it from this whl files: https://github.com/google-coral/pycoral/releases

AkkiSony, 2021-07-26:
Hi @hjonnala,
My current python version is 3.6.9.

I am just unsure which .whl file, I have to install. As I am using a virtual machine (linux) on host(windows), I am sure that I have to use linux_x86-64. However, I am not sure about cp35, cp36. Can you please elaborate to me what does this indicate? Thank you for responding! :)

manoj7410, 2021-07-26:
cp35 means that it is compatible with Python3.5 and cp36 means that it is compatible with Python3.6.

AkkiSony, 2021-07-26:
Thank you for the clarity! :) @manoj7410

[image]

I installed pycoral and tflite_runtime as per your suggestion. Since I have python version 3.6, I installed the corresponding cp36.

Now, I am getting the following error when I run the demo as shown on the coral page. Please let me know what am I missing.

manoj7410, 2021-07-26:
@AkkiSony Are you able to see the USB Accelerator in the output of the command:
- lsusb

AkkiSony, 2021-07-26:
I cannot find it! :/

However, I can see it on the  settings -> USB. Do you have any suggestions?

[image]

AkkiSony, 2021-07-26:
[image]

That's my output! Is it also called Google Inc ?

AkkiSony, 2021-07-26:
I could find Gloabal UniChip after unplugging and plugging the USB device again. 
However, I still have the "ValueError: Failed to load delegate from libedgetpu.so.1" error.

Please find the attached screenshot.
[image]

AkkiSony, 2021-07-26:
One thing that I noticed is that when I execute the classify_image.py, I can visualize the Global UniChip for few seconds, and after sometime it dissapears! :/ 
After I get this Value error, when I do 
- lsusb
I cannot find the device! :/

Do you have any suggestions?

manoj7410, 2021-07-26:
So this is how the USB Accelerator works, while connecting the device with your host PC, the Vendor name will appear as : Global Chip Inc. but once you run the first inference, the Vendor name will change to 'Google Inc'. 
VM is not officially supported by Coral because of various challenges related to IO operations and restrictions.

AkkiSony, 2021-07-26:
Sometimes it works. But it says "Aborted (core dumped)". Can you please let me know why? This is an example code as I said before.
[image]

AkkiSony, 2021-07-26:
Hi,
As VM is not supported, I decided to shift to Windows(host) installation. During the initial steps of installation. 
https://coral.ai/docs/accelerator/get-started/#runtime-on-windows
When I run the .bat file, a command prompt opens and closes every second. This cycle keeps going on. Can you please tell me why am I facing this strange issue?

manoj7410, 2021-07-26:
@AkkiSony Do you see any error too? I haven't seen this behavior yet.

AkkiSony, 2021-07-26:
I do not see any error. The cmd prompt opens and closes within one second. And this keep going on until I close the window. It just keeps flashing in and out! :/

manoj7410, 2021-07-26:
@hjonnala Do you have a Windows machine available with you? If yes, then can you please try to execute the instructions given at: https://coral.ai/docs/accelerator/get-started/#runtime-on-windows ?

AkkiSony, 2021-07-26:
Hi,
I am following the instructions as per the website. I installed Visual Studio C++ 2019 Redistributable. Later, I also unzipped the file and now when I try to install using the batch file, I get this problem of continous flashing of command prompt. I have no idea about the reason for this! :/

AkkiSony, 2021-07-26:
Hi, Sorry it was organisational issue. After I clicked on the .bat file, it required me to enter the admin right. After entering the admin rights, things worked fine. Now I am able to install them! :)

closed 2021-07-26 by manoj7410

AkkiSony, 2021-07-26:
Just a small question with the istallaton procedure for windows system.
Should this be done in an Anaconda platform? If not, then should we create a virtual enviroment and then install the packages like tensorflow_lite and pycoral?

hjonnala, 2021-07-26:
It should be fine in an Anaconda platform too. Let us know if you are facing any issues.

AkkiSony, 2021-07-26:
I am facing this issue when I execute "bash examples/install_requirements.sh classify_image.py"
[image]

hjonnala, 2021-07-26:
please install git bash and try to run those commands.

For instructions please refer to: https://coral.ai/docs/dev-board/get-started/#other-windows-requirements

AkkiSony, 2021-07-26:
Thanks for the answer. That helped me. But since I have to execute these commands on git bash, why do I require Anaconda? 

- How do I create a virtual environment to install the dependencies and libraries?

Please find the output of the command below. Still the test_data directory remain empty! :(

_**$ bash examples/install_requirements.sh examples/classify_image.py
Usage:
  install_requirements.sh [filename]

  Provide the name of an example file in this directory to
  download only the files required for that example.
  If no filename provided, it downloads all example files.
  All files go into the pycoral/test_data/ directory.**_

AkkiSony, 2021-07-26:
$ bash examples/install_requirements.sh classify_image.py
NOT FOUND: mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite
NOT FOUND: mobilenet_v2_1.0_224_inat_bird_quant.tflite
NOT FOUND: inat_bird_labels.txt
NOT FOUND: parrot.jpg

hemanthreddyjonnala, 2021-07-26:
I am not sure why the files are not getting downloaded. 

Script trying to download files form this link, can you check if this link working or not on your Machine: 
https://github.com/google-coral/test_data/raw/master/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite

[image]

AkkiSony, 2021-07-26:
The link works fine. It downloaded the model as soon as I clicked it!
But why is it not possible through the script? 
Do I have to install pycoral and edgetpu only from.whl files?

hemanthreddyjonnala, 2021-07-26:
Please install edgetpu runtime from zip file (https://coral.ai/docs/accelerator/get-started#runtime-on-windows)

tflite_runtime and pycoral from .whl files 

can you try these two commands on bash terminal.

curl -Lso /dev/null -w "%{http_code}" "https://github.com/google-coral/test_data/raw/master/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite"

curl -OL "https://github.com/google-coral/test_data/raw/master/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite"

AkkiSony, 2021-07-26:
The two commands on bash terminal gave me the output as follows.

[image]

hjonnala, 2021-07-26:
the first should give 200 response. Please check with your organization team why the connection is refused.

AkkiSony, 2021-07-26:
Please find the below screenshot. When I execute that command, neither am I getting output nor any error. :/
[image]

AkkiSony, 2021-07-26:
**tflite_runtime and pycoral from .whl files**

For this, do we have to create a virtual environment and then install? I am sorry, I have not worked much on windows side.

hjonnala, 2021-07-26:
virtual environment or without virtual environment both would work..

hjonnala, 2021-07-26:
> Please find the below screenshot. When I execute that command, neither am I getting output nor any error. :/
> [image]

can you try these two commands:

python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg

py examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg

AkkiSony, 2021-07-27:
@hjonnala Can you please give me a short description, how do you install using whl files on windows side?

> virtual environment or without virtual environment both would work.

AkkiSony, 2021-07-27:
@hjonnala Thank you for your reply!

> virtual environment or without virtual environment both would work..

Can you please give me a short description how to install .whl files on windows side. I am confused because, on Anaconda platform we can create virtual environment and then instal using pip command. But on Git bash, since there is no pip installed, how do we do it? Can you please guide me with a short description?

hemanthreddyjonnala, 2021-07-27:
For pip installations  You can use anaconda platform as you described above. 

On bash run pip install command with python executable.
sample command: c:\\pyhton36\\python.exe -m pip install numpy

AkkiSony, 2021-07-27:
Thank you for your input. It really helped to. I added the path of Python and Pip to my PATH variable.

But  when I try to install the .whl files, I am getting these error.
[image]

hjonnala, 2021-07-27:
You are trying install linux versions.

we have new releases. 
Please install 

edgetpu runtime form this (https://github.com/google-coral/libedgetpu/releases/download/release-grouper/edgetpu_runtime_20210726.zip)
and tflite runtime and pycoral from below:
if your python version is 3.6 for windows:
```
pip3 install https://github.com/google-coral/pycoral/releases/download/v2.0.0/tflite_runtime-2.5.0.post1-cp36-cp36m-win_amd64.whl
pip3 install https://github.com/google-coral/pycoral/releases/download/v2.0.0/pycoral-2.0.0-cp36-cp36m-win_amd64.whl

```

AkkiSony, 2021-07-29:
Thank you for making me notice that I was installing the linux version. I am sorry for that! :)

Whe I try to run the example, I am getting the below error. Please can you guide me for the same? :) I thank you in advance for your time! 
[image]

hjonnala, 2021-07-29:
Hi, which python version version you are using?

I think it's due to pillow version. can you try this

pip uninstall pillow
pip install Pillow==8.3.1

AkkiSony, 2021-07-29:
Hello @hjonnala ,
$ python --version
Python 3.8.8

I uninstalled Pillow version 8.2. 
While installing Pillow==8.3.1, I am getting an error

[image]

hjonnala, 2021-07-29:
can you create new virtual environment and try this command.

python3 -m pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral

if python3 not working try:
py -m pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral

AkkiSony, 2021-07-29:
I activated the default virtual environment "base". 
After executing the above command, I am getting the same error as above.

I even tried to add them to trusted hosts.
python -m pip install --trusted-host pypi.python.org --trusted-host files.pythonhosted.org --trusted-host pypi.org --upgrade pip 

Yet the same error as the image attached in the previous thread! :(

hjonnala, 2021-07-29:
Instead of base can you please create new virtual environment and paste the entire output of 
python3 -m pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral

AkkiSony, 2021-07-30:
> can you create new virtual environment and try this command.
>
> python3 -m pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral
>
> if [...]

I tried these commands after creating a new virtual environment using conda platform, yet I have these errors. 

I even added proxy by using 
set http_proxy=.....
set https_proxy=....

[image]

AkkiSony, 2021-07-30:
I solved the above error by adding an extra command. Hope it helps someone with the same error. :)

 pip install pycoral-1.0.1-cp38-cp38-win_amd64.whl -i http://pypi.douban.com/simple- -trusted-host pypi.douban.com

AkkiSony, 2021-08-02:
> the first should give 200 response. Please check with your organization team why the connection is refused.

I now getting segmentation fault as the output for both the statements. Can you help me solve the issue please?
[image]