google-coral/pycoral #35: ImportError: _pywrap_coral.cpython-38-darwin.so not found

robmarkcole, 2021-07-25:
Platform:
```
{'platform': 'Darwin',
 'platform-release': '20.5.0',
 'platform-version': 'Darwin Kernel Version 20.5.0: Sat May  8 05:10:33 PDT 2021; root:xnu-7195.121.3~9/RELEASE_X86_64',
 'architecture': 'x86_64',
 'processor': 'i386',
 'ram': '8 GB'}
```
Python 3.8.5. with venv using `python3 -m venv venv`
On running `from pycoral.utils import edgetpu` and having the usb accelerator plugged into USB, I receive:
```
ImportError: dlopen(/Users/robin/Github/pycoral-experiments/venv/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so, 2): Library not loaded: @rpath/libedgetpu.1.dylib
  Referenced from: /Users/robin/Github/pycoral-experiments/venv/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so
  Reason: image not found
```

hjonnala, 2021-07-25:
Hi, please try uninstall tflite_runtime and pycoral and install from these whl files using python:
https://github.com/google-coral/pycoral/releases

Here the links commands for python3.8 and Mac
```
pip install https://github.com/google-coral/pycoral/releases/download/v1.0.1/tflite_runtime-2.5.0-cp38-cp38-macosx_10_15_x86_64.whl

pip install https://github.com/google-coral/pycoral/releases/download/v1.0.1/pycoral-1.0.1-cp38-cp38-macosx_10_15_x86_64.whl
```

Thanks

robmarkcole, 2021-07-26:
@hjonnala having followed your instructions I receive:

```
ImportError: dlopen(/Users/robin/Github/pycoral-experiments/venv/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so, 2): Library not loaded: @rpath/libedgetpu.1.dylib

  Referenced from: /Users/robin/Github/pycoral-experiments/venv/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so
  Reason: image not found
```

hjonnala, 2021-07-27:
@ robmarkcole. I am not able to reproduce the error. Can you please share entire error details from which line the error is coming.

```
(coral_env) hemanthreddyj-macbookpro1:Desktop hemanthreddyj$ python3
Python 3.8.2 (default, Jun  8 2021, 11:59:35) 
[Clang 12.0.5 (clang-1205.0.22.11)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.uname()
uname_result(system='Darwin', node='hemanthreddyj-macbookpro1.roam.corp.google.com', release='20.5.0', version='Darwin Kernel Version 20.5.0: Sat May  8 05:10:33 PDT 2021; root:xnu-7195.121.3~9/RELEASE_X86_64', machine='x86_64', processor='i386')
>>> from pycoral.utils import edgetpu
>>> 

```

hjonnala, 2021-07-27:
we have new release. Please try to create new virtual env and install using python3 -m pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral.

And let us know if you are still facing any issues.

```
hemanthreddyj-macbookpro1:~ hemanthreddyj$ /usr/bin/python3 -m venv pycoratest_venv
hemanthreddyj-macbookpro1:~ hemanthreddyj$ source pycoratest_venv/bin/activate
(pycoratest_venv) hemanthreddyj-macbookpro1:~ hemanthreddyj$ python3
Python 3.8.2 (default, Jun  8 2021, 11:59:35) 
[Clang 12.0.5 (clang-1205.0.22.11)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> quit()
(pycoratest_venv) hemanthreddyj-macbookpro1:~ hemanthreddyj$ python3 -m pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral
Looking in indexes: https://google-coral.github.io/py-repo/, https://pypi.python.org/simple
Collecting pycoral
  Downloading https://github.com/google-coral/pycoral/releases/download/v2.0.0/pycoral-2.0.0-cp38-cp38-macosx_11_0_x86_64.whl (1.7MB)
     |████████████████████████████████| 1.8MB 3.0MB/s 
Collecting Pillow>=4.0.0 (from pycoral)
  Using cached https://files.pythonhosted.org/packages/8f/7d/1e9c2d8989c209edfd10f878da1af956059a1caab498e5bc34fa11b83f71/Pillow-8.3.1.tar.gz
Collecting tflite-runtime==2.5.0.post1 (from pycoral)
  Downloading https://github.com/google-coral/pycoral/releases/download/v2.0.0/tflite_runtime-2.5.0.post1-cp38-cp38-macosx_11_0_x86_64.whl (1.4MB)
     |████████████████████████████████| 1.4MB 24.4MB/s 
Collecting numpy>=1.16.0 (from pycoral)
  Downloading https://files.pythonhosted.org/packages/0b/a7/e724c8df240687b5fd62d8c71f1a6709d455c4c09432c7412e3e64f4cbe5/numpy-1.21.1.zip (10.3MB)
     |████████████████████████████████| 10.3MB 4.1MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: numpy
  Building wheel for numpy (PEP 517) ... done
  Created wheel for numpy: filename=numpy-1.21.1-cp38-cp38-macosx_11_0_x86_64.whl size=6941884 sha256=c23f4b26de45b8a05a3e0bca4ab28a5f7ee0eb2c871b6cf4bc47a1e3ac073827
  Stored in directory: /Users/hemanthreddyj/Library/Caches/pip/wheels/56/11/89/31a50277e9fc8e2d582974f9a491cbb1a3c07c717b2fcbcddb
Successfully built numpy
Installing collected packages: Pillow, numpy, tflite-runtime, pycoral
  Running setup.py install for Pillow ... done
Successfully installed Pillow-8.3.1 numpy-1.21.1 pycoral-2.0.0 tflite-runtime-2.5.0.post1
WARNING: You are using pip version 19.2.3, however version 21.2.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
(pycoratest_venv) hemanthreddyj-macbookpro1:~ hemanthreddyj$ python3
Python 3.8.2 (default, Jun  8 2021, 11:59:35) 
[Clang 12.0.5 (clang-1205.0.22.11)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from pycoral.utils import edgetpu
>>> 

```

robmarkcole, 2021-07-28:
Following exactly your steps:

```
Robins-MacBook:pycoral-experiments robin$ python3 -m venv venv
Robins-MacBook:pycoral-experiments robin$ source venv/bin/activate
(venv) Robins-MacBook:pycoral-experiments robin$ python3 -m pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral
Looking in indexes: https://google-coral.github.io/py-repo/, https://pypi.python.org/simple
Collecting pycoral
  Using cached https://github.com/google-coral/pycoral/releases/download/v1.0.1/pycoral-1.0.1-cp38-cp38-macosx_10_15_x86_64.whl (1.6 MB)
Collecting tflite-runtime==2.5.0
  Using cached https://github.com/google-coral/pycoral/releases/download/release-frogfish/tflite_runtime-2.5.0-cp38-cp38-macosx_10_15_x86_64.whl (1.3 MB)
Collecting numpy>=1.12.1
  Using cached numpy-1.21.1-cp38-cp38-macosx_10_9_x86_64.whl (16.9 MB)
Collecting Pillow>=4.0.0
  Using cached Pillow-8.3.1-cp38-cp38-macosx_10_10_x86_64.whl (2.9 MB)
Installing collected packages: numpy, tflite-runtime, Pillow, pycoral
Successfully installed Pillow-8.3.1 numpy-1.21.1 pycoral-1.0.1 tflite-runtime-2.5.0
WARNING: You are using pip version 20.1.1; however, version 21.2.1 is available.
You should consider upgrading via the '/Users/robin/Github/pycoral-experiments/venv/bin/python3 -m pip install --upgrade pip' command.
(venv) Robins-MacBook:pycoral-experiments robin$ python3
Python 3.8.5 (default, Jul 21 2020, 10:48:26) 
[Clang 11.0.3 (clang-1103.0.32.62)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from pycoral.utils import edgetpu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/robin/Github/pycoral-experiments/venv/lib/python3.8/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: dlopen(/Users/robin/Github/pycoral-experiments/venv/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so, 2): Library not loaded: @rpath/libedgetpu.1.dylib
  Referenced from: /Users/robin/Github/pycoral-experiments/venv/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so
  Reason: image not found
```

I notice you are on Python 3.8.2 whilst I am on Python 3.8.5

hjonnala, 2021-07-28:
we can see it is using cache to install the libraries. Can you try create virtual env with different name. You should be getting pycoral2.0.0.

robmarkcole, 2021-07-30:
@hjonnala even creating a fresh venv with a new name, I encounter the issue as it appears its not getting pycoral2.0.0. Tried adding `--upgrade` but still pulls 1.0.1

```
Robins-MacBook:pycoral-experiments robin$ python3 -m venv venv2Robins-MacBook:pycoral-experiments robin$ source venv2/bin/activate
(venv2) Robins-MacBook:pycoral-experiments robin$ python3 -m pip install --index-url https://google-coral.github.io/py-repo/ --extra-index-url=https://pypi.python.org/simple pycoral
Looking in indexes: https://google-coral.github.io/py-repo/, https://pypi.python.org/simple
Collecting pycoral
  Downloading https://github.com/google-coral/pycoral/releases/download/v1.0.1/pycoral-1.0.1-cp38-cp38-macosx_10_15_x86_64.whl (1.6 MB)
     |████████████████████████████████| 1.6 MB 6.7 MB/s 
Collecting Pillow>=4.0.0
  Using cached Pillow-8.3.1-cp38-cp38-macosx_10_10_x86_64.whl (2.9 MB)
Collecting numpy>=1.12.1
  Using cached numpy-1.21.1-cp38-cp38-macosx_10_9_x86_64.whl (16.9 MB)
Collecting tflite-runtime==2.5.0
  Downloading https://github.com/google-coral/pycoral/releases/download/release-frogfish/tflite_runtime-2.5.0-cp38-cp38-macosx_10_15_x86_64.whl (1.3 MB)
     |████████████████████████████████| 1.3 MB 2.2 MB/s 
Installing collected packages: Pillow, numpy, tflite-runtime, pycoral
Successfully installed Pillow-8.3.1 numpy-1.21.1 pycoral-1.0.1 tflite-runtime-2.5.0
WARNING: You are using pip version 20.1.1; however, version 21.2.1 is available.
You should consider upgrading via the '/Users/robin/Github/pycoral-experiments/venv2/bin/python3 -m pip install --upgrade pip' command.
(venv2) Robins-MacBook:pycoral-experiments robin$ python3
Python 3.8.5 (default, Jul 21 2020, 10:48:26) 
[Clang 11.0.3 (clang-1103.0.32.62)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from pycoral.utils import edgetpu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/robin/Github/pycoral-experiments/venv2/lib/python3.8/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: dlopen(/Users/robin/Github/pycoral-experiments/venv2/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so, 2): Library not loaded: @rpath/libedgetpu.1.dylib
  Referenced from: /Users/robin/Github/pycoral-experiments/venv2/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so
  Reason: image not found
>>> quit()
(venv2) Robins-MacBook:pycoral-experiments robin$ pip freeze
numpy==1.21.1
Pillow==8.3.1
pycoral==1.0.1
tflite-runtime==2.5.0
```

dmitriykovalev, 2021-07-30:
@robmarkcole Looks like you are missing `libedgetpu` shared library according to the error message: `2): Library not loaded: @rpath/libedgetpu.1.dylib`. Have you installed `libedgetpu` from the [runtime zip archive](https://github.com/google-coral/libedgetpu/releases)?

robmarkcole, 2021-07-31:
@dmitriykovalev thanks for the pointer. However I hit various issues and am unable to install 

```
Robins-MacBook:edgetpu_runtime robin$ brew update
Error: 
  homebrew-core is a shallow clone.
To `brew update`, first run:
  git -C /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core fetch --unshallow
This restriction has been made on GitHub's request because updating shallow
clones is an extremely expensive operation due to the tree layout and traffic of
Homebrew/homebrew-core and Homebrew/homebrew-cask. We don't do this for you
automatically to avoid repeatedly performing an expensive unshallow operation in
CI systems (which should instead be fixed to not use shallow clones). Sorry for
the inconvenience!
```

dmitriykovalev, 2021-07-31:
@robmarkcole Looks like there is an error while installing `libusb` using Homebrew. You probably need to [install](https://stackoverflow.com/questions/9329243/how-to-install-xcode-command-line-tools) the latest version of Xcode Command Line Tools and then try again. `Libusb` is a dependency for `libedgetpu`, so it should be installed anyway.

robmarkcole, 2021-08-01:
@dmitriykovalev I am currently wiping and reinstalling my command line tools after the following, lets see if this resolves it. However all indications are that my tools are already up to date

```
==> Downloading https://homebrew.bintray.com/bottles/libusb-1.0.23.catalina.bottle.1.tar.gz
#=#=-#  #                                                                     
curl: (22) The requested URL returned error: 403 Forbidden
Error: Failed to download resource "libusb"
Download failed: https://homebrew.bintray.com/bottles/libusb-1.0.23.catalina.bottle.1.tar.gz
Warning: Bottle installation failed: building from source.
==> Downloading https://github.com/libusb/libusb/releases/download/v1.0.23/libusb-1.0.23.tar.bz2
Already downloaded: /Users/robin/Library/Caches/Homebrew/downloads/c49fbf2356d71d89ba2c94312bf2c237035d8975418b55254f25ea14ce81692a--libusb-1.0.23.tar.bz2
Error: Your Command Line Tools are too outdated.
Update them from Software Update in System Preferences or run:
  softwareupdate --all --install --force

If that doesn't show you an update run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/more/.

Error: Your CLT does not support macOS 11.
It is either outdated or was modified.
Please update your CLT or delete it if no updates are available.
Update them from Software Update in System Preferences or run:
  softwareupdate --all --install --force

If that doesn't show you an update run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/more/.
```

dmitriykovalev, 2021-08-03:
@robmarkcole That's interesting. Please copy the output from the following commands:
```
otool -L /usr/local/lib/libedgetpu.1.0.dylib
```
and
```
brew --prefix libusb
```

robmarkcole, 2021-08-04:
@dmitriykovalev 
```
(venv) MacBook-Pro:pycoral-experiments robin$ otool -L /usr/local/lib/libedgetpu.1.0.dylib
/usr/local/lib/libedgetpu.1.0.dylib:
        /usr/local/lib/libedgetpu.1.dylib (compatibility version 0.0.0, current version 0.0.0)
        /usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 905.6.0)
        /usr/local/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
        /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1775.118.101)
        /usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.100.5)
        /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1775.118.101)
(venv) MacBook-Pro:pycoral-experiments robin$ brew --prefix libusb
/usr/local/opt/libusb
```

dmitriykovalev, 2021-08-04:
@robmarkcole Everything seems to be correct so far. Do you have Coral USB Accelerator properly connected to the Mac? Can you see it in the output of `ioreg -p IOUSB -w0 -l`?

Just in case, can you also try to run the python script with `sudo`, will it make any difference?

dmitriykovalev, 2021-08-04:
@robmarkcole It's interesting that you have `libusb` installed at `/usr/local/lib/libusb-1.0.0.dylib` according to `otool` output. Homebrew installs it to a different location: `$(brew --prefix libusb)/lib` which is `/usr/local/opt/libusb/lib` on your Mac.

robmarkcole, 2021-08-04:
Could be the device is not found as I am using an adapter? 

[image]

The output:
```

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
MacBook-Pro:pycoral-experiments robin$ ioreg -p IOUSB -w0 -l
+-o Root  <class IORegistryEntry, id 0x100000100, retain 20>
  | {
  |   "IOKitBuildVersion" = "Darwin Kernel Version 20.6.0: Wed Jun 23 00:26:31 PDT 2021; root:xnu-7195.141.2~5/RELEASE_X86_64"
  |   "OS Build Version" = "20G80"
  |   "IOBluetoothActive" = Yes
  |   "OSKernelCPUSubtype" = 3
  |   "OSKernelCPUType" = 16777223
  |   "OSPrelinkKextCount" = 18
  |   "IORegistryPlanes" = {"IOPower"="IOPower","IOService"="IOService","IO80211Plane"="IO80211Plane","IOUSB"="IOUSB","CoreCapture"="CoreCapture","IOACPIPlane"="IOACPIPlane","IODeviceTree"="IODeviceTree"}
  |   "IOConsoleLocked" = No
  |   "IOConsoleUsers" = ({"kCGSSessionOnConsoleKey"=Yes,"kCGSSessionIDKey"=257,"kSCSecuritySessionID"=100009,"kCGSessionLongUserNameKey"="Robin","kCGSSessionSystemSafeBoot"=No,"kCGSSessionUserIDKey"=501,"kCGSSessionAuditIDKey"=100009,"kCGSessionLoginDoneKey"=Yes,"kCGSSessionGroupIDKey"=20,"kCGSSessionUserNameKey"="robin","kCGSSessionLoginwindowSafeLogin"=No})
  |   "IOKitDiagnostics" = {"Instance allocation"=12052576,"Container allocation"=17590363,"Pageable allocation"=3555127296,"Classes"={"IONDRVFramebuffer"=0,"IGAccelCommandQueue"=11,"IOHIDEventServiceFastPathUserClient"=0,"IOKitDiagnosticsClient"=0,"IONaturalMemoryCursor"=0,"AppleUSBLegacyHub"=1,"AppleUSBDiagnostics"=0,"AppleUSBXHCIIsochronousRequestPool"=2,"IOAudioClientBufferSet"=0,"AppleSMCPMC"=0,"AppleUSBRequest"=1,"DspFuncBuzzKill"=1,"AppleHDAMikeyInternalCS4208"=0,"AppleASMedia1042USBXHCICommandRing"=0,"IGAccelSharedUserClient"=33,"IOUSBMassStorageDriverRequestTimer"=0,"AppleHDATDM_Codec"=1,"IOHDACodecDevice"=1,"IOThunderboltDeficitCommandQueue"=1,"IORegistryEntry"=128,"OSLiveNumber"=6,"AppleActuatorDevice"=1,"IOUSBMassStorageUASDriver"=0,"AppleHDAWorkLoop"=1,"IORTC"=1,"IOHIDevice"=0,"IOPCIEventSource"=0,"AppleThunderboltNHIType4"=0,"IOThunderboltAbstractMicro"=1,"IO80211DriverCommandDescriptor"=0,"AppleHDAHardwareConfigDriverLoader"=0,"DspFuncUserClient"=0,"IOMemoryCursor"=1,"AppleSmartBatteryManager"=1,"IOThunderboltXDPropertiesDirectory"=2,"IOPMGR"=0,"AppleHDATDM_CS42L81"=0,"IOBreaker"=0,"AppleIntelFramebuffer"=3,"IO80211NANDataInterfacePeerManager"=0,"IGFlushNotifyInterruptEventSource"=4,"AppleHDAFunctionGroup_80862807"=0,"AppleUSBUserHCIDevice"=0,"IOUSBLowLatencyCommandLegacy"=0,"AppleThunderboltNHIReceiveRingManager"=1,"AppleHIDKeyboardEventDriver"=0,"AppleThunderboltIPReceiveCommand"=0,"EFIData"=93,"AIDCircularBufferShared"=0,"AppleUSB20KeyboardHub"=0,"AppleUSBUserHCITransferQueue"=0,"IO80211NANPeer"=0,"AppleUSB20HubPort"=4,"AppleUSBXHCIPort"=2,"IOAccelCommandQueue"=1,"DspFunc4ChOutput"=1,"IGRangeAllocator"=37,"IOSurfaceSharedEventNotification"=0,"IOBluetoothHostControllerUserClient"=0,"AppleKeyStoreUserClient"=23,"IOThunderboltSwitchType1"=0,"AppleHDAFunctionGroupWM8800"=0,"IOSMBusRequest"=0,"IntelFBClientControl"=1,"IOSkywalkPacket"=1,"DspFuncBeamFormer"=0,"IOPMServiceInterestNotifier"=115,"AppleACPILid"=1,"AppleUpstreamUserClientDriver"=3,"IOHIDResourceQueue"=1,"IOStorage"=4,"IOAccelSegmentResourceList"=0,"AppleUSBRequestPool"=2,"IOSerialStreamSync"=1,"IOHIDConsumer"=0,"IO80211AWDLMulticastPeer"=1,"AppleUSBHostCompositeDevice"=1,"IOSharedInterruptController"=2,"IOThunderboltXDPropertiesEntry"=11,"IOHIDClientData"=4,"IOBluetoothMemoryDescriptorRetainer"=0,"IOGraphicsWorkLoop"=2,"hv_vmx_vm_t"=0,"AppleHSSPIController"=1,"AppleHDATDMBusManagerCS4208"=0,"IGAccelVideoContextMedia"=0,"IOTimeSyncClockManagerUserClient"=2,"OSSerializer"=50,"IOPCIMessagedInterruptController"=1,"OSCollection"=5,"IOThunderboltConfigMultiReadCommand"=1,"IOUserEthernetResourceUserClient"=0,"IOThunderboltSet"=26,"IO80211RealTimePeerManager"=1,"IO80211NANDataPathResponder"=0,"IOUSBDeviceUserClientV2"=1,"SmbusHandler"=1,"AppleHDAEngineOutput"=1,"AIDReporters"=0,"_IOServiceNullNotifier"=1,"AIDInterface"=0,"IO80211NANDataPathInitiator"=0,"IOUserNetworkRxCompletionQueue"=0,"IOTimeSyncUnicastUDPv4PtPPort"=0,"IOSkywalkMemorySegment"=136,"com_apple_driver_eficheckuserclient"=0,"IOBluetoothHCIUserClient"=13,"IOBluetoothPacketLoggerUserClient"=0,"AppleUSB20InternalIntelHubClient"=0,"AppleIntelPCHPMC"=1,"AppleIntelPlane"=9,"com_apple_driver_pm_cpu_reporter"=1,"IONetworkStackUserClient"=1,"AppleMultitouchEventDriverV2Wrapper"=1,"AppleDisplay"=0,"IOTimeSyncPortManager"=0,"IOBluetoothHostController"=1,"IOInterleavedMemoryDescriptor"=0,"_IOServiceNotifier"=519,"AppleAPFSUserClient"=0,"AppleHDAFunctionGroupGT216"=0,"AppleUSBLegacyInterfaceUserClient"=0,"AppleUSBUserHCIResources"=1,"AppleGraphicsDevicePolicy"=1,"IGAccel2DContext"=3,"AppleHDAFunctionGroupCS4208"=0,"IOSkywalkEthernetInterface"=1,"IGScheduler"=1,"IOTimeSyncEthernetInterfaceAdapter"=0,"IOSKArena"=17,"AppleUSBXHCILPTCommandRing"=0,"IOUSBHostDevice"=5,"IOSubMemoryDescriptor"=6,"AppleHSBluetoothInterface"=0,"AppleUSBRequestCompleter"=1,"IOThunderboltXDPropertiesCache"=1,"DspFuncVolume_4ch"=0,"IOBluetoothL2CAPSignalChannel"=0,"IOAVBNubUserClient"=0,"AppleThunderboltNHIIntelPCI"=1,"IOThunderboltXDomainLink"=0,"IOServiceUserNotification"=422,"AppleMultitouchTrackpadHIDEventDriver"=1,"AppleFDEKeyStoreUserClient"=0,"IOThunderboltConfigPrepareToSleepCommand"=1,"IO80211ThroughputCache"=0,"AppleUSBHostResourcesTypeCBPC"= [... cut ...]
          "IOPowerManagement" = {"DevicePowerState"=2,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
          "kUSBCurrentConfiguration" = 1
          "Device Speed" = 3
          "USB Vendor Name" = "GenesysLogic"
          "idVendor" = 1507
          "IOGeneralInterest" = "IOCommand is not serializable"
          "kUSBProductString" = "USB3.1 Hub"
          "kUSBVendorString" = "GenesysLogic"
          "IOClassNameOverride" = "IOUSBDevice"
        }
 ```

dmitriykovalev, 2021-08-04:
Looks like there is an issue with your USB hub. I don't see Coral device in your output. The accelerator has `"idVendor" = 6766`. That's what I have on my Mac:
```
$ ioreg -p IOUSB -w0 -l | grep -i 6766
          "idVendor" = 6766
```
You can also do `brew install usbutils` and run `lsusb` to have nicer output. On my Mac I see this line:
```
Bus 000 Device 001: ID 1a6e:089a Global Unichip Corp.
```
Notice that 0x1a6e (hex) equals to 6766 (dec).

Or you can run `System Information` util and see graphical output:
[image]

robmarkcole, 2021-08-05:
I also get
```
MacBook-Pro:~ robin$ ioreg -p IOUSB -w0 -l | grep -i 6766
            "idVendor" = 6766
```

dmitriykovalev, 2021-08-05:
I see, device is properly connected then. Please run this command and copy the output:
```
python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'
```

robmarkcole, 2021-08-05:
```
(venv) MacBook-Pro:pycoral-experiments robin$ python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'
BuildLabel(COMPILER=Apple LLVM 12.0.5 (clang-1205.0.22.9),DATE=redacted,TIME=redacted), RuntimeVersion(14)
```

dmitriykovalev, 2021-08-05:
Looks like everything is connected properly and should be working fine, it's weird that it doesn't.

Is there a chance you can try connecting USB accelerator directly to the laptop without any hubs? There are usually two ways of doing this:

- USB-C to USB-C cable
- USB to USB-C adapter

It could also be useful to try a different USB hub if possible.

robmarkcole, 2021-08-08:
Alright, using the USB-C to USB-C cable the inference runs without issue! Closing this issue and purchasing extra cables :-)

closed 2021-08-08 by robmarkcole

dmitriykovalev, 2021-08-09:
@robmarkcole I'm glad it works now! According to my experience, USB cable/power issues are the hardest to debug.