google-coral/pycoral #38: Unable to cast Python instance to C++ type

mzignis, 2021-08-03:
Hi, I'm trying to check dma-buffer and use it to run my model (on google coral dev board), but when I try call a few functions from pycoral.utils.edgetpu, pybind return error: 

Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
RuntimeError: Unable to cast Python instance to C++ type (compile in debug mode for details)

My code:
`
import pycoral.utils.edgetpu as edgetpu
interpreter = edgetpu.make_interpreter('model_edgetpu.tflite')
edgetpu.supports_dmabuf(interpreter)
`

I've try to use all pycoral versions (1.0.0, 1.0.1 and 2.0.0) but in each of it there is the same issue

hjonnala, 2021-08-04:
Hi, can you please try the code snippet below

```
import pycoral.utils.edgetpu as edgetpu
interpreter = edgetpu.make_interpreter('model_edgetpu.tflite')
edgetpu.supports_dmabuf(interpreter._native_handle())
```

mzignis, 2021-08-10:
Sorry for response, its return True right now but I still have problem to run `invoke_with_dmabuffer`.
I use GstAllocator to allocate memory in DMA but my code crash on `invoke...`

```
import os
import gi

gi.require_version('GstAllocators', '1.0')
from gi.repository import GstAllocators

import pycoral.utils.edgetpu as edgetpu

interpreter = edgetpu.make_interpreter('model_edgetpu.tflite')
interpreter.allocate_tensors()

print(interpreter)
print(edgetpu.supports_dmabuf(interpreter._native_handle()))

f = os.open('test.jpg', os.O_RDONLY)
print('File:', f)

dma_allocator = GstAllocators.DmaBufAllocator.new()
memory = GstAllocators.DmaBufAllocator.alloc(dma_allocator, f, 6400)
print(memory)
print(GstAllocators.is_dmabuf_memory(memory), GstAllocators.is_fd_memory(memory), GstAllocators.is_phys_memory(memory))
print(memory.size)

fd = GstAllocators.dmabuf_memory_get_fd(memory)
print(type(fd), fd)

edgetpu.invoke_with_dmabuffer(interpreter._native_handle(), fd, memory.size)
```

hjonnala, 2021-08-11:
Have you checked map_buffer true of false. I think it needs to False to invoke_with_dambuufer.
https://github.com/google-coral/pycoral/blob/master/pycoral/utils/edgetpu.py#L189
`interpreter_handle = interpreter._native_handle()

map_buffer = not GstAllocators.is_dmabuf_memory(memory) or not edgetpu.supports_dmabuf(interpreter_handle)
print('map_buffer', map_buffer )`

hjonnala, 2021-08-20:
Hi @mzignis are you still having issue?

mzignis, 2021-08-22:
Hi, yes I was trying few other things with dma buffer but still there is the same issue. Do you have any idea what can case that?

hjonnala, 2021-08-24:
can you please share the model that you are using?