google-coral/pycoral #39: Cannot run 'Getting started examples' for USB Accelerator on Windows 10: "ValueError: Failed to load delegate from edgetpu.dll"

asylunatic, 2021-08-04:
On two different Windows 10 machines I run into the same problem when trying to run the pycoral example for the USB Accelerator. In both cases, I followed the steps from _https://coral.ai/docs/accelerator/get-started/_ . When trying to run classify_image.py, the script crashes when calling `make_interpreter`:

```Traceback (most recent call last):
  File "C:\Users\Gebruiker\Documents\Thesis\CoralSetup\venv\lib\site-packages\tflite_runtime\interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "C:\Users\Gebruiker\Documents\Thesis\CoralSetup\venv\lib\site-packages\tflite_runtime\interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Gebruiker\Documents\Thesis\CoralSetup\coral\pycoral\examples\classify_image.py", line 121, in <module>
    main()
  File "C:\Users\Gebruiker\Documents\Thesis\CoralSetup\coral\pycoral\examples\classify_image.py", line 71, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "C:\Users\Gebruiker\Documents\Thesis\CoralSetup\venv\lib\site-packages\pycoral\utils\edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "C:\Users\Gebruiker\Documents\Thesis\CoralSetup\venv\lib\site-packages\pycoral\utils\edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "C:\Users\Gebruiker\Documents\Thesis\CoralSetup\venv\lib\site-packages\tflite_runtime\interpreter.py", line 163, in load_delegate
    library, str(e)))
ValueError: Failed to load delegate from edgetpu.dll
```
On both machines, it the first time I did the installation, so I am pretty sure it does not have to do with versions (as was suggested in issues raised earlier: #281 and #46). 

I have so far not been able to find out what causes this.

manoj7410, 2021-08-04:
@asylunatic Do you see any LED glowing on the USB Accelerator ?

asylunatic, 2021-08-04:
@manoj7410 yes! it is also listed in the device manager under USB devices, and I've added the path to edgetpu.dll to the environment variables to be sure

hjonnala, 2021-08-04:
@asylunatic could you please try with edgetpu_runtime_20210119.zip from this page https://coral.ai/software/#edgetpu-runtime

Lennyz1988, 2021-08-04:
I'm having the same issue but on an aarch64 system. I just updated everything to the latest versions.

pycoral             2.0.0
tflite-runtime      2.5.0.post1

I tried these trained models:
tf2_ssd_mobilenet_v2_coco17_ptq_edgetpu.tflite
mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite (was working fine before the update)

Traceback (most recent call last):
  File "/home/lennyz1988/.local/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/home/lennyz1988/.local/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "coralkolksluis.py", line 163, in <module>
    classify_frame(model_file)
  File "coralkolksluis.py", line 111, in classify_frame
    interpreter = edgetpu.make_interpreter(model_file)
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/home/lennyz1988/.local/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 163, in load_delegate
    library, str(e)))
ValueError: Failed to load delegate from libedgetpu.so.1

hjonnala, 2021-08-04:
@Lennyz1988 can you paste the output of command `lsusb`

Lennyz1988, 2021-08-04:
@hjonnala 
Bus 008 Device 002: ID 2109:0817 VIA Labs, Inc.
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 002: ID 2109:2817 VIA Labs, Inc.
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

hjonnala, 2021-08-04:
@Lennyz1988 I assume you are using USB accelerator. If its connected you should see  Global Unichip Corp. or Google in ouput

```
(my_app_venv) hemanthreddyj@hemanthreddyj-glaptop:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 1050:0200 Yubico.com Gnubby U2F
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 010: ID 1a6e:089a Global Unichip Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 138a:0097 Validity Sensors, Inc. 
Bus 001 Device 003: ID 13d3:5682 IMC Networks SunplusIT Integrated Camera
Bus 001 Device 002: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 010: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

AkkiSony, 2021-08-06:
$ python examples/classify_image.py -m test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite -l test_data/inat_bird_labels.txt -i test_data/parrot.jpg
Traceback (most recent call last):
  File "C:\Users\asdf\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "C:\Users\asdf\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 85, in <module>
    main()
  File "examples/classify_image.py", line 62, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "C:\Users\asdf\Anaconda3\envs\coral_env\lib\site-packages\pycoral\utils\edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "C:\Users\asdf\Anaconda3\envs\coral_env\lib\site-packages\pycoral\utils\edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "C:\Users\asdf\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from edgetpu.dll

hjonnala, 2021-08-06:
@AkkiSony can you paste the output of following command.

`python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'`

Please check if you have edgetpu.dll in this path C:\Windows\System32?
Please consider trying on personal laptop, that would help you to resolve the issues with organization laptop.

AkkiSony, 2021-08-09:
> python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'

 python -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'
BuildLabel(COMPILER=MSVC 192628805,DATE=Nov  4 2020,TIME=11:06:09), RuntimeVersion(13)
(coral_env)

> Please check if you have edgetpu.dll in this path C:\Windows\System32?

[image]

AkkiSony, 2021-08-09:
> python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'

It's strange that I am getting this output now.

python -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\asdf\Coral-USB\coral1\pycoral\pycoral\utils\edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ModuleNotFoundError: No module named 'pycoral.pybind'
(coral_env)

hjonnala, 2021-08-09:
@AkkiSony pycoral modules are fetching from this path C:\Users\asdf\Coral-USB\coral1\pycoral\pycoral\utils\edgetpu.py.. But they need to come from coral_env/lib/site-packages/pycoral..

AkkiSony, 2021-08-09:
@hjonnala Do you have any idea what modification should I do?

hjonnala, 2021-08-09:
@AkkiSony you can try renaming pycoral to pycoral1 or something else in coral1 directory..

AkkiSony, 2021-08-09:
It dint work even after renaming! :/

hjonnala, 2021-08-09:
@AkkiSony please paste the error you are getting after renaming.

AkkiSony, 2021-08-09:
$ python -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\AkshayaC\Coral-USB\coral1\pycoral1\pycoral\utils\edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ModuleNotFoundError: No module named 'pycoral.pybind'
(coral_env)

@hjonnala

hjonnala, 2021-08-09:
@AkkiSony please change pycoral name also inside pycoral1 directory

AkkiSony, 2021-08-09:
After renaming it, I got the following output.

BuildLabel(COMPILER=MSVC 192628805,DATE=Nov  4 2020,TIME=11:06:09), RuntimeVersion(13)
----------------------------------------------------------------------------------------------------------
When I run the example code, I get the following error.
 File "C:\Users\AkshayaC\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "C:\Users\AkshayaC\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 85, in <module>
    main()
  File "examples/classify_image.py", line 62, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "C:\Users\asdf\Anaconda3\envs\coral_env\lib\site-packages\pycoral\utils\edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "C:\Users\asdf\Anaconda3\envs\coral_env\lib\site-packages\pycoral\utils\edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "C:\Users\asdf\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from edgetpu.dll
__________________________________________________________________________________________________________

hjonnala, 2021-08-09:
@AkkiSony It should be working fine with runtime version 13. Please make sure USB accelerator connected to the machine.

AkkiSony, 2021-08-09:
Thanks. I am getting this error when the USB accelerator is connected.

asylunatic, 2021-08-10:
@hjonnala I uninstalled the latest version and installed edgetpu_runtime_20210119.zip as you recommended and it runs now! Thanks

closed 2021-08-10 by hjonnala