google-coral/pycoral #41: Coral Ejects at Runtime

ATLTVHEAD, 2021-08-05:
Installed edgetpu_runtime_20210726 for windows running the latest C++ distribution of Visual Studio

I am running through the demo on the getting started page, currently on the python execution of the parrot classification. 

Every time I try to execute the python script, the Google Coral "ejects" from the my pc and the terminal brings up a fresh line as though it finished properly. No output from python is displayed. 

Any ideas to try to debug? I uninstalled and reinstalled with restarts in-between and still the same problem.

manoj7410, 2021-08-05:
@ATLTVHEAD Which Coral device are you working with ? 
And are you working on a VM or on a native installation ?

ATLTVHEAD, 2021-08-05:
> @ATLTVHEAD Which Coral device are you working with ? 
> And are you working on a VM or on a native installation ? 

Using the USB Accelerator model and installing directly on a Windows 10 machine.

hjonnala, 2021-08-05:
please try with edgetpu_runtime_20210119.zip from this page https://coral.ai/software/#edgetpu-runtime

AkkiSony, 2021-08-06:
@hjonnala 
$ python examples/classify_image.py -m test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite -l test_data/inat_bird_labels.txt -i test_data/parrot.jpg
Traceback (most recent call last):
  File "C:\Users\AkshayaC\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "C:\Users\AkshayaC\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 85, in <module>
    main()
  File "examples/classify_image.py", line 62, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "C:\Users\AkshayaC\Anaconda3\envs\coral_env\lib\site-packages\pycoral\utils\edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "C:\Users\AkshayaC\Anaconda3\envs\coral_env\lib\site-packages\pycoral\utils\edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "C:\Users\AkshayaC\Anaconda3\envs\coral_env\lib\site-packages\tflite_runtime\interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from edgetpu.dll

Do you have any solution for this error?
I have installed edgetpu_runtime_20210119.zip  and working on Native windows10.

ATLTVHEAD, 2021-08-07:
> please try with edgetpu_runtime_20210119.zip from this page https://coral.ai/software/#edgetpu-runtime

With Runtime 20210119, the usb accelerator now disconnects and connects a number of times, but gives me inferencing output and results so everything seems to be running! Thank you!

closed 2021-08-07 by ATLTVHEAD