google-coral/pycoral #44: Library not loaded: /opt/local/lib/libusb-1.0.0.dylib

VelizarVESSELINOV, 2021-08-15:
# Error
```bash
python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "~/Documents/GitHub/pycoral/examples/classify_image.py", line 40, in <module>
    from pycoral.utils.edgetpu import make_interpreter
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pycoral/pybind/_pywrap_coral.cpython-39-darwin.so, 2): Library not loaded: /opt/local/lib/libusb-1.0.0.dylib
  Referenced from: /usr/local/lib/libedgetpu.1.dylib
  Reason: image not found
```
# Install
Following the instructions: https://coral.ai/docs/accelerator/get-started
```bash
python3 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0
Looking in indexes: https://pypi.org/simple, https://google-coral.github.io/py-repo/
Requirement already satisfied: pycoral~=2.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (2.0.0)
Requirement already satisfied: Pillow>=4.0.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pycoral~=2.0) (8.2.0)
Requirement already satisfied: tflite-runtime==2.5.0.post1 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pycoral~=2.0) (2.5.0.post1)
Requirement already satisfied: numpy>=1.16.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pycoral~=2.0) (1.21.0)
```

#  libusb 1.0.24 already installed
```bash
brew install libusb
Error:
  homebrew-core is a shallow clone.
To `brew update`, first run:
  git -C /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core fetch --unshallow
This command may take a few minutes to run due to the large size of the repository.
This restriction has been made on GitHub's request because updating shallow
clones is an extremely expensive operation due to the tree layout and traffic of
Homebrew/homebrew-core and Homebrew/homebrew-cask. We don't do this for you
automatically to avoid repeatedly performing an expensive unshallow operation in
CI systems (which should instead be fixed to not use shallow clones). Sorry for
the inconvenience!
Warning: libusb 1.0.24 is already installed and up-to-date.
To reinstall 1.0.24, run:
  brew reinstall libusb
```

hjonnala, 2021-08-15:
Hi, Have you installed edgetpu runtime https://coral.ai/docs/accelerator/get-started#runtime-on-mac.

can you please paste the output of below commnad:

`python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'`

hjonnala, 2021-08-24:
Hi, @VelizarVESSELINOV were you able to run the demo?

cku239, 2021-09-02:
I have the same issue . Followed the instructions and get the same error. I tried the python 3 command and all I get is
File "<string>, line 1
    from pycoral.utils.edgetpu import
SyntaxError: invalid syntax

hjonnala, 2021-09-03:
@cku239 if you are trying the code in file try this:

```
from pycoral.utils.edgetpu import get_runtime_version
print(get_runtime_version())
```

cku239, 2021-09-15:
I am getting the same error. Is this because homebrew  (on M1)is installing the libusb library in /opt/homebrew/opt/libusb/lib but the code is using the old way of doing things '/opt/local/lib/'.

Bit new to MacOS so not sure what is best way to set this to look for correct path (assuming that is the problem). 

`(CoralAcc) brian@Brians-Air pycoral % python3 examples/classify_image.py \
--model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
--labels test_data/inat_bird_labels.txt \
--input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 40, in <module>
    from pycoral.utils.edgetpu import make_interpreter
  File "/Users/brian/opt/anaconda3/envs/CoralAcc/lib/python3.8/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: dlopen(/Users/brian/opt/anaconda3/envs/CoralAcc/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so, 2): Library not loaded: /opt/local/lib/libusb-1.0.0.dylib
  Referenced from: /usr/local/lib/libedgetpu.1.dylib
  Reason: image not found`

hjonnala, 2021-09-16:
@cku239 have you installed edgetpu runtime? 

can you please paste the output of this command:

`python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'`

and which coral device are you trying to use?

cku239, 2021-09-16:
I'm trying to get USB accelerator version to work.

How would I know if the runtime has been installed? I've followed the instructions and got no errors until running the test at end of the instructions. I ran the instructions to install edgetpu got the message about whether the USB accelarator should run outside its operating temp.
i Iave two folders libedgetpu and libedgetpu_bin

This is what I get from running that command:-
>>> from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/brian/coral/pycoral/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ModuleNotFoundError: No module named 'pycoral.pybind'

>>> from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/brian/coral/pycoral/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ModuleNotFoundError: No module named 'pycoral.pybind'

hjonnala, 2021-09-16:
Have you installed [pycoral](https://coral.ai/docs/accelerator/get-started/#pycoral-on-mac-win)? can you please attach `pip freeze` output in text file.

Please try these commands on terminal and paste the output here:

`python3 -c "from ctypes.util import find_library; print(find_library(\"edgetpu\"))"`

After connecting USB accelerator: 

`lsusb -d 1a6e:089a`

cku239, 2021-09-16:
I tried reinstalling everything to make sure didn't make a mistake. Saw this message
"Installing Edge TPU runtime library [/usr/local/lib]...
Installing Edge TPU runtime library symlink [/usr/local/lib]..."

(CoralAcc) brian@Brians-Air ~ % python3 -c "from ctypes.util import find_library; print(find_library(\"edgetpu\"))"
None
(CoralAcc) brian@Brians-Air ~ % 

lsusb output:-
(CoralAcc) brian@Brians-Air ~ % lsusb -d 1a6e:089a    
(CoralAcc) brian@Brians-Air ~ % 

output of pip freeze:-
[pipfreezeoutput.txt](https://github.com/google-coral/pycoral/files/7181232/pipfreezeoutput.txt)

hjonnala, 2021-09-16:
similar to this ` python3 -c "from ctypes.util import find_library; print(find_library("edgetpu"))"` can you paste the output of this command
  `python3 -c "from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())"`

hjonnala, 2021-09-17:
> > > > from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > File "/Users/brian/coral/pycoral/pycoral/utils/edgetpu.py", line 24, in 
> > > > from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
> > > > ModuleNotFoundError: No module named 'pycoral.pybind'

you will get htis error if you are trying inside `coral/pycoral` directory. please try outside of it.

cku239, 2021-09-17:
(base) brian@Brians-Air edgetpu_runtime % python3 -c "from ctypes.util import find_library; print(find_library("edgetpu"))"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'edgetpu' is not defined

(base) brian@Brians-Air edgetpu_runtime % python3 -c "from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/brian/opt/anaconda3/lib/python3.8/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: dlopen(/Users/brian/opt/anaconda3/lib/python3.8/site-packages/pycoral/pybind/_pywrap_coral.cpython-38-darwin.so, 2): Library not loaded: /opt/local/lib/libusb-1.0.0.dylib
  Referenced from: /usr/local/lib/libedgetpu.1.dylib
  Reason: image not found

btw on the Mac M1 homebrew puts /opt/local/lib/libusb-1.0.0.dylib into /opt/homebrew/opt/libusb-1.0.0.dylib.

hjonnala, 2021-09-17:
can you try coping `/opt/homebrew/opt/libusb-1.0.0.dylib `and `/opt/homebrew/opt/libusb-1.dylib` to `/opt/local/lib/`

hjonnala, 2021-09-28:
@cku239 does your issue got resolved?