google-coral/pycoral #47: Dual Edge TPU + two_models_inference.py Example (Two TPUs not detected)

gavan1, 2021-09-17:
### Description

Dual Edge TPU installed + pycoral
No problem inferencing with a single model.

**python3 examples/two_models_inference.py   --classification_model test_data/mobilenet_v2_1.0_224_quant_edgetpu.tflite    --detection_model     test_data/ssd_mobilenet_v2_face_quant_postprocess_edgetpu.tflite   --image test_data/parrot.jpg**
```
Traceback (most recent call last):
  File "examples/two_models_inference.py", line 204, in <module>
    main()
  File "examples/two_models_inference.py", line 185, in main
    raise RuntimeError('This demo requires at least two Edge TPU available.')
RuntimeError: This demo requires at least two Edge TPU available.
```
**lspci -vvv**
```
00:00.0 PCI bridge: Fuzhou Rockchip Electronics Co., Ltd Device 3566 (rev 01) (prog-if 00 [Normal decode])
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 106
	Bus: primary=00, secondary=01, subordinate=ff, sec-latency=0
	I/O behind bridge: 0000f000-00000fff
	Memory behind bridge: fff00000-000fffff
	Prefetchable memory behind bridge: 0000000000900000-0000000000afffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	[virtual] Expansion ROM at 300b00000 [disabled] [size=64K]
	BridgeCtl: Parity- SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: <access denied>
	Kernel driver in use: pcieport

01:00.0 System peripheral: Device 1ac1:089a (prog-if ff)
	Subsystem: Device 1ac1:089a
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 105
	Region 0: Memory at 300a00000 (64-bit, prefetchable) [size=16K]
	Region 2: Memory at 300900000 (64-bit, prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: apex
	Kernel modules: apex
```

Advise on correct command and what output will look like if both TPU's were detected.

### Operating System

Ubuntu

### Coral Device

M.2 Accelerator with dual Edge TPU

hjonnala, 2021-09-17:
Hi Can you share output of below snippet:

```
hemanth@hemanth-glaptop:~$ python3
Python 3.9.7 (default, Sep  3 2021, 06:18:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pycoral.pybind._pywrap_coral import ListEdgeTpus as list_edge_tpus
>>> list_edge_tpus()
[{'type': 'usb', 'path': '/sys/bus/usb/devices/2-1'}]
>>> 
```

magic-blue-smoke, 2021-09-22:
> Advise on correct command and what output will look like if both TPU's were detected.

Some logs collected here: https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/3
You might have issue with only one PCIe routed to m.2 slot. I've made write up on this: https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter

hjonnala, 2021-09-27:
Thanks @magic-blue-smoke 

@gavan1 if both tpus detected you would be to see two device with id 089a

```
# lspci -nn | grep Coral
04:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
05:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
```

closed 2021-09-27 by hjonnala

gavan1, 2023-10-05:
@magic-blue-smoke I have your B+M M.2 adapter running the dual tpu on a windows machine... (I painfully managed to successfully installed needed dependency  following the official Coral documentation)

I am don't have the luxury to use Linux on the hardware with the dual tpu, are there any commands or examples scripts I can run to verify access to both TPU's?

magic-blue-smoke, 2023-10-06:
> @magic-blue-smoke I have your B+M M.2 adapter running the dual tpu on a windows machine... (I painfully managed to successfully installed needed dependency following the official Coral [...]

@gavan1 to see if it's there, you can use Device Manager to check if both Coral TPUs are present. If it shows up, you should be able to run actual scripts. However, I'm not the best person to help with this, as I've never used or tested it under Windows