google-coral/pycoral #48: Error ValueError: Failed to load delegate from edgetpu.dll And No module named 'pycoral.pybind'

staebchen0, 2021-09-19:
### Description

I have been trying to get the google coral usb accelerator running under windwos 10 for days.

Under Visual Studio 2019 i have 

- python 3.7.8 64bit and  
- C++ App VS 2019 Installed

- edgetpu_runtime_20210726.zip extracted
- Install.bat executed

**Install.bat Result:**
------------------------------------------------------------------------------------------------------
```
Installing UsbDk
Installing Windows drivers
Microsoft-PnP-Hilfsprogramm

Treiberpaket wird hinzugefügt:  coral.inf
Das Treiberpaket wurde erfolgreich hinzugefügt. (Ist bereits im System vorhanden)
Veröffentlichter Name:         oem75.inf

Treiberpaket wird hinzugefügt:  Coral_USB_Accelerator.inf
Das Treiberpaket wurde erfolgreich hinzugefügt. (Ist bereits im System vorhanden)
Veröffentlichter Name:         oem76.inf

Treiberpaket wird hinzugefügt:  Coral_USB_Accelerator_(DFU).inf
Das Treiberpaket wurde erfolgreich hinzugefügt. (Ist bereits im System vorhanden)
Veröffentlichter Name:         oem77.inf
Treiberpaket auf dem Gerät installiert: USB\VID_1A6E&PID_089A\5&32865703&0&17
Treiberpaket auf dem Gerät installiert: USB\VID_1A6E&PID_089A\5&32865703&0&18
Das Treiberpaket auf dem Gerät ist auf dem neuesten Stand: USB\VID_1A6E&PID_089A\5&32865703&0&19

Treiberpakete insgesamt:  3
Hinzugefügte Treiberpakete:  3
Installing performance counters

```
```
Info: Anbieter {aaa5bf9e-c44b-4177-af65-d3a06ba45fe7}, der in C:\Users\anja-\Anja_Programme\AnjaCoral\envCoralPy8\edgetpu_runtime\third_party\coral_accelerator_windows\coral.man definiert ist, ist bereits im Systemrepository installiert.
Info: Die Leistungsindikatoren wurden erfolgreich in C:\Users\anja-\Anja_Programme\AnjaCoral\envCoralPy8\edgetpu_runtime\third_party\coral_accelerator_windows\coral.man installiert.Copying edgetpu and libusb to System32
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
Install complete
Drücken Sie eine beliebige Taste . . .
```
------------------------------------------------------------------------------------------------------------

- Restart Windows
- Google coral usb accelerator connect to PC

Then I installed the following packages:
Edge-TPU-Python-API:
`pip install https://dl.google.com/coral/edgetpu_api/edgetpu-2.14.0-cp37-cp37m-win_amd64.whl`

tflite runtime 
`pip install https://github.com/google-coral/pycoral/releases/download/v2.0.0/tflite_runtime-2.5.0.post1-cp37-cp37m-win_amd64.whl`

Pycoral installieren
`pip install https://github.com/google-coral/pycoral/releases/download/v2.0.0/pycoral-2.0.0-cp37-cp37m-win_amd64.whl`

**

- test with classify_image.py:

**
`python examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg`

```
[... 233 lines cut ...]
    invoke_with_membuffer(interpreter_handle, input_data.ctypes.data,
                          expected_input_size)
  else:
    raise TypeError('input data type is not supported.')

```

then i get the error Code line:
`from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version - No module named 'pycoral.pybind'`

### Issue Type

Build/Install

### Operating System

Windows 10

### Coral Device

USB Accelerator

### Programming Language

Python 3.7

hjonnala, 2021-09-19:
please check this comment https://github.com/google-coral/libedgetpu/issues/29#issuecomment-909207700.

Thanks

staebchen0, 2021-09-20:
Thank you for the hint :-)
now I have installed the other runtime

edgetpu_runtime_20210119
>>> import tflite_runtime
>>> print (tflite_runtime.__version__)
2.5.0.post1
>>> import edgetpu 
>>> print (edgetpu.__version__)
2.12.1
>>> 

**Test works now:**
```
anja-@LAPTOP-7FIIFTGI MINGW64 ~/Anja_Programme/AnjaCoral/coral/pycoral (master)
$ python examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
----INFERENCE TIME----
Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
13.3ms
4.5ms
4.7ms
4.5ms
4.3ms
-------RESULTS--------
Ara macao (Scarlet Macaw): 0.75781
(envCoralPy8)
anja-@LAPTOP-7FIIFTGI MINGW64 ~/Anja_Programme/AnjaCoral/coral/pycoral (master)
$
```

now I wanted to do a test with my own model

Beispiel:
```
from edgetpu.classification.engine import ClassificationEngine
from PIL import Image
import cv2
import re
import os

# the TFLite converted to be used with edgetpu
modelPath = './model/model_edgetpu.tflite'

# The path to labels.txt that was downloaded with your model
labelPath = './model/labels.txt'

# This function parses the labels.txt and puts it in a python dictionary
def loadLabels(labelPath):
    p = re.compile(r'\s*(\d+)(.+)')
    with open(labelPath, 'r', encoding='utf-8') as labelFile:
        lines = (p.match(line).groups() for line in labelFile.readlines())
        return {int(num): text.strip() for num, text in lines}

# This function takes in a PIL Image and the ClassificationEngine
def classifyImage(image, engine):
    # Classify and ouptut inference
    classifications = engine.ClassifyWithImage(image)
    return classifications

def main():
    # Load your model onto your Coral Edgetpu
    engine = ClassificationEngine(modelPath)
    labels = loadLabels(labelPath)

    cap = cv2.VideoCapture(0)
    while cap.isOpened():
        ret, frame = cap.read()
        if not ret:
            break

        # Format the image into a PIL Image so its compatable with Edge TPU
        cv2_im = frame
        pil_im = Image.fromarray(cv2_im)

        # Resize and flip image so its a square and matches training
        pil_im.resize((224, 224))
        pil_im.transpose(Image.FLIP_LEFT_RIGHT)

        # Classify and display image
        results = classifyImage(pil_im, engine)
        cv2.imshow('frame', cv2_im)
        print(results)
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break

    cap.release()
    cv2.destroyAllWindows()

if __name__ == '__main__':
    main()

```

Error:
In line :from edgetpu.classification.engine import ClassificationEngine
No module named '_edgetpu_cpp_wrapper'

Although the edgetpu_cpp_wrapper.py script in the
AnjaCoral \ envCoralPy8 \ Lib \ site-packages \ edgetpu \ swig 
is available

hjonnala, 2021-09-20:
edgetpu api is deprecated. It has been modified to libcoral api for c++ and pycoral api for python.

Please refer to [examples-camera](https://github.com/google-coral/examples-camera) for opencv example scripts.

staebchen0, 2021-09-20:
okay i'm trying.
Can you still tell me how to install the compiler on Windows?
Only the commands for Linux are shown on the info page

hjonnala, 2021-09-20:
sorry, compiler does not work on windows. Please use [web compiler](https://colab.research.google.com/github/google-coral/tutorials/blob/master/compile_for_edgetpu.ipynb).

staebchen0, 2021-09-20:
okay thank you very much! compiling worked.

now I get detect.py when running the script
the error:
`list index out of range`
code line:
`objs = get_objects (interpreter, args.threshold) [: args.top_k]`
I have 4 classes in the model
label.txt values:
0 0_none
1 1_cat
2 2_mouse
3 3_other

hjonnala, 2021-09-20:
you might not be getting 4 values from `get_objects`. Please check how many objects you are getting.

staebchen0, 2021-09-20:
I think it's because I have a classification model and not a detection model!

is there also an example of a classification model?

under
examples-camera / opencv /
is just the detect.py example

hjonnala, 2021-09-20:
yes, we have only object detection with opencv.

staebchen0, 2021-09-20:
okay thanks for your help :-)

closed 2021-09-20 by staebchen0