google-coral/pycoral #54: How to run yolov5(pytroch) in google-TPU Dev Board

luckywangchenxi, 2021-10-15:
### Description

Hi, I am a student and I want to deploy yolov5 to google-TPU Dev Board.
I would like to ask how to run https://github.com/ultralytics/yolov5 on google-TPU Dev Board (https://coral.ai/products/dev-board) .
From what I have checked so far
1. Need to convert pytorch to TensorFlow via onnx.
2.  Convert tflite model to tflite-tpu model.
3. Then deploy to google-TPU development board.
If you can describe the exact process, it would be greatly appreciated, thanks!

### Issue Type

Build/Install

### Coral Device

Dev Board

### Programming Language

Python 3.7

hjonnala, 2021-11-18:
Feel free to reopen if you have questions.

closed 2021-11-18 by hjonnala

atbdeveloper, 2024-03-31:
Anyone knows about the process for run Yolo V9 inside google coral board?

CarlosNacher, 2024-04-15:
> ### Description
> Hi, I am a student and I want to deploy yolov5 to google-TPU Dev Board. I would like to ask how to run https://github.com/ultralytics/yolov5 on google-TPU Dev Board (https://coral.ai/products/dev-board) . From what I have checked so far
> 
> 1. Need to convert pytorch to TensorFlow via onnx.
> 2. Convert tflite model to tflite-tpu model.
> 3. Then deploy to google-TPU development board.
>    If you can describe the exact process, it would be greatly appreciated, thanks!
> 
> Click to expand!

Hey @luckywangchenxi ! I am trying to convrt pytorch to tensorflow but it is proving frustrating, could you please help me?

I have converted from pytorch to onnx, which is simple with `torch.onnx.export()` method. But the step to convert onnx to tensorflow, I don't find any library that achieves it. I found https://github.com/onnx/onnx-tensorflow but it has not been mantained for 2 years and when I try it, a lot of dependency errors raise. I would really appreciate your help, thank you in advance!

luckywangchenxi, 2024-04-16:
> Hey @luckywangchenxi ! I am trying to convrt pytorch to tensorflow but it is proving frustrating, could you please help me?
>
> I have converted from pytorch to onnx, which is simple with [...]

Sorry, I am no longer working on this project. I wish you all the best.

CarlosNacher, 2024-04-16:
@luckywangchenxi don't worry, thank you. As help for others with the same problem, **I have been able to convert onnx to tensorflow with the following repo**: https://github.com/PINTO0309/onnx2tf