google-coral/pycoral #60: tflite runtime conflicts with tensorflow version

corvitacvision, 2021-11-24:
### Description

```
from tensorflow.lite.python.interpreter_wrapper import _pywrap_tensorflow_interpreter_wrapper as _interpreter_wrapper
ImportError: generic_type: type "InterpreterWrapper" is already registered!
```

### Issue Type

Build/Install

### Operating System

Ubuntu

### Coral Device

USB Accelerator

### Programming Language

Python 3.6

hjonnala, 2021-11-24:
@corvitacvision can you please check this https://github.com/google-coral/pycoral/issues/57#issuecomment-949997068 issue to fix it.

corvitacvision, 2021-11-25:
@hjonnala Thank you for your reply. I have solved the issue by downgrading tensorflow version from 2.6 to 2.4 and now it runs without any runtime error.

closed 2021-11-25 by corvitacvision