google-coral/pycoral #61: Error pycoral

AnandaRauf, 2021-11-25:
### Description

Hello, i found bug in pycoral and edgetpu.

Error Message :
 from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: DLL load failed while importing _pywrap_coral: The specified module could not be found.

### Issue Type

Bug

### Operating System

Windows 10

### Coral Device

M.2 Accelerator with dual Edge TPU

### Programming Language

Python 3.7

manoj7410, 2021-11-25:
Is this duplicate of https://github.com/google-coral/pycoral/issues/62 ?

hjonnala, 2021-11-29:
Please uninstall the pycoral and [install](https://coral.ai/docs/accelerator/get-started#pycoral-on-mac-win) with the command below. 

`python3 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0`

Please share the output of below snippet if issue still exists. Thanks!

```
Python 3.9.7 (default, Sep  3 2021, 06:18:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pycoral
>>> pycoral.__version__
'2.0.0'
>>> 

```

hjonnala, 2021-12-09:
Feel free to reopen if you are having any issues with pycoral installation. Thanks

closed 2021-12-09 by hjonnala