google-coral/pycoral #62: BUG Edge TPU and ERROR Install Pycoral for Python 3.7

AnandaRauf, 2021-11-25:
### Description

Hello, i found bug in edge TPU and Error install Pycoral for python version 3.7 :
Traceback (most recent call last):
  File "C:\Users\Ananda Rauf Maududi\Downloads\Sistem Informasi Deteksi Benda Berbahaya(SIni Dek Benar) PC\sinidek.py", line 1, in <module>
    from edgetpu.classification.engine import ClassificationEngine
  File "C:\Users\Ananda Rauf Maududi\AppData\Local\Programs\Python\Python37\lib\site-packages\edgetpu\__init__.py", line 15, in <module>
    import edgetpu.basic.edgetpu_utils as utils
  File "C:\Users\Ananda Rauf Maududi\AppData\Local\Programs\Python\Python37\lib\site-packages\edgetpu\basic\edgetpu_utils.py", line 15, in <module>
    import edgetpu.swig.edgetpu_cpp_wrapper as wrapper
  File "C:\Users\Ananda Rauf Maududi\AppData\Local\Programs\Python\Python37\lib\site-packages\edgetpu\swig\edgetpu_cpp_wrapper.py", line 28, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "C:\Users\Ananda Rauf Maududi\AppData\Local\Programs\Python\Python37\lib\site-packages\edgetpu\swig\edgetpu_cpp_wrapper.py", line 24, in swig_import_helper
    _mod = imp.load_module('_edgetpu_cpp_wrapper', fp, pathname, description)
  File "C:\Users\Ananda Rauf Maududi\AppData\Local\Programs\Python\Python37\lib\imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "C:\Users\Ananda Rauf Maududi\AppData\Local\Programs\Python\Python37\lib\imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: DLL load failed: The specified module could not be found.

Error Install Pycoral Py V 3.7.4

### Issue Type

Build/Install

### Operating System

Windows 10

### Coral Device

M.2 Accelerator with dual Edge TPU

manoj7410, 2021-11-25:
Does this helphttps://github.com/google-coral/pycoral/issues/32#issuecomment-877944752 ?

hjonnala, 2021-11-30:
please uninstall edgetpu since it is deprecated. And install pycoral as described in this comment: https://github.com/google-coral/pycoral/issues/61#issuecomment-981731720..

Feel free to add comments to this issue if they are any questions: https://github.com/google-coral/pycoral/issues/61

Thanks

closed 2021-11-30 by hjonnala