google-coral/pycoral #64: Pretrained models license

StephenKThung, 2021-12-09:
### Description

Hi all, my group is interested in working with some of the [pre-trained models from the coral model zoo](https://coral.ai/models/object-detection/)

We saw that they are marked as "for demonstration purposes only" but have been unable to find a more specific license. Do they simply inherit the licenses from COCO and this repo or are there additional considerations?

Thanks!

### Issue Type

Support

hjonnala, 2021-12-09:
Hi @StephenKThung  most of our models and pycoral repo are Apache licensed.

https://github.com/google-coral/pycoral/blob/master/LICENSE
https://github.com/google-coral/test_data/blob/master/LICENSE

StephenKThung, 2021-12-10:
Thanks for the information!

closed 2021-12-10 by StephenKThung