google-coral/pycoral #66: Feature Request: Environment Variable override for Device Selection

chunky, 2021-12-15:
### Description

I'm currently writing some code to experiment with multiple coral devices, and it would be immensely helpful if I had a mechanism to inject the name of the device I want to use out of band [ie, without adding it directly to any code paths].

For now, I've added an environment variable lookup to my code:

```python
from pycoral.utils import edgetpu
edge_tpus_available = edgetpu.list_edge_tpus()
dev = os.getenv("EDGETPU_DEVICE", edge_tpus_available[0]["type"])
print("Coral TPUs available: {}, using {}".format(edge_tpus_available, dev))
self.interpreter = edgetpu.make_interpreter(checkpoint, device=dev)
```

I intuit it may even better to add such a thing to libedgetpu itself, but I'm using Python, so this is where I left this request. If y'all think this is a good idea, but would be better posted to libedgetpu, then please say so and I can open an issue there.

### Issue Type

Feature Request

### Operating System

Windows 10, Mendel Linux, Linux, Ubuntu, Mac OS

### Coral Device

USB Accelerator, Mini PCIe, M.2 Accelerator A+E, M.2 Accelerator B+M, M.2 Accelerator with dual Edge TPU

### Programming Language

Python 3.5, Python 3.6, Python 3.7, Python 3.8, Python 3.9

hjonnala, 2021-12-15:
Hi @chunky please check these sample functions to get the num devices and make interpreters using those devices:

https://github.com/google-coral/pycoral/blob/master/examples/model_pipelining_classify_image.py#L49
https://github.com/google-coral/pycoral/blob/master/examples/model_pipelining_classify_image.py#L77

If you want to pass them as params to the script,you can add  an [argumnet](https://github.com/google-coral/pycoral/blob/master/examples/model_pipelining_classify_image.py#L102) and pass the device name to it.

chunky, 2021-12-16:
Sure, and my code above shows how I am already able to select the device using code. This feature request is because sometimes it would be helpful to have a mechanism to specify the device in a manner that doesn't need an extra parameter to be passed around. For example:

* Specify device in tools developed by someone else where they didn't provide an option. eg [the minigo example on coral.ai](https://github.com/tensorflow/minigo)
* Control device from a harness shell script
* Useful override during debugging