google-coral/pycoral #67: PyCoral on RPi Bullseye 32Bit (Python3.9)

TimPearson, 2021-12-23:
### Description

Failed to run examples (classify_image.py) on latest install RPi4 operating system (11 aka Bullseye) with error No module named 'pycoral.adapters'.

I suspect this is because it comes with python 3.9. (alternatively I have done something stupid and am an idiot)

I tried downgrading to 3.8 but got into version hell with NUMPy etc. 

### Issue Type

Bug

### Operating System

Linux

### Coral Device

USB Accelerator

### Programming Language

Python 3.9

### Relevant Log Output

```shell
ModuleNotFoundError: No module named 'pycoral.adapters'
```

hjonnala, 2021-12-23:
Hello @TimPearson Can you please share how did you install and the pycoarl and output of below snippet.

```
Python 3.9.7 (default, Sep  3 2021, 06:18:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pycoral
>>> pycoral.__version__
'2.0.0'
>>> 

```

TimPearson, 2021-12-23:
Hello @hjonnala sure but i will have to recreate system as have since messed it up trying to find a fix, will try and get that done tomorrow.

TimPearson, 2021-12-24:
From clean system:
sudo apt install git
sudo apt install python3-virtualenv
sudo apt-get update
echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
sudo apt-get install libedgetpu1-std
sudo apt-get install python3-coral
mkdir coral
cd coral
git clone https://github.com/google-coral/pycoral.git
cd pycoral
virtualenv pycoralvenv
source pycoralvenv/bin/activate
bash examples/install_requirements.sh classify_image.py
pip3 install pillow
pip3 install numpy
sudo apt-get install libatlas-base-dev
python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg

Then I get :
(pycoralvenv) pi@raspberrypi:~/coral/pycoral $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pycoral
>>> pycoral.__version__
'2.0.0'
>>>

hjonnala, 2021-12-24:
can you try uninstall` python3-coral` and install ` python3-pycoral` (https://coral.ai/software/#debian-packages)

TimPearson, 2021-12-24:
i was still getting errors. Turned out that simply giving myself root rights via sudo solved the problem - i have no idea why? so:

sudo python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg

gives 👎 
----INFERENCE TIME----
Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
129.3ms
15.2ms
15.9ms
16.0ms
15.7ms
-------RESULTS--------
Ara macao (Scarlet Macaw): 0.75781

closed 2021-12-24 by TimPearson

crittercam, 2023-12-27:
Just in case anybody runs into this, I got it working with Bullseye.  Here are the details: 

pi@raspberrypi:~/coral/pycoral $ python --version
Python 3.9.2

pi@raspberrypi:~/coral/pycoral $ pip list
Package           Version
----------------- -----------
arandr            0.1.10
astroid           2.5.1
asttokens         2.0.4
automationhat     0.2.0
beautifulsoup4    4.9.3
blinker           1.4
blinkt            0.1.2
buttonshim        0.0.2
Cap1xxx           0.1.3
certifi           2020.6.20
chardet           4.0.0
click             7.1.2
colorama          0.4.4
colorzero         1.1
cryptography      3.3.2
cupshelpers       1.0
dbus-python       1.2.16
distro            1.5.0
docutils          0.16
drumhat           0.1.0
envirophat        1.0.0
ExplorerHAT       0.4.2
Flask             1.1.2
fourletterphat    0.1.0
gpiozero          1.6.2
html5lib          1.1
idna              2.10
isort             5.6.4
itsdangerous      1.1.0
jedi              0.18.0
Jinja2            2.11.3
lazy-object-proxy 0.0.0
logilab-common    1.8.1
lxml              4.6.3
MarkupSafe        1.1.1
mccabe            0.6.1
microdotphat      0.2.1
mote              0.0.4
motephat          0.0.3
mypy              0.812
mypy-extensions   0.4.3
numpy             1.19.5
oauthlib          3.1.0
opencv-python     4.6.0.66
opencv-python-headless 4.8.1.78
pantilthat        0.0.7
parso             0.8.1
pexpect           4.8.0
pgzero            1.2
phatbeat          0.1.1
pianohat          0.1.0
picamera2         0.3.12
pidng             4.0.9
piexif            1.1.3
piglow            1.2.5
pigpio            1.78
Pillow            8.1.2
pip               20.3.4
psutil            5.8.0
pycairo           1.16.2
pycoral           2.0.0
pycups            2.0.1
pygame            1.9.6
Pygments          2.7.1
PyGObject         3.38.0
pyinotify         0.9.6
PyJWT             1.7.1
pylint            2.7.2
PyOpenGL          3.1.5
pyOpenSSL         20.0.1
PyQt5             5.15.2
PyQt5-sip         12.8.1
pyserial          3.5b0
pysmbc            1.0.23
python-apt        2.2.1
python-prctl      1.7
rainbowhat        0.1.0
reportlab         3.5.59
requests          2.25.1
requests-oauthlib 1.0.0
responses         0.12.1
roman             2.0.0
RPi.GPIO          0.7.0
RTIMULib          7.2.1
scrollphat        0.0.7
scrollphathd      1.2.1
Send2Trash        1.6.0b1
sense-hat         2.6.0
setuptools        52.0.0
simplejpeg        1.6.4
simplejson        3.17.2
six               1.16.0
skywriter         0.0.7
sn3218            1.2.7
soupsieve         2.2.1
spidev            3.5
ssh-import-id     5.10
tflite-runtime    2.5.0.post1
thonny            4.0.1
toml              0.10.1
touchphat         0.0.1
twython           3.8.2
typed-ast         1.4.2
typing-extensions 3.7.4.3
unicornhathd      0.0.4
urllib3           1.26.5
v4l2-python3      0.3.2
webencodings      0.5.1
Werkzeug          1.0.1
wheel             0.34.2
wrapt             1.12.1

skol101, 2024-04-24:
@crittercam you could have shared your cli commands so that others could repeat.