google-coral/pycoral #68: No module named 'pycoral.pybind._pywrap_coral'

saga0611, 2021-12-25:
### Description

Hi all

I try to run on my raspi the image classifier example with the bird photo and get the error no module named 'pycoral.pybind._pywrap_coral'. Can someone help me please?

I did all the steps of this installation guide:
https://coral.ai/docs/accelerator/get-started

All installations were successfully done.

Info:
-Raspberry Pi 3B
-Python 3.5

Thanks for help!

### Operating System

Linux

### Coral Device

USB Accelerator

### Other Devices

Raspberry Pi 3

hjonnala, 2021-12-25:
Hello @saga0611 pycoral module works with Python 3.6, 3.7, 3.8, or 3.9 versions only. Please upgrade the python version and try the demo.

saga0611, 2021-12-25:
Hi @hjonnala

Thanks for your help!

I installed python 3.8 now but I get the same error. Do you have an other hint?
[image]

hjonnala, 2021-12-25:
Please check the pycoral version got installed.  and try the command as a super user.

`sudo python3.8 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg`

```
Python 3.9.7 (default, Sep  3 2021, 06:18:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pycoral
>>> pycoral.__version__
'2.0.0'
>>> 
```

saga0611, 2021-12-25:
Here the version check of pycoral:

```
pi@eee-00433:~/Desktop/coral/pycoral $ python3.8
Python 3.8.4 (default, Dec 24 2021, 15:23:21) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pycoral
>>> pycoral.__version__
'2.0.0'
```

I tryed with "sudo" but unfortunately without success. Now with "sudo" I get this error:
```
sudo python3.8 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 37, in <module>
    from pycoral.adapters import classify
ModuleNotFoundError: No module named 'pycoral'
```

Do you maybe know why I get this error although the installation of pycoral seems okay?

hjonnala, 2021-12-25:
hmm.. Have you uninstall and  install the pycoral after upgrading the python version?

Can please try to uninstall pycoral `sudo apt-get purge python3-pycoral` and install the pycoral using this command `sudo apt-get install python3-pycoral` and try the demo. 

If this is not working please uninstall pycoral and install it using pip and try the demo. Thanks!

`python3.8 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0`

saga0611, 2021-12-25:
I unistalled pycroal and installed it with `sudo apt-get install python3-pycoral` but the example was still not working (same error as before).

With this command `python3.8 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0` I get this installation error:
```
pi@eee-00433:~/Desktop/coral/pycoral $ python3.8 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://google-coral.github.io/py-repo/
Collecting pycoral~=2.0
  Downloading https://github.com/google-coral/pycoral/releases/download/v2.0.0/pycoral-2.0.0-cp38-cp38-linux_armv7l.whl (326 kB)
     |████████████████████████████████| 326 kB 90 kB/s             
Collecting tflite-runtime==2.5.0.post1
  Downloading https://github.com/google-coral/pycoral/releases/download/v2.0.0/tflite_runtime-2.5.0.post1-cp38-cp38-linux_armv7l.whl (1.3 MB)
     |████████████████████████████████| 1.3 MB 1.4 MB/s            
Requirement already satisfied: numpy>=1.16.0 in /usr/local/lib/python3.8/site-packages (from pycoral~=2.0) (1.21.5)
Collecting Pillow>=4.0.0
  Using cached Pillow-8.4.0.tar.gz (49.4 MB)
  Preparing metadata (setup.py) ... done
Using legacy 'setup.py install' for Pillow, since package 'wheel' is not installed.
Installing collected packages: tflite-runtime, Pillow, pycoral
    Running setup.py install for Pillow ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/local/bin/python3.8 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-cx8rh74u/pillow_4f680e23643940adbb0daf0aeffaf35a/setup.py'"'"'; __file__='"'"'/tmp/pip-install-cx8rh74u/pillow_4f680e23643940adbb0daf0aeffaf35a/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-nodj_fxk/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/pi/.local/include/python3.8/Pillow
         cwd: /tmp/pip-install-cx8rh74u/pillow_4f680e23643940adbb0daf0aeffaf35a/
    Complete output (175 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.8
    creating build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageFilter.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/GimpGradientFile.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/PdfImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/GribStubImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/_binary.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/PngImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageDraw.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/GbrImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageMath.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageMode.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/FliImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/JpegPresets.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/SunImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/PcfFontFile.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/BmpImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/__init__.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/IcoImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageTk.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/PsdImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/GdImageFile.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageTransform.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/FpxImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/GimpPaletteFile.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/PcdImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/Image.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/BlpImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageFile.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/__main__.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/CurImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/PalmImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/TarIO.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/MicImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
    copying src/PIL/ImageStat.py -> build/lib.linux-armv7l-3.8/PIL
[... 109 lines cut ...]
        self.build_extensions()
[... cut ...] ze, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-nodj_fxk/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/pi/.local/include/python3.8/Pillow Check the logs for full command output.
```

Pillow seems to be a problem...

saga0611, 2021-12-26:
Hi @hjonnala 

I set up my raspi new and now it works. I don't know what was wrong with the system before.

Anyway thank you very much for your help!

closed 2021-12-26 by saga0611