google-coral/pycoral #69: NOT FOUND errors

ImanHosseini, 2021-12-29:
### Description

Doing everything as in the tutorial, I get:
```
mendel@bored-apple:~/coral/pycoral$ bash examples/install_requirements.sh
Download all files...
NOT FOUND: mobilenet_v1_1.0_224_quant_embedding_extractor_edgetpu.tflite
NOT FOUND: sunflower.bmp
NOT FOUND: mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite
NOT FOUND: mobilenet_v2_1.0_224_inat_bird_quant.tflite
NOT FOUND: inat_bird_labels.txt
NOT FOUND: parrot.jpg
NOT FOUND: ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite
NOT FOUND: coco_labels.txt
NOT FOUND: grace_hopper.bmp
NOT FOUND: mobilenet_v1_1.0_224_l2norm_quant_edgetpu.tflite
NOT FOUND: cat.bmp
NOT FOUND: pipeline/inception_v3_299_quant_segment_0_of_2_edgetpu.tflite
NOT FOUND: pipeline/inception_v3_299_quant_segment_1_of_2_edgetpu.tflite
NOT FOUND: imagenet_labels.txt
NOT FOUND: parrot.jpg
NOT FOUND: deeplabv3_mnv2_pascal_quant_edgetpu.tflite
NOT FOUND: bird.bmp
NOT FOUND: ssd_mobilenet_v2_coco_quant_no_nms_edgetpu.tflite
NOT FOUND: coco_labels.txt
NOT FOUND: kite_and_cold.jpg
NOT FOUND: mobilenet_v2_1.0_224_quant_edgetpu.tflite
NOT FOUND: ssd_mobilenet_v2_face_quant_postprocess_edgetpu.tflite
NOT FOUND: parrot.jpg
NOT FOUND: movenet_single_pose_lightning_ptq_edgetpu.tflite
NOT FOUND: squat.bmp
```

### Issue Type

Bug, Support, Documentation Bug

### Operating System

Mendel Linux

### Coral Device

Dev Board Mini

### Programming Language

Python 3.7, Python 3.8, Python 3.9

manoj7410, 2021-12-29:
@ImanHosseini Just tried on my machine: 
bash examples/install_requirements.sh 
Download all files...
DOWNLOAD: mobilenet_v1_1.0_224_quant_embedding_extractor_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   191  100   191    0     0   1417      0 --:--:-- --:--:-- --:--:--  1425
100 3404k  100 3404k    0     0  4065k      0 --:--:-- --:--:-- --:--:-- 4065k
DOWNLOAD: sunflower.bmp
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   143  100   143    0     0   1482      0 --:--:-- --:--:-- --:--:--  1489
100  510k  100  510k    0     0  1322k      0 --:--:-- --:--:-- --:--:-- 1322k
DOWNLOAD: mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   181  100   181    0     0   1707      0 --:--:-- --:--:-- --:--:--  1723
100 4197k  100 4197k    0     0  5296k      0 --:--:-- --:--:-- --:--:-- 9271k
DOWNLOAD: mobilenet_v2_1.0_224_inat_bird_quant.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   173  100   173    0     0   1566      0 --:--:-- --:--:-- --:--:--  1558
100 3495k  100 3495k    0     0  4628k      0 --:--:-- --:--:-- --:--:-- 4628k
DOWNLOAD: inat_bird_labels.txt
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   150  100   150    0     0   1265      0 --:--:-- --:--:-- --:--:--  1271
100 37146  100 37146    0     0   130k      0 --:--:-- --:--:-- --:--:--  130k
DOWNLOAD: parrot.jpg
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   140  100   140    0     0   1253      0 --:--:-- --:--:-- --:--:--  1261
100  582k  100  582k    0     0  1146k      0 --:--:-- --:--:-- --:--:-- 1146k
DOWNLOAD: ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   184  100   184    0     0   1646      0 --:--:-- --:--:-- --:--:--  1657
100 6749k  100 6749k    0     0  6279k      0  0:00:01  0:00:01 --:--:-- 9546k
DOWNLOAD: coco_labels.txt
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   145  100   145    0     0   1455      0 --:--:-- --:--:-- --:--:--  1464
100   661  100   661    0     0   2453      0 --:--:-- --:--:-- --:--:--  322k
DOWNLOAD: grace_hopper.bmp
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   146  100   146    0     0   1251      0 --:--:-- --:--:-- --:--:--  1258
100  900k  100  900k    0     0  1962k      0 --:--:-- --:--:-- --:--:-- 1962k
DOWNLOAD: mobilenet_v1_1.0_224_l2norm_quant_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   178  100   178    0     0   1555      0 --:--:-- --:--:-- --:--:--  1547
100 5447k  100 5447k    0     0  6155k      0 --:--:-- --:--:-- --:--:-- 6155k
DOWNLOAD: cat.bmp
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   137  100   137    0     0   1415      0 --:--:-- --:--:-- --:--:--  1427
100  511k  100  511k    0     0  1366k      0 --:--:-- --:--:-- --:--:-- 1366k
DOWNLOAD: pipeline/inception_v3_299_quant_segment_0_of_2_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   191  100   191    0     0    163      0  0:00:01  0:00:01 --:--:--   163
100 11.9M  100 11.9M    0     0  3765k      0  0:00:03  0:00:03 --:--:-- 6248k
DOWNLOAD: pipeline/inception_v3_299_quant_segment_1_of_2_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   191  100   191    0     0   1029      0 --:--:-- --:--:-- --:--:--  1032
100 12.1M  100 12.1M    0     0  6639k      0  0:00:01  0:00:01 --:--:-- 7861k
DOWNLOAD: imagenet_labels.txt
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
[... 35 lines cut ...]
                                 Dload  Upload   Total   Spent    Left  Speed
100   171  100   171    0     0   1570      0 --:--:-- --:--:-- --:--:--  1583
100 4182k  100 4182k    0     0  5268k      0 --:--:-- --:--:-- --:--:-- 6597k
DOWNLOAD: ssd_mobilenet_v2_face_quant_postprocess_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   184  100   184    0     0   1729      0 --:--:-- --:--:-- --:--:--  1735
[... cut ...]
100   178  100   178    0     0     20      0  0:00:08  0:00:08 --:--:--    39
100 3014k  100 3014k    0     0   327k      0  0:00:09  0:00:09 --:--:-- 11.1M
DOWNLOAD: squat.bmp
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   139  100   139    0     0   1314      0 --:--:-- --:--:-- --:--:--  1323
100  893k  100  893k    0     0  2041k      0 --:--:-- --:--:-- --:--:-- 2041k

I am able to download the test data. Can you please check you internet connection?

ImanHosseini, 2021-12-29:
Internet connection was ok (I'd just cloned the repo), but since then I unplugged it _unsafely_ and now it doesn't show up in 'mdt devices' anymore. The light goes green, but doesn't show in 'mdt devices' also doesn't show under '/dev/tty*' (even if using GPIO). Is it possible that it has bricked?

hjonnala, 2021-12-29:
@ImanHosseini please try to re flash the dev board mini using [fast boot](https://coral.ai/docs/dev-board-mini/reflash/#force-boot-into-fastboot-mode) and try the tutorial.

Munna-Manoj, 2022-01-17:
ii  libedgetpu1-std:arm64                      16.0                                  arm64        Support library for Edge TPU
ii  python3-pycoral                            2.0.0                                 arm64        Coral Python API
>>> import pycoral 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pycoral'

Even after installing pycoral and necessary dependencies, I am not able to import pycoral. I tried many other suggestions from internet including from coral.ai but the issue is still the same.

My Ubuntu version:
Distributor ID:	Ubuntu
Description:	Ubuntu 21.10
Release:	21.10

Hardware: Raspberry Pi 4(64-bit)

Can you give me necessary suggestions and help to solve this issue?

ImanHosseini, 2022-10-07:
I reflashed and fixed.

closed 2022-10-07 by ImanHosseini