google-coral/pycoral #70: New, looking for way to detect and then classify in a simple file....

Swiftnesses, 2022-01-19:
### Description

Hello,

I've just downloaded pycoral and am really enjoying learning this new tool.

Essentially, I'm trying to combine the detected and classify examples to:

Look at an image (webcam jpg capture 1920x1080) and firstly identify if there are 'birds' present. If there are, I'd like to somehow utilise the bbox coordinates and run them though MobileNet V2 to classify the bird type.

I'd be really greatful if someone could get me started, I've researched for hours how to combine the functionality without luck...

Thank you.

### Issue Type

Support

### Operating System

Linux

### Coral Device

M.2 Accelerator A+E

### Programming Language

Python 3.9

Namburger, 2022-01-19:
@Swiftnesses maybe you can take a look at manufacturing demo? 
https://github.com/google-coral/demo-manufacturing

The right side does a detection on the box and then run it through and classifier to see if the image in the bbox is a good or bad apple :)

Swiftnesses, 2022-01-19:
Hi @Namburger - I'll take a look but as a beginner, my first impression is that I'll not stand a chance understanding it!

I'm currently trying to use the box coordinates to resize the image, frustrating... :(