google-coral/pycoral #72: failed to load delegate - aarch64, Debian

tom-gall, 2022-02-10:
### Description

Trying to run the initial example:

tgall@viper:~/coral/pycoral-examples$ python3 examples/classify_image.py  --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "/home/tgall/.local/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/home/tgall/.local/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 121, in <module>
    main()
  File "examples/classify_image.py", line 71, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/home/tgall/.local/lib/python3.7/site-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/home/tgall/.local/lib/python3.7/site-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/home/tgall/.local/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 163, in load_delegate
    library, str(e)))
ValueError: Failed to load delegate from libedgetpu.so.1

-------

applying some of the common things to looking at:

tgall@viper:~/coral/pycoral-examples$ python3 -c "from ctypes.util import find_library; print(find_library(\"edgetpu\"))"
libedgetpu.so.1

tgall@viper:~/coral/pycoral-examples$ lsusb -d 1a6e:089a
Bus 001 Device 004: ID 1a6e:089a Global Unichip Corp. 

I had made sure to add my userid into plugdev with sudo usermod -aG plugdev <my user id>

likewise looking at udev rules /etc/udev/rules.d/ I have
tgall@viper:~/coral/pycoral-examples$ cat /etc/udev/rules.d/99-edgetpu-accelerator.rules 
SUBSYSTEM=="usb",ATTRS{idVendor}=="1a6e",GROUP="plugdev"
SUBSYSTEM=="usb",ATTRS{idVendor}=="18d1",GROUP="plugdev"

and from dmesg :
[210360.152395] usb 1-1: new high-speed USB device number 3 using ehci-pci
[210360.307907] usb 1-1: New USB device found, idVendor=1a6e, idProduct=089a, bcdDevice= 1.00
[210360.307908] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[214202.364639] usb 1-1: USB disconnect, device number 3
[214221.585606] usb 1-2: new high-speed USB device number 4 using ehci-pci
[214221.744194] usb 1-2: New USB device found, idVendor=1a6e, idProduct=089a, bcdDevice= 1.00
[214221.744196] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0

No other error messages.

Suggestions?

### Issue Type

Build/Install

### Operating System

Linux

### Coral Device

USB Accelerator

### Programming Language

Python 3.7

hjonnala, 2022-02-10:
Can you please try below commands and after replacing the libedgetpu path in the following commands and try the demo. 

```
sudo chmod +rwx /usr/lib/aarch64-linux-gnu/libedgetpu.so.1.0
sudo chmod +rwx /usr/lib/aarch64-linux-gnu/libedgetpu.so.1
```

tom-gall, 2022-02-11:
Here is my output:

tgall@viper:/usr/include$ python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'
BuildLabel(COMPILER=6.3.0 20170516,DATE=redacted,TIME=redacted), RuntimeVersion(14)

and for libedgetpu I used apt-get

tgall@viper:~$ dpkg -s libedgetpu1-std
Package: libedgetpu1-std
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 1130
Maintainer: Coral <coral-support@google.com>
Architecture: arm64
Multi-Arch: same
Source: libedgetpu
Version: 16.0
Provides: libedgetpu1 (= 16.0)
Depends: libc6, libgcc1, libstdc++6, libusb-1.0-0
Conflicts: libedgetpu1, libedgetpu1-legacy
Description: Support library for Edge TPU
 Support library (standard speed) for the Edge TPU
Homepage: https://coral.ai/

hjonnala, 2022-02-11:
Okay, can you please share the output of following snippet. Thanks
https://github.com/google-coral/edgetpu/issues/491#issuecomment-948206410

hjonnala, 2022-02-25:
@tom-gall  are you still facing the issue?

sharoseali, 2022-05-22:
I am running the code in raspberry PI 4 and facing the issue
```
>>> from pycoral.adapters import classify
>>> from pycoral.adapters import common
>>> from pycoral.utils.dataset import read_label_file
>>> from pycoral.utils.edgetpu import make_interpreter
>>> labels = read_label_file("pycoral/test_data/inat_bird_labels.txt")
>>> interpreter = make_interpreter(["pycoral/test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite"])
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 163, in load_delegate
    library, str(e)))
ValueError: Failed to load delegate from libedgetpu.so.1

```

emepetres, 2023-04-22:
If you are using the usb accelerator, and it its plugged to your computer, setting the following udev rules should work:

```console
foo@bar:~$ lsusb -d 1a6e:089a
Bus 002 Device 004: ID 1a6e:089a Global Unichip Corp.
foo@bar:~$ echo 'SUBSYSTEMS=="usb", ATTRS{idVendor}=="1a6e", ATTRS{idProduct}=="089a", MODE="0664", TAG+="uaccess"' | sudo tee -a /etc/udev/rules.d/71-edgetpu.rules > /dev/null
foo@bar:~$ echo 'SUBSYSTEMS=="usb", ATTRS{idVendor}=="18d1", ATTRS{idProduct}=="9302", MODE="0664", TAG+="uaccess"' | sudo tee -a /etc/udev/rules.d/71-edgetpu.rules > /dev/null
foo@bar:~$ sudo udevadm control --reload-rules && sudo udevadm trigger
```