google-coral/pycoral #73: Run coral demo fail : Failed to load delegate from libedgetpu.so.1

yan-wyb, 2022-02-24:
### Description

1. My board is Khadas VIM3 (Amlogic A311D).
2. I follow [coral get-started doc](https://coral.ai/docs/m2/get-started#pycoral-on-linux) to test  
3. Aarch64 Ubuntu 20.04
4. Kernel version: 4.9. (It work fine with mainline kernel on VIM3)
 
I've tried everything I can search but it doesn't work

### Issue Type

Bug

### Operating System

Ubuntu

### Coral Device

M.2 Accelerator B+M

### Programming Language

Python 3.8, Python 3.9

### Relevant Log Output

```shell
khadas@Khadas:~/coral/pycoral$ python3 examples/classify_image.py \
> --model ~/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
> --input ~/parrot.jpg \
> --labels ~/inat_bird_labels.txt 
[  393.441760@3] apex 0000:01:00.0: Permission checking failed.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 121, in <module>
    main()
  File "examples/classify_image.py", line 71, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1
```

manoj7410, 2022-02-24:
@yan-wyb Can you paste the output of:

lspci -vvv

yan-wyb, 2022-02-24:
```
khadas@Khadas:~$ lspci -vvv
00:00.0 PCI bridge: Synopsys, Inc. DWC_usb3 / PCIe bridge (rev 01) (prog-if 00 [Normal decode])
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 71
        Bus: primary=00, secondary=01, subordinate=ff, sec-latency=0
        I/O behind bridge: 0000f000-00000fff [disabled]
        Memory behind bridge: fff00000-000fffff [disabled]
        Prefetchable memory behind bridge: fca00000-fcbfffff [size=2M]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        Expansion ROM at fcc00000 [virtual] [disabled] [size=64K]
        BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>
        Kernel driver in use: pcieport
lspci: Unable to load libkmod resources: error -12

01:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU (prog-if ff)
        Subsystem: Global Unichip Corp. Coral Edge TPU
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 104
        Region 0: Memory at fcb00000 (64-bit, prefetchable) [size=16K]
        Region 2: Memory at fca00000 (64-bit, prefetchable) [size=1M]
        Capabilities: <access denied>
        Kernel driver in use: apex
```

manoj7410, 2022-02-24:
@yan-wyb The Coral device is detected but I see, "        Capabilities: <access denied>".  I have seen this before butunable to recall the exact issue. I'll try to look in the previous issues and will get back to you.

manoj7410, 2022-02-24:
@yan-wyb Is the correct kernel header installed on your machine ? 
Have you tried with the recent kernel versions?

yan-wyb, 2022-02-25:
I'm sure the kernel headers are installed, the system is maintained by ourselves, I'm from the Khadas team.
The latest 5.16 kernel can be used normally.
4.9  kernel is the version provided by the vendor, which is also very important to us. Many users hope to support this version.

yan-wyb, 2022-03-03:
Any update ? @manoj7410

boredcoder411, 2023-12-31:
I'm having the exact same problem, on a similar system. I've ran the basic example (the one with the parrot) successfully, but I'm trying the movenet out, to no avail