google-coral/pycoral #74: Build for Python 3.8 on Ubuntu 18.04

roflcoopter, 2022-03-01:
### Description

I need to build pycoral and tflite for for Python 3.8 on Ubuntu 18.04.

My application runs on a Jetson Nano (which only supports Ubuntu 18.04) and it requires Python 3.8.
If i install the pre-built wheels for Python 3.8 i get a dependency issue with GLIBC 2.29, which is not available in Ubuntu 18.04.

When i try to build this repo from source, and specify that i want to build for Python 3.8 it defaults to Ubuntu 20.04 which is a no go for me.
I edited the `scripts/build.sh` to only build for python 3.8, and changed python 3.8 to build inside the `ubuntu:18.04` image, but the build only outputs wheels for python 3.6. (eg `tflite_runtime-2.5.0.post1-cp36-cp36m-linux_aarch64.whl`)

Why do the scripts not respect my chosen python version? 

### Issue Type

Build/Install

### Operating System

Ubuntu

### Coral Device

USB Accelerator

### Programming Language

Python 3.8

hjonnala, 2022-03-01:
Hello @roflcoopter can you please check whether your default python version is 3.6 or 3.8. 

https://github.com/google-coral/pycoral/blob/master/Makefile#L17

roflcoopter, 2022-03-01:
Thanks for your response!

Well, the script build.sh builds inside a docker container so my hosts python version wouldn't matter right?

I modified the script to launch an ubuntu:18.04 image as base but that produces python 3.6 binaries.

Tried modifying the Dockerfile as well to include python 3.8 but that results in dependency issues during build.

Is there a way to build without using docker?

hjonnala, 2022-03-01:
We don't have any steps to build with out docker.

I think, Docker image ubuntu:18.04 itself has python 3.6. Can you try to build the ubuntu image with python 3.8 and pass that image here: https://github.com/google-coral/pycoral/blob/master/scripts/build.sh#L44

roflcoopter, 2022-03-02:
Managed to do it!

I modified `scripts/build.sh` to launch an `ubuntu:18.04` image when Python 3.8 was selected.
Then i modified the `libcoral/docker/Dockerfile` and added these lines to install Python 3.8 and all required libraries
```Dockerfile
RUN add-apt-repository ppa:deadsnakes/ppa
RUN DEBIAN_FRONTEND=noninteractive apt install -y \
  python3.8 \
  python3-pip \
  python3.8-dev \
  python3.8-full \
  libpython3.8-dev \
  libpython3.8-dev:armhf \
  libpython3.8-dev:arm64
RUN python3.8 -m pip install cython numpy

ENV CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:/usr/include/python3.8/:/usr/include/aarch64-linux-gnu/python3.8/"
```

And lastly i modified https://github.com/google-coral/pycoral/blob/master/Makefile#L15 to:
```python
PYTHON ?= $(shell which python3.8)
```

closed 2022-03-02 by roflcoopter