google-coral/pycoral #75: python -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0                                         (edgetpu) Looking in indexes: https://pypi.org/simple, https://google-coral.github.io/py-repo/ ERROR: Could not find a version that satisfies the requirement pycoral~=2.0 (from versions: 0.0.4, 0.0.5, 0.0.6, 0.0.7, 0.0.8, 0.0.9) ERROR: No matching distribution found for pycoral~=2.0

shaunsingh, 2022-03-02:
### Description

Trying to install PyCoral, I always get the following error:

python -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0                                         (edgetpu)
Looking in indexes: https://pypi.org/simple, https://google-coral.github.io/py-repo/
ERROR: Could not find a version that satisfies the requirement pycoral~=2.0 (from versions: 0.0.4, 0.0.5, 0.0.6, 0.0.7, 0.0.8, 0.0.9)
ERROR: No matching distribution found for pycoral~=2.0

I'm using python3 with miniforge/conda, but the same issue occurs with builtin python3 and python3 from Homebrew on macOS. I'm using an m1 cpu if that makes any different 

It can only find and install the `PyCoral` library from `pypi`, not google's. 

### Issue Type

Bug, Build/Install

### Operating System

Mac OS

### Coral Device

USB Accelerator

### Programming Language

Python 3.9

hjonnala, 2022-03-02:
Hello @shaunsingh  please try to install from whl files:  https://github.com/google-coral/edgetpu/issues/539#issuecomment-1040754426

shaunsingh, 2022-03-02:
Hi, those wheel files are for x86_64, while I am on `aarch64`. Would it still work?

hjonnala, 2022-03-03:
@shaunsingh   Can you please confirm which python version you are using. I think, you are using python3.9..If so, can you please try the below whl files. Thanks!

https://github.com/google-coral/pycoral/releases/download/v2.0.0/tflite_runtime-2.5.0.post1-cp39-cp39-macosx_11_0_x86_64.whl

https://github.com/google-coral/pycoral/releases/download/v2.0.0/pycoral-2.0.0-cp39-cp39-macosx_11_0_x86_64.whl

shaunsingh, 2022-03-04:
@hjonnala While I am using python 3.9, I'm running on an arm-based machine so wheels build for x86_64 will not work. I imagine this is the cause of the error.  I see that you build for `linux-aarch64`, would be be possible to build for aarch64-darwin as well? For example, numpy produces universal2 wheels, which can be used on m1 and intel machines: (https://github.com/numpy/numpy/releases/tag/v1.21.0 onwards)

hjonnala, 2022-03-04:
@shaunsingh Okay, with current build scripts its not possible to build the Pycoral for M1 chip.. We would try to make changes and update the repo to support build for M1 CPU.

dmitriykovalev, 2022-03-06:
We have uploaded wheels for Apple M1 to https://github.com/google-coral/py-repo, please try to install `pycoral` by running 

```
python3 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral
```

shaunsingh, 2022-03-06:
Works great, thank you!

closed 2022-03-06 by shaunsingh

shaunsingh, 2022-03-06:
Could you kindly release an updated build of the `edgetpu_runtime` as well? Currently my python script will fail to run as the latest released runtime still runs on x86
`/usr/local/lib/libedgetpu.1.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e'))`

hjonnala, 2022-03-07:
Apple M1 build support is added for libedgetpu also. Please try to build the [edgetpu_runtime](https://github.com/google-coral/libedgetpu) for your platform. Thanks!

https://github.com/google-coral/libedgetpu/commit/f9ad0c42f415ac66591b7ac8f26b3f8fd3bf4cf5

dmitriykovalev, 2022-03-08:
New runtime is now available at  [release-grouper](https://github.com/google-coral/libedgetpu/releases/tag/release-grouper) or you can build it locally by running `make runtime`.

felisida, 2023-08-31:
hi, I tried this procedure on my iMac with OS 12.6.5 and intel processor and Python 3.9 installed, I did something wrong?

homeassistant@IMAC-4A7A7E ~ % python3 -m pip install https://github.com/google-coral/pycoral/releases/download/v2.0.0/tflite_runtime-2.5.0.post1-cp39-cp39-macosx_11_0_x86_64.whl
ERROR: tflite_runtime-2.5.0.post1-cp39-cp39-macosx_11_0_x86_64.whl is not a supported wheel on this platform.
WARNING: You are using pip version 20.2.3; however, version 23.2.1 is available.
You should consider upgrading via the '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3 -m pip install --upgrade pip' command.
homeassistant@IMAC-4A7A7E ~ % python3 -m pip install https://github.com/google-coral/pycoral/releases/download/v2.0.0/pycoral-2.0.0-cp39-cp39-macosx_11_0_x86_64.whl
ERROR: pycoral-2.0.0-cp39-cp39-macosx_11_0_x86_64.whl is not a supported wheel on this platform.
WARNING: You are using pip version 20.2.3; however, version 23.2.1 is available.
You should consider upgrading via the '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3 -m pip install --upgrade pip' command.
homeassistant@IMAC-4A7A7E ~ %