google-coral/pycoral #77: Can't run PyCoral on Ubuntu 18.04

samsterckval, 2022-05-03:
### Description

When trying to run the parrot demo on Ubuntu 18.04 VM, errors about GLIBC version are thrown, however, libc6 is newest available version.

`ImportError: /lib/x86_64-linux-gnu/libm.so.6: version 'GLIBC_2.29' not found`

`libc6 is already the newest version (2.27-3ubuntu1.5).`

`$ uname -a`
`Linux xxxxxx-VirtualBox 5.4.0-109-generic #123~18.04.1-Ubuntu SMP Fri Apr 8 09:48:52 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

`$ lsusb`
`Bus 002 Device 002: ID 1a6e:089a Global Unichip Corp.`

### Issue Type

Build/Install

### Operating System

Ubuntu

### Coral Device

USB Accelerator

### Programming Language

Python 3.9

Namburger, 2022-05-03:
@samsterckval hello, would you be able to use a newer ubuntu version? 18.04 has an older version of glibc and is not compatible with the new binaries

samsterckval, 2022-05-03:
Unfortunately I am not.  I guess compiling both pycoral and tflite_runtime locally would fix this issue?

hjonnala, 2022-05-03:
@samsterckval  please check this comment to build pycoral on ubuntu 18.04: https://github.com/google-coral/pycoral/issues/74#issuecomment-1056905305

closed 2022-05-16 by hjonnala