google-coral/pycoral #78: Cannot Set Up SSH Key

nihalraman, 2022-05-03:
### Description

Hi! When I try  connect to my devboard via "mdt shell," I get the following error message:

I followed the instructions at https://coral.ai/docs/dev-board/mdt/#install-mdt, and it looks like the same key is on both my local device and on the coral. Any help would be very appreciated. Thanks!

Waiting for a device...
Connecting to wishful-orange at 172.20.10.4
Key not present on wishful-orange -- pushing

It looks like you're trying to connect to a device that isn't connected
to your workstation via USB and doesn't have the SSH key this MDT generated.
To connect with `mdt shell` you will need to first connect to your device
ONLY via USB.

Cowardly refusing to attempt to push a key to a public machine.

### Issue Type

Support

### Operating System

Mendel Linux

### Coral Device

Dev Board

hjonnala, 2022-05-03:
@nihalraman can you please try the solution provided here changing ip address from 192.168.10 to 172.20.10 in site-packages/mdt/sshclient.py line 86: https://stackoverflow.com/questions/58938727/unable-to-connect-google-coral-using-otg-port/59035796#59035796

nihalraman, 2022-05-07:
Ah, ok that makes sense. Thanks!