google-coral/pycoral #81: Can not install pycoral in Ubuntu Server 22.04 LTS for Raspberry Pi

tyouritsugun, 2022-06-30:
Hi,

Here is the steps to reproduce
```
sudo apt-get install python3-pycoral
```
I got 
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-pycoral : Depends: python3-tflite-runtime (= 2.5.0.post1) but it is not going to be installed
                   Depends: python3 (< 3.10) but 3.10.4-0ubuntu2 is to be installed
E: Unable to correct problems, you have held broken packages.

```
Then I tried to install python3-tflite-runtime, I got
```
ubuntu@ubuntu:~$ sudo apt-get install python3-tflite-runtime 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-tflite-runtime : Depends: python3 (< 3.10) but 3.10.4-0ubuntu2 is to be installed
E: Unable to correct problems, you have held broken packages.
```
It seems that my python version is incorrect, so I installed the python 3.8 and the result is
```
ubuntu@ubuntu:~$ python --version
Python 3.8.13
ubuntu@ubuntu:~$ ls -al /usr/bin/python
lrwxrwxrwx 1 root root 24 Jun 30 00:02 /usr/bin/python -> /etc/alternatives/python
ubuntu@ubuntu:~$ ls -al /etc/alternatives/python
lrwxrwxrwx 1 root root 18 Jun 30 00:04 /etc/alternatives/python -> /usr/bin/python3.8

```
However, the problem remains unchanged.

Is there anyway to solve this problem?

Tyou

tyouritsugun, 2022-06-30:
I also tried Ubuntu Desktop 22.04 LTS  for Raspberry Pi, I met the same problem.

Thank you in advance

Tyou

hjonnala, 2022-06-30:
Please try building the pycoral on Ubuntu 22.04. Here is an example with Ubuntu 18.04 and python 3.8 : https://github.com/google-coral/pycoral/issues/74#issuecomment-1056905305

Thanks!

tyouritsugun, 2022-07-01:
Hi

Thank you for your reply. I changed `scripts/build.sh` as below:
```
set -ex

readonly SCRIPT_DIR="$(cd "$(dirname "${BASH_SOURCE[0]}")" && pwd)"
readonly MAKEFILE="${SCRIPT_DIR}/../Makefile"
readonly DOCKER_CPUS="${DOCKER_CPUS:=k8 aarch64 armv7a}"
PYTHON_VERSIONS="36 37 38 39 310"

while [[ $# -gt 0 ]]; do
  case "$1" in
    --clean)
      make -f "${MAKEFILE}" clean
      shift
      ;;
    --python_versions)
      PYTHON_VERSIONS=$2
      shift
      shift
      ;;
    *)
      shift
      ;;
  esac
done

function docker_image {
  case $1 in
    36) echo "ubuntu:18.04" ;;
    37) echo "debian:buster" ;;
    38) echo "ubuntu:20.04" ;;
    39) echo "debian:bullseye" ;;
    310) echo "ubuntu:22.04" ;;
    *) echo "Unsupported python version: $1" 1>&2; exit 1 ;;
  esac
}

for python_version in ${PYTHON_VERSIONS}; do
  make DOCKER_CPUS="${DOCKER_CPUS}" \
       DOCKER_IMAGE=$(docker_image "${python_version}") \
       DOCKER_TARGETS="pybind tflite wheel tflite-wheel" \
       -f "${MAKEFILE}" \
       docker-build
done

```
Then I ran the script, after one hour I got the error as blow:
```
Processing triggers for libc-bin (2.35-0ubuntu3) ...
Reading package lists...
Building dependency tree...
Reading state information...
Package python is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python2-minimal:armhf python2:armhf python2-minimal:arm64 python2:arm64
  python2-minimal python2 dh-python 2to3 python-is-python3

E: Package 'python' has no installation candidate
The command '/bin/sh -c apt-get update     && DEBIAN_FRONTEND=noninteractive apt-get install -y       libc6-dev:arm64       libc6-dev:armhf     && DEBIAN_FRONTEND=noninteractive apt-get install -y       sudo       debhelper       python       python3-all       python3-numpy       python3-setuptools       python3-six       python3-wheel       libpython3-dev       libpython3-dev:armhf       libpython3-dev:arm64       build-essential       crossbuild-essential-armhf       crossbuild-essential-arm64       libusb-1.0-0-dev       libusb-1.0-0-dev:arm64       libusb-1.0-0-dev:armhf       zlib1g-dev       zlib1g-dev:armhf       zlib1g-dev:arm64       pkg-config       p7zip-full       zip       unzip       curl       wget       git       vim       mc       software-properties-common' returned a non-zero code: 100

```
It seems that it is not easy to modify `scripts/build.sh`.

Do you have official support for Ubuntu Desktop(SERVER) 22.04 LTS?

Thank you in advance
Tyou

hjonnala, 2022-07-01:
We don't have any official support for Ubuntu Desktop(SERVER) 22.04 LTS. Please check this tutorial(https://www.youtube.com/watch?v=QqTLkw1IenI&list=PLQY2H8rRoyvz_anznBg6y3VhuSMcpN9oe&index=4) and try with [tflite-support](https://github.com/tensorflow/tflite-support) API.

tyouritsugun, 2022-07-01:
Thank you for your reply. I think that it is better to use 20.04 for now.

Tyou

closed 2022-07-01 by tyouritsugun