google-coral/pycoral #83: Cannot install python3-pycoral

halogen28372, 2022-07-24:
### Description

When I attempt to install pycoral I get the error below:

The following packages have unmet dependencies:
 python3-pycoral : Depends: python3-tflite-runtime (= 2.5.0.post1) but it is not going to be installed
                   Depends: python3 (< 3.10) but 3.10.4-0ubuntu2 is to be installed
E: Unable to correct problems, you have held broken packages.

I am installing on a fresh install of Ubuntu 22.04 server edition.

### Issue Type

Build/Install

### Operating System

Ubuntu

### Coral Device

Mini PCIe

### Relevant Log Output

```shell
# sudo apt-get install python3-pycoral
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-pycoral : Depends: python3-tflite-runtime (= 2.5.0.post1) but it is not going to be installed
                   Depends: python3 (< 3.10) but 3.10.4-0ubuntu2 is to be installed
E: Unable to correct problems, you have held broken packages.
```

hjonnala, 2022-07-25:
Please try building the pycoral on Ubuntu 22.04. Here is an example with Ubuntu 18.04 and python 3.8 : https://github.com/google-coral/pycoral/issues/74#issuecomment-1056905305

Thanks!

halogen28372, 2022-07-25:
Thanks for the tip, but I'd really like an easier way to install. Is there a Ubuntu/Linux version that just works?

halogen28372, 2022-08-28:
I ended up using an older distro of Ubuntu and that worked fine.

closed 2022-09-12 by hjonnala

dnguyen800, 2022-10-28:
> I ended up using an older distro of Ubuntu and that worked fine.

Which version of Ubuntu did you use?

halogen28372, 2022-10-28:
It was 18 point something. It was the most up to date 18 available at the time. 
> > I ended up using an older distro of Ubuntu and that worked fine.
> 
> 
> 
> Which version of Ubuntu did you use?

Eschguy, 2023-02-09:
Using Ubuntu 22.04.1 and am getting the same error STILL.

halogen28372, 2023-02-09:
Unless you are skilled with Linux I recommend just using Ubuntu 18. You will also have to use an older version of python, but that should be standard for Ubuntu 18.

tokenwizard, 2023-03-01:
Will this package be updated to work with current versions of Ubuntu? Version 18 (Bionic) will reach EOL in May. It was released in 2018 and at this point is nearly 5 years old. It seems silly to keep telling people to use such old OS versions rather than updating the software to work current-gen versions of the OS. 

Just my $0.02

tokenwizard, 2023-03-01:
FYI, it seems to install properly on 20.04 if you don't want to use nearly 5 year old Ubuntu 18.

jfiraben, 2024-02-28:
So this doesn't work on Debian 12?