google-coral/pycoral #85: Python 3.10 and 3.11 support?

baudneo, 2022-08-10:
Anyone compiled pycoral for 3.10 and 3.11? I am attempting to build them but I was wondering if someone else has already done this?

fenil1608, 2022-08-13:
Hi, any update on this ? I'm trying to use pycoral on ubuntu 22 with python 3.10.

baudneo, 2022-08-15:
I'm assuming no since the assignee hasn't replied to this issue but did reply to a newer issue.

It seems maybe it's based on dependencies that aren't available in 3.10+ yet. Only option I can see is the community forking the project and attempting to build for 3.10+.

I will give it a try when I have a bit of spare time.

travisariggs, 2022-10-03:
@baudneo , did you ever find any spare time to try building for 3.10?  I am very interested in running on 3.10. I tried simply updating the `build.sh` docker image to use `ubuntu:22.04`, but that wasn't enough to get it to work.

baudneo, 2022-10-03:
It's based on wheels for scikit or tensorflow from what I understand. When the upstream team releases the wheels than 3.x integration can happen

peteh, 2022-10-15:
As far as I understand both scikit and tensorflow have 3.10 compatibility marked in pipy. However, I'm failing to run the examples on ubuntu 22.04

baudneo, 2022-10-15:
Compatibility yes, wheels no. Wheels are needed by pycoral to build their 3.10 stuff.

hjonnala, 2022-11-07:
please try the [wheels](https://github.com/hjonnala/snippets/tree/main/wheels/python3.10), built locally on Linux, for pyhton3.10 and let us know if those are working on ubuntu:22.04. Thanks!

baudneo, 2022-11-07:
Will give her a go tonight!

baudneo, 2022-11-08:
Ill have to make a build of Ubuntu 21.10 (glibc 2.34) or 22.04 (glibc 2.35) as 21.04 uses glibc 2.33 and throws this error ->
```
>>> from pycoral.adapters import common
>>> from pycoral.adapters import common, detect
>>> from pycoral.utils.edgetpu import make_interpreter as make_interpreter, list_edge_tpus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tyler/.local/lib/python3.10/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/tyler/.local/lib/python3.10/site-packages/pycoral/pybind/_pywrap_coral.cpython-310-x86_64-linux-gnu.so)
>>>
```

travisariggs, 2022-11-08:
I tried it on my yocto kirkstone system with the same error (though different version of GLIBC)

```
      from pycoral.utils import edgetpu
  File "/opt/ros/noetic/lib/python3.10/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
      from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /opt/ros/noetic/lib/python3.10/site-packages/pycoral/pybind/_pywrap_coral.cpython-310-x86_64-linux-gnu.so)
```

Do you have instructions to build this shared object?

hjonnala, 2022-11-09:
Please install the [bazel](https://bazel.build/install/ubuntu) and try building the [pycoral](https://github.com/google-coral/pycoral) itself and share the complete logs if there are any errors.

```
git clone --recurse-submodules https://github.com/google-coral/pycoral
cd pycoral
make
make wheel
make tflite-wheel
```

travisariggs, 2022-11-09:
I was not able to build on a fresh image of Ubuntu 22.04.

I installed bazelisk, cloned the pycoral repo and ran `make`.

My error log is attached.

[build-fail.log](https://github.com/google-coral/pycoral/files/9974782/build-fail.log)

hjonnala, 2022-11-10:
Please check whether /usr/include/python3.10 path exist and change  _get_python_include function in this file as below and rerun the make command: /home/ubuntu/.cache/bazel/_bazel_ubuntu/d8383202a9b6366beeb5d874af8d5d3f/external/org_tensorflow/third_party/py/python_configure.bzl

```
def _get_python_include(repository_ctx, python_bin):
    """Gets the python include path."""
    return "/usr/include/python3.10"
```

hjonnala, 2022-11-10:
You may get some other errors after rerun. Please add the below lines to /home/ubuntu/.cache/bazel/_bazel_ubuntu/d8383202a9b6366beeb5d874af8d5d3f/external/ruy/ruy/block_map.cc

```
#include <stdexcept>
#include <limits>
```

travisariggs, 2022-11-10:
The path `/usr/include/python3.10` did not exist on my Ubuntu 22 system.  I installed "python3-numpy", which did create that base path.

I modified the `_get_python_include` to return "/usr/include/python3.10".  I then ran `make`.  Here is the log of that failure:

[20221110-build-fail-1.log](https://github.com/google-coral/pycoral/files/9984712/20221110-build-fail-1.log)

I then added <stdexcept> and <limits> to `block_map.cc` at the path you gave.

Unfortunately, running make resulted in an error.  It was unable to find `Python.h`.  That is captured in this log:

[20221110-build-fail-2.log](https://github.com/google-coral/pycoral/files/9984724/20221110-build-fail-2.log)

I searched the filesystem and found that Python.h was not installed.  I ran `apt install python3-dev`, which installed `Python.h` to `/usr/include/python3.10/Python.h`.

I ran make again with the same error.  It still wasn't able to find Python.h.  I tried rebooting, running make clean and then make again...same error.

[20221110-build-fail-3.log](https://github.com/google-coral/pycoral/files/9984730/20221110-build-fail-3.log)

Do I need to reset some cached state in bazel?

Also, do I need to run some sort of setup script to install the dependencies that you expect to be present at build time?  I certainly needed to install python3-dev, but are there others?

baudneo, 2022-11-10:
It has compiled and is working as expected on a new 22.04 Jammy install

travisariggs, 2022-11-11:
I started with a brand new install of Ubuntu 22.  This time, I installed these apt packages before attempting to build the wheel files:

1. build-essential
2. git
3. python3-dev
4. python3-wheel
5. python3-numpy

Then, I did the following:

1. Installed bazelisk
2. Cloned pycoral
3. Ran make (failed)
4. Edited python_configure.bzl to return the python include path
5. Ran make (failed)
6. Edited block_make.cc to add the extra headers
7. Ran make
8. Ran make wheel
9. Ran make tflite-wheel

Wheels built successfully!

However, after installing them on my non-Ubuntu (Yocto Kirkstone) system, I get a runtime error involving GLIBCXX:
```
Traceback (most recent call last):
     from pycoral.utils import edgetpu
   File "/opt/ros/noetic/lib/python3.10/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
     from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
 ImportError: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /opt/ros/noetic/lib/python3.10/site-packages/pycoral/pybind/_pywrap_coral.cpython-310-x86_64-linux-gnu.so)
```

According to this (https://gcc.gnu.org/onlinedocs/libstdc++/manual/abi.html), it looks like I need to compile with gcc 12.1, but ubuntu 22 defaults to gcc 11.1.

I am going to try installing gcc 12.1 and rebuilding next week.

Does that seem reasonable?

baudneo, 2022-11-11:
> I started with a brand new install of Ubuntu 22.  This time, I installed these apt packages before attempting to build the wheel files:
>
> 1. build-essential
> 2. git
> 3. python3-dev
> 4. python3-wheel
> 5. [...]

Build it on your yocto system using its libs.

rdejana, 2023-03-07:
Any updates on this?

baudneo, 2023-03-07:
> Any updates on this?

Idk if 3.10 has easily installable wheels but you can build it yourself by following the instructions in this thread.

nandra, 2023-03-20:
I've tried to add python 3.10 support directly to build all binaries using docker but it's failing on various places. Does anybody work on it please? Thanks.

jakew009, 2023-03-20:
Is there any reason why only a very limited subset of wheels for python 3.10 have been released? Is this repo essentially abandoned? Building the wheels from scratch is a nightmare.

jakew009, 2023-03-20:
@travisariggs did you ever managed to build a wheel successfully for kirkstone? We are trying to do the same thing

rgriffogoes, 2023-03-31:
Just followed the steps above and it works, but ensure that **you have an older version of bazel.** 

I'm on a fresh Ubuntu 22.04 Server, and installing latest Bazel (6.0.0) the process failed (with something like `Please use constraints from @platforms repository embedded in Bazel`)

Had to purge it + install an old version:

```
sudo apt-get --purge remove bazel

sudo apt install bazel-5.3.2
sudo ln -s /usr/bin/bazel-5.3.2 /usr/bin/bazel
```
Then the process described by [travisariggs](https://github.com/travisariggs) on Nov 11th worked (including the 2 edits in the cache files).

rancerbeta, 2023-05-02:
Note for: https://github.com/google-coral/pycoral/issues/85#issuecomment-1491249583 
if you've already got bazelisk installed, all you should need to do to change bazel versions is to point bazel at bazelisk, set USE_BAZEL_VERSION= and then, use as usual. bazelisk should handle the version change.

adding some notes for investigation in case it's useful for someone:

Seems like the one of the cause of python 3.10 not building is a 
failure in a utility from tensorflow, which is currently being pulled at a commit from
 https://github.com/tensorflow/tensorflow/commit/a4dfb8d1a71385bd6d122e4f27f86dcebb96712d 
according to ./libedgetpu/workspace.bzl:9:TENSORFLOW_COMMIT = "a4dfb8d1a71385bd6d122e4f27f86dcebb96712d"
which is from May 12, 2021

The utility failure is at: https://github.com/tensorflow/tensorflow/blob/master/third_party/py/python_configure.bzl#L156
this errors out because their execute utility considers anything in stderr to be a failure:
https://github.com/tensorflow/tensorflow/blob/master/third_party/remote_config/common.bzl#L229

The latest version does something different, which doesn't trigger the deprecation msg 
https://github.com/tensorflow/tensorflow/commit/7988440dfaa83bf8e147d879c742f29464530baa
and was added Feb 7, 2022

Ruy fails due to missing includes needed with newer gcc,  which was fixed in:
https://github.com/google/ruy/pull/271
June 15, 2021

Ruy is pulled in by tensorflow and at the hash that tensorflow is at in the commit at https://github.com/tensorflow/tensorflow/commit/f779f42004d1873784cb2b4cb351d4b342ad0816 
Mar 4, 2021

tensorflow bump of ruy https://github.com/tensorflow/tensorflow/commit/6f4b5a666f650da68c3fa1b3c9bb38640eaa2236 on June 18. 2021 should pull in the newer fix

It seems like upgrading tensorflow to a newer version would also pull in these fixes, instead of needing to fix the bazel cache, but upgrades are often tricky. I'm not sure what else would need to be updated.

seems like you'd want at least tf 2.9, but looking at the release notes... there's quite a few breaking changes from 2.5 to now

rikardn, 2023-05-04:
About the GLIBCXX import error:

To my understanding wheels should not be built with dynamic linking to the C++ library. Static linking should be used. GLIBC is ok to dynamically link. I checked wheels from other C++ projects and they don't dynamically depend on libstdc++. If using gcc this can be accomplished by adding the `-static-libgcc` and the `-static-libstdc++` options.

cgranetgithub, 2023-06-12:
**+1 for more recent Python versions support.**
(I'm using Coral with a Raspberry Pi, developing a robot with ROS2 - Robot Operating System. To ensure compatibility between pycoral/ROS2/Ubuntu, I have to go back to Ubuntu Focal 20.04 and ROS2 Foxy which is not supported anymore...)

krsureshmca82, 2023-06-15:
> I started with a brand new install of Ubuntu 22. This time, I installed these apt packages before attempting to build the wheel files:
>
> 1. build-essential
> 2. git
> 3. python3-dev
> 4. python3-wheel
> 5. [...]

Hi,

As per your procedure I am trying to build pycoral for python3.10 but facing following build error.
Could you please help me on this

rnd@rnd-TRX40-AORUS-XTREME:~/Project/Coral/pycoral$ make
Loading: 0 packages loaded
Loading: 0 packages loaded
PYTHON_BIN_PATH=/home/rnd/.pyenv/shims/python3 bazel build --compilation_mode=opt --copt=-DNPY_NO_DEPRECATED_API=NPY_1_7_API_VERSION --cpu=k8 --linkopt=-L/home/rnd/Project/Coral/pycoral/libedgetpu_bin/direct/k8 --linkopt=-l:libedgetpu.so.1 --linkopt=-Wl,--strip-all \
    --embed_label='TENSORFLOW_COMMIT=a4dfb8d1a71385bd6d122e4f27f86dcebb96712d' \
    --stamp \
    //src:_pywrap_coral
ERROR: /home/rnd/.cache/bazel/_bazel_rnd/83230b4bceeb8dd10b1bc66e418a7aa5/external/bazel_tools/platforms/BUILD:19:6: in alias rule @bazel_tools//platforms:x86_64: Constraints from @bazel_tools//platforms have been removed. Please use constraints from @platforms repository embedded in Bazel, or preferably declare dependency on https://github.com/bazelbuild/platforms. See https://github.com/bazelbuild/bazel/issues/8622 for details.
ERROR: /home/rnd/.cache/bazel/_bazel_rnd/83230b4bceeb8dd10b1bc66e418a7aa5/external/bazel_tools/platforms/BUILD:19:6: Analysis of target '@bazel_tools//platforms:x86_64' failed
ERROR: /home/rnd/Project/Coral/pycoral/src/BUILD:67:17: While resolving toolchains for target //src:_pywrap_coral: Target @local_execution_config_platform//:platform was referenced as a platform, but does not provide PlatformInfo
ERROR: Analysis of target '//src:_pywrap_coral' failed; build aborted: 
INFO: Elapsed time: 0.064s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets configured)
make: *** [Makefile:152: pybind] Error 1

Thanks,
Sureshkumar

hjonnala, 2023-06-15:
> ERROR: /home/rnd/.cache/bazel/_bazel_rnd/83230b4bceeb8dd10b1bc66e418a7aa5/external/bazel_tools/platforms/BUILD:19:6: Analysis of target '@bazel_tools//platforms:x86_64' failed
> ERROR: [...]

https://github.com/google-coral/pycoral/issues/85#issuecomment-1491249583

krsureshmca82, 2023-06-17:
> > I started with a brand new install of Ubuntu 22. This time, I installed these apt packages before attempting to build the wheel files:
> > 
> > 1. build-essential
> > 2. git
> > 3. python3-dev
> > 4. python3-wheel
> > 5. python3-numpy
> > 
> > Then, I did the following:
> > 
> > 1. Installed bazelisk
> > 2. Cloned pycoral
> > 3. Ran make (failed)
> > 4. Edited python_configure.bzl to return the python include path
> > 5. Ran make (failed)
> > 6. Edited block_make.cc to add the extra headers
> > 7. Ran make
> > 8. Ran make wheel
> > 9. Ran make tflite-wheel
> > 
> > Wheels built successfully!
> > However, after installing them on my non-Ubuntu (Yocto Kirkstone) system, I get a runtime error involving GLIBCXX:
> > ```
> > Traceback (most recent call last):
> >      from pycoral.utils import edgetpu
> >    File "/opt/ros/noetic/lib/python3.10/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
> >      from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
> >  ImportError: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /opt/ros/noetic/lib/python3.10/site-packages/pycoral/pybind/_pywrap_coral.cpython-310-x86_64-linux-gnu.so)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > According to this (https://gcc.gnu.org/onlinedocs/libstdc++/manual/abi.html), it looks like I need to compile with gcc 12.1, but ubuntu 22 defaults to gcc 11.1.
> > I am going to try installing gcc 12.1 and rebuilding next week.
> > Does that seem reasonable?
> 
> Hi,
> 
> As per your procedure I am trying to build pycoral for python3.10 but facing following build error. Could you please help me on this
> 
> rnd@rnd-TRX40-AORUS-XTREME:~/Project/Coral/pycoral$ make Loading: 0 packages loaded Loading: 0 packages loaded PYTHON_BIN_PATH=/home/rnd/.pyenv/shims/python3 bazel build --compilation_mode=opt --copt=-DNPY_NO_DEPRECATED_API=NPY_1_7_API_VERSION --cpu=k8 --linkopt=-L/home/rnd/Project/Coral/pycoral/libedgetpu_bin/direct/k8 --linkopt=-l:libedgetpu.so.1 --linkopt=-Wl,--strip-all --embed_label='TENSORFLOW_COMMIT=a4dfb8d1a71385bd6d122e4f27f86dcebb96712d' --stamp //src:_pywrap_coral ERROR: /home/rnd/.cache/bazel/_bazel_rnd/83230b4bceeb8dd10b1bc66e418a7aa5/external/bazel_tools/platforms/BUILD:19:6: in alias rule @bazel_tools//platforms:x86_64: Constraints from @bazel_tools//platforms have been removed. Please use constraints from @platforms repository embedded in Bazel, or preferably declare dependency on https://github.com/bazelbuild/platforms. See [bazelbuild/bazel#8622](https://github.com/bazelbuild/bazel/issues/8622) for details. ERROR: /home/rnd/.cache/bazel/_bazel_rnd/83230b4bceeb8dd10b1bc66e418a7aa5/external/bazel_tools/platforms/BUILD:19:6: Analysis of target '@bazel_tools//platforms:x86_64' failed ERROR: /home/rnd/Project/Coral/pycoral/src/BUILD:67:17: While resolving toolchains for target //src:_pywrap_coral: Target @local_execution_config_platform//:platform was referenced as a platform, but does not provide PlatformInfo ERROR: Analysis of target '//src:_pywrap_coral' failed; build aborted: INFO: Elapsed time: 0.064s INFO: 0 processes. FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets configured) make: *** [Makefile:152: pybind] Error 1
> 
> Thanks, Sureshkumar

> > I started with a brand new install of Ubuntu 22. This time, I installed these apt packages before attempting to build the wheel files:
> > 
> > 1. build-essential
> > 2. git
> > 3. python3-dev
> > 4. python3-wheel
> > 5. python3-numpy
> > 
> > Then, I did the following:
> > 
> > 1. Installed bazelisk
> > 2. Cloned pycoral
> > 3. Ran make (failed)
> > 4. Edited python_configure.bzl to return the python include path
> > 5. Ran make (failed)
> > 6. Edited block_make.cc to add the extra headers
> > 7. Ran make
> > 8. Ran make wheel
[... 32 lines cut ...]
> Thanks, Sureshkumar

Hi,

Thanks for the reply.
As of now, I am able to build Pycoral3.8 with "make wheel", but I need Pycoral 3.10
Now I'm trying to build Pycoral on Ubuntu 22.04 (it seems that Python is being taken from the default system path).

Could you please assist me with the following issue while building a pycoral in ubuntu 22.04?

ERROR: An error occurred during the fetch of repository 'local_execution_config_python':
   Traceback (most recent call last):
	File "/root/.cache/bazel/_bazel_root/eab0d61a99b6696edb3d2aff87b585e8/external/org_tensorflow/third_party/py/python_configure.bzl", line 212, column 41, in _create_local_python_repository
		python_include = _get_python_include(repository_ctx, python_bin)
	File "/root/.cache/bazel/_bazel_root/eab0d61a99b6696edb3d2aff87b585e8/external/org_tensorflow/third_party/py/python_configure.bzl", line 152, column 21, in _get_python_include
		result = execute(
	File "/root/.cache/bazel/_bazel_root/eab0d61a99b6696edb3d2aff87b585e8/external/org_tensorflow/third_party/remote_config/common.bzl", line 219, column 13, in execute
		fail(
Error in fail: Problem getting python include path.
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
Is the Python binary path set up right? (See ./configure or PYTHON_BIN_PATH.) Is distutils installed?

Thanks and Regards,
Sureshkumar R

FoUStep, 2023-06-21:
Trying to figure out how to fix this gcc error (with python 3.11):
[make.log](https://github.com/google-coral/pycoral/files/11815802/make.log)

I was planning to do a quick install on Debian 12 but apparently it seems some of the dependencies are outdated?

ranrinc, 2023-06-24:
> Trying to figure out how to fix this gcc error (with python 3.11): [make.log](https://github.com/google-coral/pycoral/files/11815802/make.log)
> 
> I was planning to do a quick install on Debian 12 but apparently it seems some of the dependencies are outdated?

I also need to know how to install it on Debian 12 since proxmox already run Debian 12

tuxthepenguin84, 2023-06-27:
Hi @Lockie85

You can use pyenv to manage multiple versions of Python https://github.com/pyenv/pyenv

For apt gpg keys the steps listed above are in fact deprecated. The correct way to manage the key is to place the key here ```/usr/share/keyrings/coral.gpg``` and then create ```/etc/apt/sources.list.d/coral.list``` with the following contents ```deb [signed-by=/usr/share/keyrings/coral.gpg] https://packages.cloud.google.com/apt coral-edgetpu-stable main``` afterwards you can run ```sudo apt update``` and then you should be able to install the necessary packages.

microspino, 2023-09-02:
Same here: tried on Ubuntu 22.04.3 LTS Jammy got the same error. Tried everything @travisariggs and @hjonnala said being sure bazel (installed via bazelisk) stayed at 5.3.2 like @rgriffogoes wrote but nothing, I'm stuck here:

```
Target //src:_pywrap_coral failed to build
INFO: Elapsed time: 2.267s, Critical Path: 0.16s
INFO: 8 processes: 7 internal, 1 linux-sandbox.
FAILED: Build did NOT complete successfully
make: *** [Makefile:152: pybind] Error 1
```

cappittall, 2023-09-04:
Fastest and easy way:

- create python 3.10 environment.  (not working at 3.9 either 3.11) 
- Download those two .whl files  [pycoral](https://drive.google.com/drive/folders/1Pt89Tde6CIipLxiAQTJn9PfYS-gSjKUF?usp=sharing) 
- pip install pycoral-2.0.0-cp310-cp310-linux_x86_64.whl 
- pip install  tflite_runtime-2.5.0.post1-cp310-cp310-linux_x86_64.whl
Thants all.

baudneo, 2023-09-04:
> Fastest and easy way:
> 
> * Create python 3.10 environment.  (not working at 3.9 either 3.11)
> * Download those two .whl files  [pycoral](https://drive.google.com/drive/folders/1Pt89Tde6CIipLxiAQTJn9PfYS-gSjKUF?usp=sharing)
> * pip install pycoral-2.0.0-cp310-cp310-linux_x86_64.whl
> * pip install  tflite_runtime-2.5.0.post1-cp310-cp310-linux_x86_64.whl
> 
> thats all.

Installing random wheels hosted on google drive by unknown author(s) may not be up everyone's alley though....

cappittall, 2023-09-04:
@baudneo, you are right!  where do you suggest to upload?

baudneo, 2023-09-04:
Hard to say, I don't really have a recommendation besides getting @hjonnala to host them here or somewhere linked to this project. I'm thinking more in terms of production use case, a lot of companies won't allow installing unknown wheels, for home users it's their call.

cappittall, 2023-09-04:
@baudneo I uploaded here github, I found and used those files for many times also. 
https://github.com/cappittall/pycoral_whl_4_python3.10

cappittall, 2023-09-05:
@cgranetgithub, I don't know if its work but just an idea, try to flash with mendel OS. (all necessary packages are embeded) 
https://coral.ai/software/#mendel-linux

baudneo, 2023-10-11:
Its crazy that this is a repo controlled by a multi billion dollar company and they cant even bother to build wheels for python versions that have seen some of the biggest performance increases in years.  Good thing we all went out and spent money on hardware that we cant take full advantage of due to software limitations and abandonment by the owners.

Christ, even Intel updates their software for old ass compute sticks. 

Lesson learned, don't buy hardware from google or any company google has a stake in.

tuxthepenguin84, 2023-10-11:
So to clarify there are two ways to get this to work if you are on Python >= 3.10:

- Install the wheels located here: https://github.com/hjonnala/snippets/tree/main/wheels/python3.10
or
- Install Python 3.9 with pyenv: https://realpython.com/intro-to-pyenv/

Is this correct?

BradleyFord, 2023-10-11:
> So to clarify there are two ways to get this to work if you are on Python >= 3.10:
> 
> * Install the wheels located here: https://github.com/hjonnala/snippets/tree/main/wheels/python3.10
>   or
> * Install Python 3.9 with pyenv: https://realpython.com/intro-to-pyenv/
> 
> Is this correct?

This could be an interesting temporary path; are there any instructions on how to install this?

David-OConnor, 2023-10-30:
What's the word?

David-OConnor, 2023-10-30:
> > So to clarify there are two ways to get this to work if you are on Python >= 3.10:
> > 
> > * Install the wheels located here: https://github.com/hjonnala/snippets/tree/main/wheels/python3.10
> >   or
> > * Install Python 3.9 with pyenv: https://realpython.com/intro-to-pyenv/
> > 
> > Is this correct?
> 
> This could be an interesting temporary path; are there any instructions on how to install the wheels version?

Any way on Windows?

David-OConnor, 2023-10-30:
It appears to work on Windows, from Pip, with Python 3.1, and pycoral 0.2.0.

David-OConnor, 2023-11-02:
Hero

oberluz, 2023-11-02:
I haven't uploaded them up yet, still backing. I'll post here when done in the next hour or so

baudneo, 2023-11-02:
Sick, good thing the community took the time to support a multi billion dollar companies hardware.

Thank you @oberluz

oberluz, 2023-11-02:
> Sick, good thing the community took the time to support a multi billion dollar companies hardware.
> 
> Thank you @oberluz

No worries.

If anyone has cloned the repo since i made the announcement please do:

$ git pull
$ git submodules update --init --recursive

as I've added a commit that modifies the libcoral submodule.

I'm rebuilding the wheels and will upload them shortly.

@baudneo you mentioned you where interested in 3.11 which I haven't tested yet. I'll add it shortly using debian bookworm to build it...

oberluz, 2023-11-02:
@David-OConnor @baudneo Here are the wheels. Let me know how you go...:

https://github.com/oberluz/pycoral/releases/tag/2.7.0

David-OConnor, 2023-11-02:
Love it. Should I be able to build for Windows using the source and PyCoral instructions?

oberluz, 2023-11-02:
> Love it. Should I be able to build a wheel for Windows using the source versions you posted and PyCoral instructions?

I don't have a windows machine or VM. All I can say is try it. I haven't modified the way it build so if you were able to do it before you should be able to now.

If you do it would be good if we can include the windows wheels in the release for others.

David-OConnor, 2023-11-02:
Awesome; I'll post back once I try if it works

David-OConnor, 2023-11-02:
On Ubuntu 20, Python 3.11, installing the wheel results in `Error: pycoral...wheel is not a supported wheel on this platform.` I wonder if I could build for 3.11 using your sources though.

David-OConnor, 2023-11-02:
I attempted to build on Windows/Py3.11, and Ubuntu/Py3.11. Same error on both:

```
ERROR: Could not find a version that satisfies the requirement tflite-runtime==2.7.0.post1 (from pycoral) (from versions: none)
ERROR: No matching distribution found for tflite-runtime==2.7.0.post1
```

oberluz, 2023-11-02:
Yes, I'm still adding support for 3.11. I'll need to try the next
tensorflow version that has a recent version of pybind11 in its bazel
configuration.

There is a branch for 3.11 but us failing due to the reasons I've explained

Can you try building the wheels for 3.6, 3.7, 3.8, 3.9 and 3.10? Those
should work.

oberluz, 2023-11-05:
> I attempted to build on Windows/Py3.11, and Ubuntu/Py3.11. Same error on both:
> 
> ```
> ERROR: Could not find a version that satisfies the requirement tflite-runtime==2.7.0.post1 (from pycoral) (from versions: none)
> ERROR: No matching distribution found for tflite-runtime==2.7.0.post1
> ```
> 
> I wonder if the fact the assets are split into pycoral and tflite_runtime wheels is relevant? Of note, I don't know how this system works, and was previously confused about the difference between the two libs.

I guess they split them originally so that tflite-runtime could be used for other things then pycoral.

Regarding your error, you need to install tflite-runtime before pycoral..

oberluz, 2023-11-05:
> I attempted to build on Windows/Py3.11, and Ubuntu/Py3.11. Same error on both:
>
> ```
> ERROR: Could not find a version that satisfies the requirement tflite-runtime==2.7.0.post1 (from pycoral) (from [...]

I guess they split them originally so that tflite-runtime could be used for other things then pycoral.

Regarding your error, you need to install tflite-runtime before pycoral..

David-OConnor, 2023-11-05:
I appreciate the info; I have it installed.

[20 more replies]