google-coral/pycoral #86: E: Unable to locate package edgetpu-compiler

soundarsurya, 2022-08-12:
### Description

I'm trying to install edgetpu-compiler in my ubuntu machine. Followed the steps provided here [edgetpu-compiler](https://coral.ai/docs/edgetpu/compiler/#download). But i cannot able to install it.

Error i'm getting
```
ubuntu@my-machine:~$ sudo apt-get install edgetpu-compiler
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package edgetpu-compiler
ubuntu@my-machine:~$
```
Machine which i'm trying to install edgetpu-compiler
```
ubuntu@my-machine:~$ uname -a
Linux my-machine 5.4.0-120-generic #136~18.04.1-Ubuntu SMP Fri Jun 10 18:00:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
ubuntu@my-machine:~$
```
Logs of the steps i followed 
```
ubuntu@my-machine:/etc/apt$ curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2537  100  2537    0     0   1800      0  0:00:01  0:00:01 --:--:--  1799
OK
ubuntu@my-machine:/etc/apt$
ubuntu@my-machine:/etc/apt$ echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list
deb https://packages.cloud.google.com/apt coral-edgetpu-stable main
ubuntu@my-machine:/etc/apt$
```

### Issue Type

Build/Install

### Operating System

Ubuntu

### Coral Device

Dev Board

hjonnala, 2022-08-12:
Please try running `sudo apt-get update` before running `sudo apt-get install edgetpu-compiler `.

soundarsurya, 2022-08-16:
Thanks, it worked! 
Actually some proxy connection issue at my end.
Now Installed edgetpu-compiler into my system successfully.

closed 2022-08-16 by soundarsurya