google-coral/pycoral #87: Yocto hardknott arm64, and No module named 'pycoral.adapters'

fenil1608, 2022-08-15:
### Description

Hi, I'm trying to use the USB accelerator with IMX som module running on yocto hardknott with python version 3.9.5, after installing the pycoral from the release tarball, I'm still unable to use the pycoral API to run inference on USB Accelerator. The edgetpu library is already installed. 

### Issue Type

Bug

### Operating System

Linux

### Coral Device

USB Accelerator

### Programming Language

Python 3.9

### Relevant Log Output

```shell
Traceback (most recent call last):
  File "/home/root/coral/pycoral/examples/classify_image.py", line 37, in <module>
    from pycoral.adapters import classify
ModuleNotFoundError: No module named 'pycoral.adapters'
```

hjonnala, 2022-08-15:
can you please share the output of below commands. Thanks!

```
Python 3.9.7 (default, Sep  3 2021, 06:18:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tflite_runtime as tflite
>>> import pycoral
>>> from pycoral.utils.edgetpu import get_runtime_version
>>> get_runtime_version()
'BuildLabel(COMPILER=6.3.0 20170516,DATE=redacted,TIME=redacted), RuntimeVersion(14)'
>>> tflite.__version__
'2.5.0.post1'
>>> pycoral.__version__
'2.0.0'
>>> 
```

fenil1608, 2022-08-17:
root@imx8mpsolidrun:~# python3
Python 3.9.5 (default, May  3 2021, 15:11:33)
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tflite_runtime as tflite
>>> import pycoral
>>> from pycoral.utils.edgetpu import get_runtime_version
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pycoral.utils'
>>> get_runtime_version()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'get_runtime_version' is not defined
>>> tflite.__version__
'2.6.0'
>>> pycoral.__version__
'2.0.0'

closed 2022-08-17 by fenil1608