google-coral/pycoral #91: `make_interpreter` works on runtime version 20210119 but not on version 20220308 (Windows 11)

benank, 2022-09-25:
### Description

I am using Windows 11. I was having the exact [same issue as this person](https://github.com/google-coral/pycoral/issues/57) where `make_interpreter` fails silently, but then I installed the earlier recommended runtime from [this reply](https://github.com/google-coral/pycoral/issues/57#issuecomment-981782175) and it started working. Is there an issue with the latest runtime, or am I missing something?

From my testing:
 - 20210119: Works!
 - 20210726: Does not work
 - 20220308: Does not work

On Linux (Raspberry Pi 4), it works flawlessly on the latest version.

Thanks.

### Issue Type

Bug, Build/Install

### Operating System

Windows 10

### Coral Device

USB Accelerator

### Other Devices

Rapsberry Pi 4

### Programming Language

Python 3.8, Python 3.9

hjonnala, 2022-09-26:
Yes, there is an issue with latest edgeTPU runtime for windows platform.

nmaas87, 2022-10-01:
I am seeing the same thing with Windows 10. Is there an update planed / which issue does prevent this? Is it still the same as https://github.com/google-coral/pycoral/issues/39 ?

hjonnala, 2022-10-02:
Hello @nmaas87 , @benank 
Please try the below steps to fix the issues with edgetpu runtime 14.

- uninstall any previous runtimes.
- install edgetpu runtime from edgetpu_runtime_20220308 zip file: https://coral.ai/software/#edgetpu-runtime
- install `edgetpu_runtime_20220308/edgetpu_runtime/third_party/usbdk/UsbDk_1.0.22_x64` manually.
- replace `c:/windows/system32/libusb-1.0.dll` with `edgetpu_runtime_20210119/edgetpu_runtime/third_party/libsub_win/libusb-1.0.dll`

nmaas87, 2022-10-02:
Thank you very much @hjonnala - just tried it and it did work perfectly on Windows 10 x64! 💯

closed 2022-10-04 by hjonnala