google-coral/pycoral #92: Pycoral python API accuracy loss when running inference on a series of images.

hillyuyichu, 2022-10-06:
### Description

I'm running Pycoral python API on my Raspberry Pi 4 with Coral USB Accelerator. 

My program basically captures images from camera and upload them to the google cloud. Before going to the cloud, the images go through the image classification with pre-trained tflite model (used Google's vertex AI service). 

My problem is when I used my program, the image classification suffers an accuracy loss and delivers false positives and negatives. However, when I input a single image that was taken earlier into the Pycoral's example program [classify_image.py](https://github.com/google-coral/pycoral/blob/master/examples/classify_image.py),  I get a very high accuracy ( like 90 % ish). Same thing for all the other images in the cloud. 

Here is my main code for  classification:
[ classification.py ](https://github.com/hillyuyichu/Pycoral-python-API/blob/main/classification.py)
[config.py](https://github.com/hillyuyichu/Pycoral-python-API/blob/main/config.py)  (needed in classification.py)

PS. I have no problem with the inference speed, just the prediction results.

### Issue Type

Performance

### Operating System

Linux

### Other Devices

Rapsberry Pi 4

### Programming Language

Python 3.9

hjonnala, 2022-10-24:
Hello @hillyuyichu seems to be you are creating multiple interpreters for different classifiers on single TPU at: https://github.com/hillyuyichu/Pycoral-python-API/blob/main/pycoral_classification.py#L163. If you want to run/load multiple models on same TPU either you have compile the models with [co comiplation](https://coral.ai/docs/edgetpu/compiler/#co-compiling-multiple-models), recommended only if all the model sizes are less than 8MB, or you have to create the interpreter each time when switching form one model to another model.

hillyuyichu, 2022-10-24:
Thank you for the reply @hjonnala ! 

I'm loading 5 models into a library but only one is set as active and being run each time. [config.json](https://github.com/hillyuyichu/Pycoral-python-API/blob/05f175a5197eef2c14c12b5db7bf2326cc765fec/config.json#L34)

Here is how my program looks like on front end[ UI screenshot ](https://github.com/hillyuyichu/Pycoral-python-API/blob/main/Screenshot_20221024_012526.png)

Here is how the program works in the main [classify.start()](https://github.com/hillyuyichu/Pycoral-python-API/blob/179c0a5c39cc685610a5b52790f0f1eb86f30079/main.py#L106)

All my models are under 8MB and exported as Edge TPU TF lite.

hillyuyichu, 2022-10-24:
Is there an updated version for downloading EdgeTPU compiler?

This seems to be deprecated:

```
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -

echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list

sudo apt-get update

sudo apt-get install edgetpu-compiler
```

hjonnala, 2022-10-25:
can you try creating the interpreter inside the for loop at: https://github.com/hillyuyichu/Pycoral-python-API/blob/main/pycoral_classification.py#L70 instead pre loading the models. 

We don't have any updated version of EdgeTPU compiler. Fell free to use the [colab tutorial](https://colab.sandbox.google.com/github/google-coral/tutorials/blob/master/compile_for_edgetpu.ipynb) if you are having any issues.

hillyuyichu, 2022-10-26:
I've moved the interpreter inside the for loop. It doesn't seem to change anything. 

I have also done the co-compilation for two of my models. The performance has improved a little bit. Before, it was way off. However, it still doesn't give the correct predictions. 
For example, when I put an empty pan in front of the camera, I got a prediction result of 0.80 for empty_pan label on your Pycoral's example code [classfiy_image.py](https://github.com/google-coral/pycoral/blob/master/examples/classify_image.py). But when I run it on mine classification.py, I get a prediction result of 0.11 for empty_pan label. My classification.py is bascially the same as the Pycoral's example except there's an dictionary added.