google-coral/pycoral #94: Coral EdgeTPU USB Support - Not recognized/not changing vendor ID

dollarusername2, 2022-11-26:
### Description

Hey folks,

I am trying to get the awesome frigate project to run with my newly bought Coral EdgeTPU USB.

It works fine on the "old" server (details see below), but somehow only shows as the generic USB device in the new, will not be picked up by frigate and will not change vendor ID.

The strange thing: I am using exactly the same docker container of frigate (tested both, latest stable 0.11.1 and 0.10.1), incl. same docker-compose.yaml and frigate.conf file. (Only thing I changed is the name of the HDD volumes). Even the Host OS (Ubuntu Server) is nearly identical. Old server running 22.04.0, new 22.04.1. Both machines where only used for frigate, so no other software installed.

Yet, somehow, the Coral does not seem to work with the newer server.

I tried direct connection, different USB Cable, and two types of external powered HUBs. ALL of them work at the old server, none at the new. So I guess this is not really related to hardware.

The hardware of the "new" (used) server is actually pretty mainstream: Fujitsu ESPRIMO P756 MT | i5-6500 CPU.

Do you have any idea where/how to start debugging this?

Thanks a lot!

PS: The frigate projected pointed me here - if this is the wrong place for this question, could you kindly point me to the right project? Thanks!

### Issue Type

Support

### Operating System

Ubuntu

### Coral Device

USB Accelerator

### Programming Language

Other

### Relevant Log Output

```shell
frigate  | Traceback (most recent call last):
frigate  |   File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
frigate  |     delegate = Delegate(library, options)
frigate  |   File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
frigate  |     raise ValueError(capture.message)
frigate  | ValueError
frigate  | 
frigate  | During handling of the above exception, another exception occurred:
frigate  | 
frigate  | Traceback (most recent call last):
frigate  |   File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
frigate  |     self.run()
frigate  |   File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
frigate  |     self._target(*self._args, **self._kwargs)
frigate  |   File "/opt/frigate/frigate/edgetpu.py", line 135, in run_detector
frigate  |     object_detector = LocalObjectDetector(
frigate  |   File "/opt/frigate/frigate/edgetpu.py", line 43, in __init__
frigate  |     edge_tpu_delegate = load_delegate("libedgetpu.so.1.0", device_config)
frigate  |   File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
frigate  |     raise ValueError('Failed to load delegate from {}\n{}'.format(
frigate  | ValueError: Failed to load delegate from libedgetpu.so.1.0
frigate  |
```

soysauce10, 2022-11-27:
I'm also trying to fix something similar with my setup, but have you seen this? https://github.com/blakeblackshear/frigate/discussions/2599#discussioncomment-3621686

dollarusername2, 2022-11-27:
Hey @soysauce10 ,

Thanks for pointing this out. I tried this at the beginning, but it didn't help. Just to make sure: I just tried again to move the USB from devices to volumes and it didn't help... 

Where are you stuck?

hjonnala, 2022-11-28:
Hi @dollarusername2 can you try  the demo adding  the  below lines at: https://github.com/google-coral/pycoral/blob/master/examples/classify_image.py#L41 and share the logs..

```
from pycoral.pybind._pywrap_coral import SetVerbosity as set_verbosity
set_verbosity(10)
```

dollarusername2, 2022-11-28:
Good evening @hjonnala ,

the demo works fine (after installing virtualenv and the old python...). Vendor ID is changed after running the demo. (However frigate is still not working).

Anyway, please find the verbose logs attached. 

`(pycoral-test) mira@esprimo:~/pycoral-test/pycoral$ python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
I tflite/edgetpu_manager_direct.cc:453] No matching device is already opened for shared ownership.
I driver/driver_factory_default.cc:31] Failed to open /sys/class/apex: No such file or directory
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x1a6e, product:0x89a
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x18d1, product:0x9302
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:998] EnumerateDevices: found [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/beagle/beagle_usb_driver_provider.cc:225] Enumerate: adding path [/sys/bus/usb/devices/2-3]
I tflite/edgetpu_manager_direct.cc:471] No device of type Apex (PCIe) is available.
I tflite/edgetpu_context_direct.cc:106] USB always DFU: False (default)
I tflite/edgetpu_context_direct.cc:147] USB bulk-in queue capacity: 8
I tflite/edgetpu_context_direct.cc:63] Performance expectation: High when USB connected EdgeTpu is throttled
I driver/usb/usb_driver.cc:1383] Open device and check if DFU is needed
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:1081] OpenDevice: device opened 0x561e7b38d730
I driver/usb/local_usb_device.cc:184] LocalUsbDevice
I driver/usb/usb_standard_commands.cc:36] UsbStandardCommands
I driver/usb/usb_dfu_commands.cc:37] UsbDfuCommands
I driver/usb/usb_standard_commands.cc:43] GetDeviceDescriptor
I driver/usb/local_usb_device.cc:398] GetDescriptor
I driver/usb/usb_standard_commands.cc:78] Vender ID: 0x18d1
I driver/usb/usb_standard_commands.cc:79] Product ID: 0x9302
I driver/usb/usb_driver.cc:1410] Device is already in application mode, skipping DFU
I driver/usb/usb_driver.cc:1422] Resetting device
I driver/usb/local_usb_device.cc:243] Close: closing device 0x561e7b38d730 
I driver/usb/local_usb_device.cc:216] DoCancelAllTransfers: cancelling 0 async transfers
I driver/usb/local_usb_device.cc:224] DoCancelAllTransfers: waiting for all async transfers to complete
I driver/usb/local_usb_device.cc:234] DoCancelAllTransfers: all async transfers have completed
I driver/usb/local_usb_device.cc:276] Close: releasing 0 transfer buffers
I driver/usb/local_usb_device.cc:289] Close: performing graceful reset
I driver/usb/local_usb_device.cc:322] Close: final clean up completed
I driver/usb/usb_driver.cc:1364] Opening device expecting application mode
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:1081] OpenDevice: device opened 0x561e7b397de0
I driver/usb/local_usb_device.cc:184] LocalUsbDevice
I driver/usb/usb_standard_commands.cc:36] UsbStandardCommands
I driver/usb/usb_ml_commands.cc:47] UsbMlCommands
I driver/usb/usb_dfu_commands.cc:40] ~UsbDfuCommands
I driver/usb/usb_standard_commands.cc:39] ~UsbStandardCommands
I driver/usb/local_usb_device.cc:196] ~LocalUsbDevice
I driver/usb/local_usb_device.cc:243] Close: closing device (nil) 
I driver/usb/local_usb_device.cc:352] ClaimInterface
I driver/usb/usb_ml_commands.cc:80] ReadRegister32 offset 0x1a30c
I driver/usb/local_usb_device.cc:514] SendControlCommandWithDataIn
I driver/usb/local_usb_device.cc:521] SYNC CTRL WITH DATA IN begin
I driver/usb/local_usb_device.cc:538] SYNC CTRL WITH DATA IN end
I driver/usb/usb_ml_commands.cc:110] ReadRegister32 [0x1A30C] == 0xF0059
I driver/usb/usb_ml_commands.cc:153] WriteRegister32 [0x1A30C] := 0xF0059
I driver/usb/local_usb_device.cc:475] SendControlCommandWithDataOut
I driver/usb/local_usb_device.cc:482] SYNC CTRL WITH DATA OUT begin
I driver/usb/local_usb_device.cc:499] SYNC CTRL WITH DATA OUT end
I driver/usb/usb_ml_commands.cc:80] ReadRegister32 offset 0x1a314
I driver/usb/local_usb_device.cc:514] SendControlCommandWithDataIn
I driver/usb/local_usb_device.cc:521] SYNC CTRL WITH DATA IN begin
I driver/usb/local_usb_device.cc:538] SYNC CTRL WITH DATA IN end
I driver/usb/usb_ml_commands.cc:110] ReadRegister32 [0x1A314] == 0x110000
I driver/usb/usb_ml_commands.cc:80] ReadRegister32 offset 0x1a318
[... 1228 lines cut ...]
I driver/usb/local_usb_device.cc:482] SYNC CTRL WITH DATA OUT begin
I driver/usb/local_usb_device.cc:499] SYNC CTRL WITH DATA OUT end
I driver/usb/usb_ml_commands.cc:80] ReadRegister32 offset 0x1a318
I driver/usb/local_usb_device.cc:514] SendControlCommandWithDataIn
I driver/usb/local_usb_device.cc:521] SYNC CTRL WITH DATA IN begin
I driver/usb/local_usb_device.cc:538] SYNC CTRL WITH DATA IN end
[... cut ...]
I driver/usb/local_usb_device.cc:263] Close: releasing claimed interface 0
I driver/usb/local_usb_device.cc:216] DoCancelAllTransfers: cancelling 0 async transfers
I driver/usb/local_usb_device.cc:224] DoCancelAllTransfers: waiting for all async transfers to complete
I driver/usb/local_usb_device.cc:234] DoCancelAllTransfers: all async transfers have completed
I driver/usb/local_usb_device.cc:276] Close: releasing 0 transfer buffers
I driver/usb/local_usb_device.cc:322] Close: final clean up completed`

hjonnala, 2022-11-29:
Can you please share similar logs form the frigate container and the output of `lsusb` command inside the container.

dollarusername2, 2022-12-01:
Hello @hjonnala ,

thanks for helping me out! Unfortunately there is no lsusb inside that specific container. However I managed to run the test example:

`@757605ae6e65:~/pycoral-test/tmp/pycoral-test/pycoral# python3 examples/classify_image.py \
>   --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite  \
>   --labels test_data/inat_bird_labels.txt \
>   --input test_data/parrot.jpg
I tflite/edgetpu_manager_direct.cc:453] No matching device is already opened for shared ownership.
I driver/driver_factory_default.cc:31] Failed to open /sys/class/apex: No such file or directory
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x1a6e, product:0x89a
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x18d1, product:0x9302
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:998] EnumerateDevices: found [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/beagle/beagle_usb_driver_provider.cc:225] Enumerate: adding path [/sys/bus/usb/devices/2-3]
I tflite/edgetpu_manager_direct.cc:471] No device of type Apex (PCIe) is available.
I tflite/edgetpu_context_direct.cc:106] USB always DFU: False (default)
I tflite/edgetpu_context_direct.cc:147] USB bulk-in queue capacity: 8
I tflite/edgetpu_context_direct.cc:67] Performance expectation: Max (default)
I driver/usb/usb_driver.cc:1383] Open device and check if DFU is needed
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
I driver/usb/local_usb_device.cc:60] ConvertLibUsbError: USB error -3 [OpenDevice]
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/2-3]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[3]
and so on and so on...`

hjonnala, 2022-12-02:
> I tflite/edgetpu_context_direct.cc:401] Failed to open device [Apex (USB)] at [/sys/bus/usb/devices/2-3]: Permission denied: USB error -3 [OpenDevice]

can you try the example with sudo and share the dockerfile and docker command used to run the container..

dollarusername2, 2022-12-02:
As seen in the log, there is only root inside the container, and the test was run as root (inside the container): **root**@757605ae6e65:~/pycoral-test/tmp/pycoral-test/pycoral# python3 examples/classify_image.py

The container itself is run in a rootless docker setup via command "docker compose up".

The docker-compose.yaml is here:

### docker-compose.yaml

```yaml
version: "3.9"
services:
  frigate:
    container_name: frigate
    privileged: true # this may not be necessary for all setups
    restart: unless-stopped
    image: blakeblackshear/frigate:stable
    shm_size: "64mb" # update for your cameras based on calculation above
    devices:
      - /dev/bus/usb:/dev/bus/usb # passes the USB Coral, needs to be modified for other versions
      #- /dev/apex_0:/dev/apex_0 # passes a PCIe Coral, follow driver instructions here https://coral.ai/docs/m2/get-started/#2a-on-linux
      - /dev/dri/renderD128 # for intel hwaccel, needs to be updated for your hardware
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /home/mira/frigate/config.yml:/config/config.yml:ro
      - /media/hdd/frigate-media:/media/frigate
      - type: tmpfs # Optional: 1GB of memory, reduces SSD/SD Card wear
        target: /tmp/cache
        tmpfs:
          size: 1000000000
    ports:
      - "5000:5000"
      - "1935:1935" # RTMP feeds
    environment:
      FRIGATE_RTSP_PASSWORD: "pw"
```

hjonnala, 2022-12-02:
The initial steps mentioned [here](https://github.com/blakeblackshear/frigate/discussions/2599#discussioncomment-3621686) working for me with `docker run -ti --rm --entrypoint=/bin/bash --privileged -v /dev/bus/usb:/dev/bus/usb blakeblackshear/frigate:stable`

Can you try adding `- /dev/bus/usb:/dev/bus/usb ` to volumes instead of devices..Thanks!

dollarusername2, 2022-12-02:
Hi, as written in the issue at the frigate project: unfortunately this did not help in my case :-/

(To be sure, I just checked again)

dollarusername2, 2022-12-08:
Any ideas left?

hjonnala, 2022-12-09:
Please try with the below steps..if its not working please share the [verbose logs](https://github.com/google-coral/pycoral/issues/94#issuecomment-1328604113).

1. run `docker run -ti --rm --entrypoint=/bin/bash --privileged --mount type=bind,source=/dev,target=/dev blakeblackshear/frigate:stable
`

2. inside container, run
```
cd ~ \
 && apt-get -y update \
 && apt-get -y install curl \
 && mkdir test_data \
 && cd test_data \
 && curl -LO https://coral.ai/static/docs/images/parrot.jpg \
 && curl -LO https://raw.githubusercontent.com/google-coral/test_data/master/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
 && curl -LO https://raw.githubusercontent.com/google-coral/test_data/master/inat_bird_labels.txt \
 && cd .. \
 && curl -LO https://raw.githubusercontent.com/google-coral/pycoral/master/examples/classify_image.py \
 && python3 classify_image.py \
  --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite  \
  --labels test_data/inat_bird_labels.txt \
  --input test_data/parrot.jpg
```

dollarusername2, 2022-12-09:
Hi @hjonnala ,

thanks for helping me!

Running `1` just results in an error:#
`$ docker run -ti --rm --entrypoint=/bin/bash --privileged --mount type=bind,source=/dev,target=/dev blakeblackshear/frigate:stable
docker: Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: open /dev/console: permission denied: unknown.`

Running with `sudo` the container starts and seems to be able to access the frigate:
`
----INFERENCE TIME----
Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
11.8ms
2.8ms
2.7ms
2.7ms
2.9ms
-------RESULTS--------
Ara macao (Scarlet Macaw): 0.76953
`

There seems to be some issue with the rootless docker setup?

hjonnala, 2022-12-09:
> There seems to be some issue with the rootless docker setup?

I think so, can you try the instructions at : https://linuxhandbook.com/rootless-docker/ or https://docs.docker.com/engine/security/rootless/

dollarusername2, 2022-12-09:
Hi @hjonnala , 

I double just the install (actually just did it again) and the error still persists. I am checking with the frigate people at their project, If anybody has an idea. Checking google this might also be a permission issue at the docker file. However, why would this only occur at the new machine? 

Strange issue...

hjonnala, 2022-12-12:
can you try the below commands and `reboot` the device and try the demo with docker.

```
sudo addgroup docker
sudo usermod -aG docker $USER 
```

dollarusername2, 2022-12-12:
Hi @hjonnala ,

The group was already exisiting and my user was already part of the user group. Anyway, I added it again, rebooted, and (no surprise): Issue stays the same like in https://github.com/google-coral/pycoral/issues/94#issuecomment-1344118195 :-/

dollarusername2, 2022-12-12:
Wohoo... that is a good tip!

Somehow the $DOCKER_HOST was missing from my env.

When I set it manually in my .bashrc, it works! Yeah!

Last issue left: Obviously I do want to start docker inkl. frigate at reboot automatically without login. This used to work with the systemctl approach, however the $DOCKER_HOST seems to be missing in this context as well. I need to research how to fix this (unless anybody knows by chance?).

Thanks!!!

closed 2022-12-12 by hjonnala