google-coral/pycoral #97: Import error when running classify_image.py

GameMonkey, 2022-12-02:
### Description

I have used the Google Coral Dev Board's Getting Started guide and installed mendel. I have also installed pycoral, but when running the command

```
python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
```

I get the error:
```
Traceback (most recent call last):
  File "examples/classify_image.py", line 40, in <module>
    from pycoral.utils.edgetpu import make_interpreter
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: No module named 'pycoral.pybind._pywrap_coral'
```

When using `pip3 list -v`, I can see that the versions of the modules are:

- pycoral (2.0.0)
- tflite-runtime (2.5.0.post1)
- edgetpu (1.9.2)
- edgetpuvision (1.0)

The Python version I am running is Python 3.5.2

Any suggestions on how to fix the issue? Thanks

### Issue Type

Build/Install

### Operating System

Mendel Linux

### Coral Device

Dev Board

### Programming Language

Python 3.5

### Relevant Log Output

```shell
python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 40, in <module>
    from pycoral.utils.edgetpu import make_interpreter
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: No module named 'pycoral.pybind._pywrap_coral'
```

hjonnala, 2022-12-02:
Hello @GameMonkey Dev board comes with python3.7.3 and no need to install any packages(pycoarl) to run the demo. 

Please [reflash](https://coral.ai/docs/dev-board/get-started/#requirements) the board and try the demo. Thanks!!

GameMonkey, 2022-12-06:
Hey, I have problems installing new versions of Mendel, but I made a separate [issue](https://github.com/google-coral/edgetpu/issues/688) for that. So let us close this issue for now.

closed 2022-12-06 by GameMonkey